esphome/issues #2771: bang bang realy sometimes stays on idle

IsaacBreuer, 2021-12-01:
### The problem

I use the bang bang climate for controlling my waterheater,  it is working nice for over a year. however I recently discovered the realy sometimes stays on even the max temperature was reached,  the climate status shows as idle but realy is on, it just happens every now and then. 
was wondering if that's a bug or there is something wrong with my setup.
I can have some automation in ha to check for climate is ideal and realy is on, and trigger the realy off, but since its a waterheater I realy want the esp device to handle it on its own. 

### Which version of ESPHome has the issue?

1.18.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5. (also older versions)

### What platform are you using?

ESP8266

### Board

sonoff RE5V1C (5v dry contact)

### Component causing the issue

bang bang

### Example YAML snippet

```yaml
esphome:
  name: waterheater
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2
  esp8266_restore_from_flash: true
  
  

wifi:
  ssid: redacted
  password: redacted 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Waterheater Fallback Hotspot"
    password: redacted

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Water Heater Button"
    on_press:
      - switch.toggle: relay
# Enable Web server.
web_server:
  port: 80
switch:
  - platform: gpio
    restore_mode: RESTORE_DEFAULT_OFF
    name: "Water Heater Relay"
    pin: GPIO12
    id: relay

dallas:
  - pin: 2
    id: "dallas_hub_2"
    update_interval: 10s

sensor:
  - platform: dallas
    address: 0x123C01B556C41528
   # index: 0
    name: "Tank Temperature"
    id: temperature1
    dallas_id: dallas_hub_2
#  - platform: dallas
  #  address: 0xDD0000031EFB0428
  #  name: "Pipe Temperature"
    
climate:
  - platform: bang_bang
    name: "Water Heater controller"
    sensor: temperature1
    default_target_temperature_low:  51.6667 °C
    default_target_temperature_high: 57.2222 °C

    heat_action:
      - switch.turn_on: relay
    idle_action:
      - switch.turn_off: relay
status_led:
  pin:
    number: GPIO13
    inverted: yes
```

martgras, 2021-12-02:
Can you add logs that show the state of the relay. 
Do you have a multimeter to measure the gpio pin when the relay is on but should be off? 
And finally does the relay turn off if you knock on it (it might stick due to micro welding)

IsaacBreuer, 2021-12-12:
controlling the realy with onboard button or even in Homeassistant (realy is exposed to ha) turns the realy off, so I don't expect it to be a mechanical issue. 
I assume it related to Homeassistant restart or unavailable when the relay has to go off although the automation is not handled be ha at all.

martgras, 2021-12-12:
That's the reason why I asked for logs

closed 2022-04-26 by github-actions[bot]