esphome/issues #2773: I get an error compiling a device that has a pulse_meter_sensor implementation

VdR1960, 2021-12-02:
### The problem

When compiling an error is thrown. This started after updating, without changing the yaml file. If I comment out the sensor implementation all is well (well, except not having the sensor I need). See snippet from the log below.

[image]

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

pulse_meter

### Example YAML snippet

```yaml
sensor:

  # Water flow sensor
  - platform: pulse_meter
    name: "Water flow"
    pin: D1
    unit_of_measurement: "ltr/min"
    internal_filter: 100ms
    accuracy_decimals: 1
    timeout: 30s                              # mininum flow is 2.0 ltr/min

    total:      
      name: 'Water consumption'
      unit_of_measurement: 'm3'
      accuracy_decimals: 3
      
      filters:
        - multiply: 0.001                     # 1.0 ltr/imp / 1000 ltr/m3
```

### Additional information

This device "water" also started showing the name of the yaml file on the card. No idea what that means or if it is related. See screenshot.

[image]

martgras, 2021-12-02:
It seems that not all of your files are up to date. 
pulse_meter_sensor.h was changed in September to include `hal.h` instead of `esphal.h`
That change is included in 2021.11.4. 
Try if removing the folder  /data/water/.pioenvs fixes the issue

VdR1960, 2021-12-02:
I tried long and hard, but I cannot find a /data directory or a .pioenvs file. I searched the whole homeassistant directory through samba access. Where should I look?

I did find the pulse_meter_sensor.h file that tries to include esphal.h. Changing that call to hal.h does not help. ?

martgras, 2021-12-03:
Did you try to use "Clean Build Files?" - Maybe that resolves the problem already
Otherwise the files to delete are in the esphome holder .  

You can either delete the complete build folder 
[image]

or drill down to the water folder and delete just that 
(In my case the device is called sdm-120 you should see water instead
[image]

VdR1960, 2021-12-03:
Thanks again for your support.

Unfortunately I keep getting the same error.

I tried 'Clean Build Files, that gave me hope (see below), but did not help.

[image]

I deleted the /build/water directory, but did not help.

There still is the #include "esphome/core/esphal.h" statement in the pulse_meter_sensor.h file.

[image]

That file gets rebuild with that same include statement every time I compile. No matter if I change the include statement in the file or delete the file.

What's next?

VdR1960, 2021-12-03:
You mentioned a change with version 2021.11.4. I noticed that the version number in \\homeassistant\config\esphome\water\src\esphome\core is not the same as the version in the log. Can that be related?

[image]

[image]

martgras, 2021-12-03:
Yes that is the problem but I'm not familiar enough with the HA Addon to dig deeper. 
Personally I'd backup my yaml files and remove and reinstall the ESPHome add-on but that is not an "offical" advise

VdR1960, 2021-12-03:
Yes I thought of that, but I'm afraid I then cannot connect to the devices over wireless anymore. I would hate to dig them out the wall to get them going again.

Thank you for your help.

ssieb, 2021-12-05:
You can remove the add-on and re-install it.  That won't remove your yaml configs (you should be backing them up anyway).  And as long as the config is the same, they will all still work after.

VdR1960, 2021-12-06:
This was all related to failed updates of ESPHome. An un-install, cleanup, re-install has fixed the issue.

closed 2021-12-06 by VdR1960