esphome/issues #2774: ESP32 WROOM - Does not compile

p0rc0jet, 2021-12-02:
### The problem

Tried to flash new ESP32 WROOM, firmware doesn't compile.
Tried to update old one with no serial at all, same issue.

### Which version of ESPHome has the issue?

2022.1.0.dev0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

wifi_component_esp32.cpp.

### Example YAML snippet

```yaml
esphome:
  name: iot-sen-volt
  platform: ESP32
  board: nodemcu-32s

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: "123"

ota:
  password: "123"

wifi:
  ssid: "home_assistant_01"
  password: "123"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ha Fallback Hotspot"
    password: "123"

captive_portal:

uart:
  - id: volt_uart
    tx_pin: 13
    rx_pin: 12
    baud_rate: 9600
    stop_bits: 1

sensor:
  - platform: pzem004t
    current:
      name: "volt - current"
    voltage:
      name: "volt - voltage"
    power:
      name: "volt - power"
    energy:
      name: "volt - energy"
    update_interval: 60s

  - platform: wifi_signal
    name: "volt - wifi"
    update_interval: 3600s
```

### Anything in the logs that might be useful for us?

```txt
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling .pioenvs/iot-sen-volt/src/esphome/components/uart/uart_esp32.cpp.o
Compiling .pioenvs/iot-sen-volt/src/esphome/components/wifi/wifi_component_esp32.cpp.o
In file included from src/esphome/core/hal.h:4:0,
                 from src/esphome/core/application.h:8,
                 from src/esphome/components/wifi/wifi_component_esp32.cpp:19:
src/esphome/core/gpio.h:7:0: warning: "LOG_PIN" redefined
 #define LOG_PIN(prefix, pin) \
 ^
In file included from src/esphome/components/wifi/wifi_component_esp32.cpp:18:0:
src/esphome/core/esphal.h:32:0: note: this is the location of the previous definition
 #define LOG_PIN(prefix, pin) \
 ^
src/esphome/components/uart/uart_esp32.cpp:23:36: error: no 'uint32_t esphome::uart::UARTComponent::get_config()' member function declared in class 'esphome::uart::UARTComponent'
 uint32_t UARTComponent::get_config() {
                                    ^
src/esphome/components/uart/uart_esp32.cpp:70:27: error: no 'void esphome::uart::UARTComponent::setup()' member function declared in class 'esphome::uart::UARTComponent'
 void UARTComponent::setup() {
                           ^
src/esphome/components/uart/uart_esp32.cpp:91:33: error: no 'void esphome::uart::UARTComponent::dump_config()' member function declared in class 'esphome::uart::UARTComponent'
 void UARTComponent::dump_config() {
                                 ^
src/esphome/components/uart/uart_esp32.cpp:107:6: error: redefinition of 'void esphome::uart::UARTComponent::write_byte(uint8_t)'
 void UARTComponent::write_byte(uint8_t data) {
      ^
In file included from src/esphome/components/uart/uart.h:7:0,
                 from src/esphome/components/uart/uart_esp32.cpp:2:
src/esphome/components/uart/uart_component.h:35:8: note: 'void esphome::uart::UARTComponent::write_byte(uint8_t)' previously defined here
   void write_byte(uint8_t data) { this->write_array(&data, 1); };
        ^
src/esphome/components/uart/uart_esp32.cpp: In member function 'virtual void esphome::uart::UARTComponent::write_array(const uint8_t*, size_t)':
src/esphome/components/uart/uart_esp32.cpp:112:9: error: 'class esphome::uart::UARTComponent' has no member named 'hw_serial_'
   this->hw_serial_->write(data, len);
         ^
src/esphome/components/uart/uart_esp32.cpp: At global scope:
src/esphome/components/uart/uart_esp32.cpp:117:6: error: redefinition of 'void esphome::uart::UARTComponent::write_str(const char*)'
 void UARTComponent::write_str(const char *str) {
      ^

In file included from src/esphome/components/uart/uart.h:7:0,
                 from src/esphome/components/uart/uart_esp32.cpp:2:
src/esphome/components/uart/uart_component.h:36:8: note: 'void esphome::uart::UARTComponent::write_str(const char*)' previously defined here
   void write_str(const char *str) {
        ^
src/esphome/components/uart/uart_esp32.cpp:121:6: error: redefinition of 'bool esphome::uart::UARTComponent::read_byte(uint8_t*)'
 bool UARTComponent::read_byte(uint8_t *data) {
      ^
In file included from src/esphome/components/uart/uart.h:7:0,
                 from src/esphome/components/uart/uart_esp32.cpp:2:
src/esphome/components/uart/uart_component.h:43:8: note: 'bool esphome::uart::UARTComponent::read_byte(uint8_t*)' previously defined here
   bool read_byte(uint8_t *data) { return this->read_array(data, 1); };
        ^
src/esphome/components/uart/uart_esp32.cpp: In member function 'virtual bool esphome::uart::UARTComponent::peek_byte(uint8_t*)':
src/esphome/components/uart/uart_esp32.cpp:131:17: error: 'class esphome::uart::UARTComponent' has no member named 'hw_serial_'
   *data = this->hw_serial_->peek();
                 ^
src/esphome/components/uart/uart_esp32.cpp: In member function 'virtual bool esphome::uart::UARTComponent::read_array(uint8_t*, 
size_t)':
src/esphome/components/uart/uart_esp32.cpp:137:9: error: 'class esphome::uart::UARTComponent' has no member named 'hw_serial_'
   this->hw_serial_->readBytes(data, len);
         ^
src/esphome/components/uart/uart_esp32.cpp: In member function 'virtual int esphome::uart::UARTComponent::available()':
src/esphome/components/uart/uart_esp32.cpp:158:47: error: 'class esphome::uart::UARTComponent' has no member named 'hw_serial_'
 int UARTComponent::available() { return this->hw_serial_->available(); }
                                               ^
src/esphome/components/uart/uart_esp32.cpp: In member function 'virtual void esphome::uart::UARTComponent::flush()':
src/esphome/components/uart/uart_esp32.cpp:161:9: error: 'class esphome::uart::UARTComponent' has no member named 'hw_serial_'
   this->hw_serial_->flush();
         ^
*** [.pioenvs/iot-sen-volt/src/esphome/components/uart/uart_esp32.cpp.o] Error 1
In file included from src/esphome/core/hal.h:4:0,
                 from src/esphome/core/application.h:8,
                 from src/esphome/components/wifi/wifi_component_esp32.cpp:19:
src/esphome/core/gpio.h:50:7: error: redefinition of 'class esphome::GPIOPin'
 class GPIOPin {
       ^

In file included from src/esphome/components/wifi/wifi_component_esp32.cpp:18:0:
src/esphome/core/esphal.h:70:7: error: previous definition of 'class esphome::GPIOPin'
 class GPIOPin {
       ^
In file included from src/esphome/core/hal.h:4:0,
                 from src/esphome/core/application.h:8,
                 from src/esphome/components/wifi/wifi_component_esp32.cpp:19:
src/esphome/core/gpio.h:66:7: error: redefinition of 'class esphome::ISRInternalGPIOPin'
 class ISRInternalGPIOPin {
       ^
In file included from src/esphome/components/wifi/wifi_component_esp32.cpp:18:0:
src/esphome/core/esphal.h:41:7: error: previous definition of 'class esphome::ISRInternalGPIOPin'
 class ISRInternalGPIOPin {
       ^
In file included from src/esphome/core/hal.h:4:0,
                 from src/esphome/core/application.h:8,
                 from src/esphome/components/wifi/wifi_component_esp32.cpp:19:
src/esphome/core/gpio.h:91:8: error: 'bool esphome::InternalGPIOPin::is_internal()' marked 'override', but does not override
   bool is_internal() override { return true; }
        ^

src/esphome/components/wifi/wifi_component_esp32.cpp:142:39: error: no 'IPAddress esphome::wifi::WiFiComponent::wifi_sta_ip_()' member function declared in class 'esphome::wifi::WiFiComponent'
 IPAddress WiFiComponent::wifi_sta_ip_() {
                                       ^
src/esphome/components/wifi/wifi_component_esp32.cpp:562:45: error: no 'wl_status_t esphome::wifi::WiFiComponent::wifi_sta_status_()' member function declared in class 'esphome::wifi::WiFiComponent'
 wl_status_t WiFiComponent::wifi_sta_status_() { return WiFiClass::status(); }
                                             ^
src/esphome/components/wifi/wifi_component_esp32.cpp:697:11: error: prototype for 'IPAddress esphome::wifi::WiFiComponent::wifi_soft_ap_ip()' does not match any in class 'esphome::wifi::WiFiComponent'
 IPAddress WiFiComponent::wifi_soft_ap_ip() {
           ^
In file included from src/esphome/components/wifi/wifi_component_esp32.cpp:1:0:
src/esphome/components/wifi/wifi_component.h:228:22: error: candidate is: esphome::network::IPAddress esphome::wifi::WiFiComponent::wifi_soft_ap_ip()
   network::IPAddress wifi_soft_ap_ip();
                      ^
*** [.pioenvs/iot-sen-volt/src/esphome/components/wifi/wifi_component_esp32.cpp.o] Error 1
========================== [FAILED] Took 2.21 seconds ==========================
```

martgras, 2021-12-02:
Can you please upgrade your version.  There was a lot of refactoring happing and the current version doesn't even use uart_esp32.cpp anymore

p0rc0jet, 2021-12-02:
Did totally fresh git clone. 
Everything compiled with flying colors. 
Thank you.

closed 2021-12-02 by p0rc0jet