esphome/issues #2775: MCP23017 inputs aren't properly configured on boot

ficueu, 2021-12-02:
### The problem

If previous firmware set **_pullup: true_** - pullup is enabled, next while trying to upload new firmware with changed _**pullup: false**_ input has still enabled pullup resistor. Looks like MCP remember last set registers while ESPHome not setting those properly after reboot.

Steps to reproducing this problem:
1. Measured voltage after uploading - pin 0 to GND - should be ~VCC

```
mcp23017:
  - id: 'mcp23017_hub'
    address: 0x20

binary_sensor:
  - platform: gpio
    name: "MCP23017 Pin #1"
    pin:
      mcp23xxx: mcp23017_hub
      number: 0
      mode:
        input: true
        pullup: true
      inverted: false
```
2. Measured voltage after uploading - pin 0 to GND - should be 0.

```
mcp23017:
  - id: 'mcp23017_hub'
    address: 0x20

binary_sensor:
  - platform: gpio
    name: "MCP23017 Pin #1"
    pin:
      mcp23xxx: mcp23017_hub
      number: 0
      mode:
        input: true
        pullup: false
      inverted: false
```
Next I wrote sample sketch with Arduino and execute command to disable pullup on pin 0 then measured voltage and it was 0V.

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.5

### What platform are you using?

ESP32

### Board

esp32-dev

### Component causing the issue

mcp23017

oxan, 2021-12-03:
Can you try esphome/esphome#2855?
```yaml
external_components:
- source: github://pr#2855
  components: [ mcp23x17_base ]
```

ficueu, 2021-12-03:
@oxan, tried, working great. Tested on few gpios.

closed 2021-12-06 by jesserockz