esphome/issues #2777: DMSR option REQUEST_PIN cannot be set in configuration .yaml

Niiccoo, 2021-12-03:
### The problem

My d1_mini P1 slimme meter uses D7 to trigger the request to read data form P1 meter. (and not D5). 
When I try to use the setting  request_pin: D7 I get an error. My set up works when I manually set D7 to high. Only the the result are poor and only getting few reliable reading a day.

INFO Reading configuration /config/esphome/meterp1.yaml...
Failed config
dsmr: [source /config/esphome/meterp1.yaml:58]
  id: dsmr_instance
  **[request_pin] is an invalid option for [dsmr]**. Please check the indentation.
  request_pin: D7 [source /config/esphome/meterp1.yaml:60]

The REQUEST_PIN is present in the `dsmr/__init__.py /` file:
`CONF_REQUEST_PIN = "request_pin" `

### Which version of ESPHome has the issue?

ESPHome Current version: 2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

dsmr

### Example YAML snippet

```yaml
uart:
  # D6 is GIO12
  - rx_pin: D6
    baud_rate: 115200
    id: uart_bus
    data_bits: 8
    parity: NONE
    stop_bits: 1

# Test om D7 aan en uit te zetten
switch:
  - platform: gpio
    pin: D7
    name: "Lees van slimmemeter"
    icon: "mdi:power-from-grid"

dsmr:
  id: dsmr_instance
  request_pin: D7
```

mmakaay, 2021-12-03:
Version 2021.11.4 does not have the `request_pin` option yet. This feature was implemented after 2021.11.0 was released, and since it was not a bugfix it didn't get into the 2021.11.x releases.
I don't know why you would see the constant for that option in your `__init__.py`. I checked the 2021.11.4 code, and there's no such definition in the code.

The upcoming 2021.12.0 is the first release that has this feature availble.

mmakaay, 2021-12-03:
Oh, and BTW: better use D7 for RX if you are having issues with reliable data receival. D7 can do hardware UART, D5 can't.

I wrote some info in the docs for this, currently visible at the beta docs:
https://beta.esphome.io/components/sensor/dsmr.html#improving-reader-results

Niiccoo, 2021-12-03:
Thanks for the answer. 

Oops, was not aware that it is not release yet. Will wait for next release. 
Searching for an explanation I got the information form the ESPHome-GITHUB. 

I am aware that D7 would be better. But I already hooked it up to D5 and mount the d1_mini more or less permanent next to the "slimmemeter". So a OTA fix would be perfect. 
 
This is my set up: 

[image]

After reading your docs will start on making a better one. 
Thanks!

mmakaay, 2021-12-03:
If you are going to build a new one, maybe you'd like one of these?

[image]

That's from my [DSMR reader project](https://github.com/mmakaay/dsmr-reader-for-esphome/blob/main/images/PCB_soldered.jpg). I've got some spare PCB's lying around, and these provide request pin support along with hardware signal inversion, and a header for an optional OLED display.

If you're interested, ping me on the [ESPHome Discord](https://discord.gg/KhAMKrd) (@MauriceMakaay there)

mmakaay, 2021-12-10:
Closing this issue, since the upcoming version of ESPHome will have support for the request pin.

closed 2021-12-10 by mmakaay

Niiccoo, 2021-12-12:
Just update to 2021.12.0. It works perfect!

Thanks @mmakaay

mmakaay, 2021-12-13:
Thanks for reporting back, always nice to read a success story 😄 
Enjoy!