esphome/issues #2778: WLED not found! [Unable to find effect with the name 'wled']

xGuIInE, 2021-12-03:
### The problem

I can´t use WLED effect for my fastled_clockless platform light. I have tested another esphome versions, and same problem. Wled component appears to be installed in python pacakes...

Error: Unable to find effect with the name 'wled'.

My template:
`light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO12
    num_leds: 60
    rgb_order: GRB
    name: "FastLED WS2812B Light"
    
    effects:
      - wled:
          port: 21324`

### Which version of ESPHome has the issue?

2021.12.0b2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

light effects wled

martgras, 2021-12-03:
You seem to miss the `wled:` entry 
Maybe you have included it in your yaml but omitted here.

```
wled:
light:

  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO12
    num_leds: 60
    rgb_order: GRB
    name: "FastLED WS2812B Light"
    effects:
      - wled:
         port: 21324
```

Btw: wrap your yaml with 3 backticks to make it easier to read
 \`\`\`
yaml or other code
 \`\`\`

xGuIInE, 2021-12-04:
Absolutely right... I didn't realize

In any case, i can´t connect WLED with hyperion. Changed to e131 and works perfect!

closed 2021-12-04 by xGuIInE