esphome/issues #2780: Climare - Thermostat not updating mode or action

kubik369, 2021-12-04:
### The problem

Hi,

I have a climate - thermostat setup. It appears to work correctly: it regulates the temperature around the set point, you can also set the set point from HA. However, ESPHome does not seem to report/update the state of the thermostat, it is always shown as being off in HA. I have attached a part of the logs, they are from a period when the heating was on, but as you can see, both Mode and Action are being logged as off. I am not sure whether this is a misconfiguration on my part, but it seems weird to me that it would show the mode and action being off if the heating is working. The current and target temperature is being shown correctly in HA. Thank you for your help :)

### Which version of ESPHome has the issue?

2021.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

sonoff_dualr2_001

### Component causing the issue

climate

### Example YAML snippet

```yaml
climate:
  - platform: thermostat
    name: "Sliepky klíma"
    id: sliepky_klima
    visual:
      min_temperature: 10 °C
      max_temperature: 30 °C
      temperature_step: 0.1 °C
    sensor: sliepky_temperature
    default_target_temperature_low: 15 °C
    min_heating_off_time: 300s
    min_heating_run_time: 300s
    min_idle_time: 30s
    heat_action:
      - switch.turn_on: relay2
    idle_action:
      - switch.turn_off: relay2
```

### Anything in the logs that might be useful for us?

```txt
[15:11:11][D][climate:381]: 'Sliepky klíma' - Sending state:
[15:11:11][D][climate:384]:   Mode: OFF
[15:11:11][D][climate:386]:   Action: OFF
[15:11:11][D][climate:404]:   Current Temperature: 17.50°C
[15:11:11][D][climate:410]:   Target Temperature: 17.50°C

[15:12:11][D][dallas.sensor:136]: 'Teplota u sliepok': Got Temperature=17.7°C

[15:12:11][D][climate:381]: 'Sliepky klíma' - Sending state:
[15:12:11][D][climate:384]:   Mode: OFF
[15:12:11][D][climate:386]:   Action: OFF
[15:12:11][D][climate:404]:   Current Temperature: 17.69°C
[15:12:11][D][climate:410]:   Target Temperature: 17.50°C
```

### Additional information

I am running the Home assistant OS VM, there wasn't such an option in the drop-down. It probably doesn't change anything, just wanted to include it for completion

kubik369, 2021-12-04:
I have encountered another user on Discord (myreal#7710) who seems to be having the same problem. Here is his message if he doesn't get to post it here:

Screenshot version:
[image]
Text version:
```
Hey, 
I try to set up simple thermostat for heating only. Have this config:  https://pastebin.com/QjGX6WL5
Temperature is way below set default_target_temperature_low, however, heat_action is not initiated. Get this in logs: https://pastebin.com/RiHZUqsP

Seems that I miss something obvious 🙂 can you help?
```

kubik369, 2021-12-05:
Apparently the other user has fixed his problem by updating to `2021.11.4`. I have tried doing the same, but no change for me sadly :/ Only additional piece of information that I have noticed is that I get a log of the switch for the heater bulb getting turned off and on correctly. The device still keeps reporting both `MODE` and `ACTION` as `OFF`, even though the actions specified in `heat_action` and `idle_action` get executed, as the previously mentioned logs also indicate.

kbx81, 2021-12-08:
This is...interesting. I'm not seeing this behavior on any of my thermostats and I've just updated them to current `dev`. I'd be interested to know if this issue/behavior persists into 2021.12 which is due to be released late this week.

kbx81, 2022-01-03:
Is this still an issue on the latest releases of HA and ESPHome?

bcotton, 2022-01-14:
Just getting started with the thermostat components. I'm seeing same issue:

```
[16:01:52][D][climate:381]: 'Basement Thermostat' - Sending state:
[16:01:52][D][climate:384]:   Mode: OFF
[16:01:52][D][climate:386]:   Action: OFF
[16:01:52][D][climate:404]:   Current Temperature: 44.90°C
[16:01:52][D][climate:410]:   Target Temperature: 80.28°C

```

I'm on HA  core-2021.12.9

lal12, 2022-10-02:
Maybe this is obvious, but I had the same effect at first until I activated it by pressing the "flame" on the dashboard controls. So additional to temperature there seems to be an off/on mode.

kbx81, 2022-10-03:
Yeah I don't know what this is -- it's not something I've seen before and I haven't been able to reproduce it here.

Is this behavior that occurs on a newly-installed device? If manually selecting the mode in HA resolves the behavior and it doesn't happen again, I suspect this may be the case -- maybe it is assuming some invalid state the first time it's flashed.

I'd be curious to know exactly what versions of both HA and ESPHome you're using when you saw this behavior. I'm also curious if the latest updates in current `dev` (and available in the next release) will fix it -- some minor changes have been made to the start-up behavior to accommodate restoring state _and_ restoring from a (default) preset.

There is an "off" (climate) mode but not an "on" mode.

woferry, 2024-01-14:
I was just getting started with ESPHome's Climate/Thermostat, and I'm seeing the same thing (2023.12.5).  Simple, heating-only setup (one switch), and I'm stuck in mode: OFF so the heat_action isn't firing no matter how high I set the min temp.  I had already toggled presets in HA and the mode remained OFF, though I didn't have the optional mode setting in my presets (just name and default_target_temperature_low).  Once I added mode to one of the presets and selected that preset in HA suddenly the thermostat card 'arc' switched from grey to orange and the mode changed to HEAT.