esphome/issues #2781: Update failed, now my ESPHome installation is corrupt. How can I fix this without loosing the nodes?

VdR1960, 2021-12-04:
### The problem

I get a compiler error when using the pulse_meter integration. With help from here and the HA forum, it is clear that that is caused by an **incomplete update of ESPHome**, causing an #include of on old file version (epshal.h) instead of the new version (hal.h).

[image]

I do not find a way to repair the ESPHome install. I'm afraid that if I uninstall and re-install I will loose the connection the nodes and some are really hard to get to re-build.

[image]

[image]

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Anything in the logs that might be useful for us?

```txt
The log report ESPHome version 2021.11.4.

But the version.h file shows 2021.10.3
```

editicalu, 2021-12-04:
The ESPHome integration can be used regardless of the esphome add-on (you can even use it without it). Normally you wouldn't lose the devices as HomeAssistant devices.

You can simulate the behavior by stopping the ESPhome add-on and checking whether existing devices are still connected to the Home Assistant instance.

I've tried this locally (stopping, not deleting) and it seems to work.

If you still don't trust this, you can always use an external esphome instance (for example, on your personal computer) and use that to update the esp devices in your network.

mmakaay, 2021-12-04:
All you need for ESPHome are the yaml config files. If you backup those, and dump them back in a fresh install, you'll have everything back. As @editicalu said: the ESPHome dashboard is not needed for the connection to Home Assistant. It is only used for building and uploading firmware to the devices.

VdR1960, 2021-12-06:
I have un-installed, cleaned up and re-installed. The problem is solved.

Something had seriously gone wrong with the ESPHome updates, there where a lot of extra files in the esphome directory. Maybe that was somehow unique to my setup.

Thank you all for your help.

JeeCee1, 2022-01-17:
I had the same problem, ESPHome 2021.12.3 on Supervised HA.
Resolved as well be uninstalling ESP Home and needed to manually remove the .esphome folder and all project folders.

mmakaay, 2022-01-17:
Thanks for the extra info @JeeCee1 
Closing this issue now, since the issue for the original author was resolved.

closed 2022-01-17 by mmakaay