esphome/issues #2782: [SPS30] "pm_size" not working?

lostcontrol, 2021-12-04:
### The problem

Looking at the [code](https://github.com/esphome/esphome/blob/dev/esphome/components/sps30/sensor.py#L94), it seems that `pm_size` should be supported. However, it does not seem to be taken into account in the config:

> [15:54:22][C][sps30:067]: sps30:
[15:54:22][C][sps30:068]:   Address: 0x69
[15:54:22][W][sps30:084]: Unable to request sensor firmware version
[15:54:22][C][sps30:094]:   Update Interval: 60.0s
[15:54:22][C][sps30:095]:   Serial Number: ''
[15:54:22][C][sps30:096]:   PM1.0 'PM <1µm Weight concentration'
[15:54:22][C][sps30:096]:     Device Class: 'pm1'
[15:54:22][C][sps30:096]:     State Class: 'measurement'
[15:54:22][C][sps30:096]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:096]:     Accuracy Decimals: 2
[15:54:22][C][sps30:096]:     Icon: 'mdi:chemical-weapon'
[15:54:22][C][sps30:097]:   PM2.5 'PM <2.5µm Weight concentration'
[15:54:22][C][sps30:097]:     Device Class: 'pm25'
[15:54:22][C][sps30:097]:     State Class: 'measurement'
[15:54:22][C][sps30:097]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:097]:     Accuracy Decimals: 2
[15:54:22][C][sps30:097]:     Icon: 'mdi:chemical-weapon'
[15:54:22][C][sps30:098]:   PM4 'PM <4µm Weight concentration'
[15:54:22][C][sps30:098]:     State Class: 'measurement'
[15:54:22][C][sps30:098]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:098]:     Accuracy Decimals: 2
[15:54:22][C][sps30:098]:     Icon: 'mdi:chemical-weapon'
[15:54:22][C][sps30:099]:   PM10 'PM <10µm Weight concentration'
[15:54:22][C][sps30:099]:     Device Class: 'pm10'
[15:54:22][C][sps30:099]:     State Class: 'measurement'
[15:54:22][C][sps30:099]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:099]:     Accuracy Decimals: 2
[15:54:22][C][sps30:099]:     Icon: 'mdi:chemical-weapon'

It is also not present in the [documentation](https://esphome.io/components/sensor/sps30.html).

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Component causing the issue

sps30

### Example YAML snippet

```yaml
sensor:
  - platform: scd30
    co2:
      name: "CO2"
      on_value:
        then:
          - http_request.send:
              method: PUT
              url: "${openhab_rest_url}/AirQuality_CO2/state"
              headers:
                Content-Type: text/plain
                Accept: application/json
              body: !lambda "return esphome::to_string(x);"
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    automatic_self_calibration: false
    altitude_compensation: 750m
    address: 0x61
    update_interval: 60s
  - platform: sps30
    pm_1_0:
      name: "PM <1µm Weight concentration"
      id: "PM_1_0"
    pm_2_5:
      name: "PM <2.5µm Weight concentration"
      id: "PM_2_5"
    pm_4_0:
      name: "PM <4µm Weight concentration"
      id: "PM_4_0"
    pm_10_0:
      name: "PM <10µm Weight concentration"
      id: "PM_10_0"
    pmc_0_5:
      name: "PM <0.5µm Number concentration"
      id: "PMC_0_5"
    pmc_1_0:
      name: "PM <1µm Number concentration"
      id: "PMC_1_0"
    pmc_2_5:
      name: "PM <2.5µm Number concentration"
      id: "PMC_2_5"
    pmc_4_0:
      name: "PM <4µm Number concentration"
      id: "PMC_4_0"
    pmc_10_0:
      name: "PM <10µm Number concentration"
      id: "PMC_10_0"
    pm_size:
      name: "PM average size"
      id: "PM_SIZE"
    address: 0x69
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[15:54:22][C][sps30:067]: sps30:
[15:54:22][C][sps30:068]:   Address: 0x69
[15:54:22][W][sps30:084]: Unable to request sensor firmware version
[15:54:22][C][sps30:094]:   Update Interval: 60.0s
[15:54:22][C][sps30:095]:   Serial Number: ''
[15:54:22][C][sps30:096]:   PM1.0 'PM <1µm Weight concentration'
[15:54:22][C][sps30:096]:     Device Class: 'pm1'
[15:54:22][C][sps30:096]:     State Class: 'measurement'
[15:54:22][C][sps30:096]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:096]:     Accuracy Decimals: 2
[15:54:22][C][sps30:096]:     Icon: 'mdi:chemical-weapon'
[15:54:22][C][sps30:097]:   PM2.5 'PM <2.5µm Weight concentration'
[15:54:22][C][sps30:097]:     Device Class: 'pm25'
[15:54:22][C][sps30:097]:     State Class: 'measurement'
[15:54:22][C][sps30:097]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:097]:     Accuracy Decimals: 2
[15:54:22][C][sps30:097]:     Icon: 'mdi:chemical-weapon'
[15:54:22][C][sps30:098]:   PM4 'PM <4µm Weight concentration'
[15:54:22][C][sps30:098]:     State Class: 'measurement'
[15:54:22][C][sps30:098]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:098]:     Accuracy Decimals: 2
[15:54:22][C][sps30:098]:     Icon: 'mdi:chemical-weapon'
[15:54:22][C][sps30:099]:   PM10 'PM <10µm Weight concentration'
[15:54:22][C][sps30:099]:     Device Class: 'pm10'
[15:54:22][C][sps30:099]:     State Class: 'measurement'
[15:54:22][C][sps30:099]:     Unit of Measurement: 'µg/m³'
[15:54:22][C][sps30:099]:     Accuracy Decimals: 2
[15:54:22][C][sps30:099]:     Icon: 'mdi:chemical-weapon'
```

### Additional information

I don't have the sensor yet, I'm just preparing my setup. Also, I discovered ESPHome today. Amazing project. Planned on implementing things myself but this project is just perfect for my needs.

martgras, 2021-12-05:
`pm_size`  is missing  the documentation. submitted esphome/esphome-docs#1699 to update the documentation
It's also not mentioned in dump_config which shows the configured sensors. Submitted https://github.com/esphome/esphome/pull/2866 to correct it
Despite these glitches pm_size is already working  in the released version with a SPS30 connected.

lostcontrol, 2021-12-05:
@martgras awesome, thank you :smile: I don't know when you close tickets, feel free to close if fine for you.

closed 2021-12-06 by jesserockz