esphome/issues #2791: Ethernet and i2c bus problem on boot

stefkamarton, 2021-12-06:
### The problem

Hy,
I have an ESP32 (nodemcu-32s) and I connected LAN8720 component to my ESP, I don't use esp's WIFI, and I use i2c connection.
When I reset my ESP then it didn't connect to the internet through LAN8720. But If I've disconnect sda pin for a second, it begins work and get ip from dhcp and connect to the internet. Can you solve my problem?

```
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1

i2c:
  - id: bus_a
    sda: GPIO14
    scl: GPIO12
    scan: true

tca9548a:
  - address: 0x72
    id: multiplex0
    i2c_id: bus_a
    channels:
      - bus_id: tca0
        channel: 0
```

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

ethernet, LAN8720, i2c

### Example YAML snippet

```yaml
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1

i2c:
  - id: bus_a
    sda: GPIO14
    scl: GPIO12
    scan: true

tca9548a:
  - address: 0x72
    id: multiplex0
    i2c_id: bus_a
    channels:
      - bus_id: tca0
        channel: 0
```

oxan, 2021-12-06:
Is there anything in the logs?

stefkamarton, 2021-12-08:
> Is there anything in the logs?

There aren't any logs because I can't connect to ESP32 until I disconnect sda pin for a second. 
If I disconnect it and connect it then after that works well and connect to the internet immediately!

randybb, 2021-12-08:
For ESP32 GPIO12 must be LOW during boot. Remove external i2c pull-ups and maybe it will help. Would be better to stick with the default I2C pins - SDA GPIO21, SDL GPIO22

_Startup failure if in PULL-UP mode
Pin 12 is used as the boot pin to select the output voltage of an internal regulator that powers the flash chip (VDD_SDIO). This pin has an internal pull-down resistor. If it is not connected, it will display a low level when resetting (selection of default 3.3V operation). Make sure it is not pulled up by a device during boot or the ESP32 might not boot._

closed 2021-12-12 by jesserockz