esphome/issues #2795: ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

Tokn59, 2021-12-08:
### The problem

Since shortly I get this error when trying to flash a Wemos D1 Mini Pro. Never had this kind of error before.

ESPHome version 2021.11.4 running on latest builds HASSOS

Board definition :

esphome:
name: co2box-ian-1
platform: ESP8266
board: d1_mini_pro

The only way out is to flash it again manually.

Any clue as to what causes this ?

Thank You !!!!!

### Which version of ESPHome has the issue?

ESPHome  2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

HASSOS core-2021.11.5 - supervisor-2021.10.8

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini Pro

### Component causing the issue

OTA Update

### Example YAML snippet

```yaml
esphome:
  name: co2box-ian-1
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: "IOT"
  password: "xxxxxxxxxxxx"
# Optional manual IP
#  manual_ip:
#    static_ip: 192.168.1.253
#    gateway: 192.168.1.1
#    subnet: 255.255.255.0  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "co2box Ian 1 Fallback Hotspot"
    password: "xxxxxxxxxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxxxxxx"

ota:
  password: "xxxxxxxxxxx"

web_server:
  port: 80
  
i2c:
  sda: D3
  scl: D4
  scan: True
  
uart:
  tx_pin: GPIO01
  rx_pin: GPIO03 
  baud_rate: 9600

#D1: Green Wire / mhz19
#D2: Blue Wire / mhz19

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2811
    pin: D6
    num_leds: 1
    name: "CO2 Status Light"
  
sensor:
  - platform: mhz19
    id: my_mhz19_id
    co2:
      name: "co2box CO2 Value"
      id: co2value
    temperature:
      name: "co2box Temperature"
      id: co2temp
      filters:
       - offset: -5.5
    automatic_baseline_calibration: false  
    update_interval: 60s
  - platform: adc
    pin: VCC
    name: "co2box Battery Voltage"    
  - platform: wifi_signal
    name: "co2box WiFi Signal Sensor"
    update_interval: 60s
    id: wifisignal

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
      id: espip
    ssid:
      name: ESP Connected SSID
      id: espssid    
# Define Switches

switch:
  - platform: restart
    name: "co2box Ian 1 Restart"

  - platform: template
    name: "CO2 Calibrate"
    optimistic: true
    on_turn_on:
      mhz19.abc_enable: my_mhz19_id
    on_turn_off:
      mhz19.abc_disable: my_mhz19_id
time:
  - platform: sntp
    id: time1
    
font:
  - file: "arial.ttf"
    id: slk
    size: 12
  - file: "arial.ttf"
    id: bebas
    size: 12
  - file: "arial.ttf"
    id: bebas20
    size: 32   
  - file: "roboto regular.ttf"
    id: roboto
    size: 24
  - file: "Orbitron Bold.ttf"
    id: orbitronb
    size: 36
    
display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    reset_pin: D0
    address: 0x3C
    rotation: 180
    id: my_display
    pages:
      - id: page1
        lambda: |-
          it.rectangle(0, 0, 128, 64);
          if (id(co2value).has_state()) { 
          it.printf(64, 10, id(orbitronb), TextAlign::TOP_CENTER, "%.0f", id(co2value).state);
          }
          it.print(45, 50, id(bebas), "PPM");
      - id: page2
        lambda: |-
           it.rectangle(0, 0, 128, 64);
           if (id(co2temp).has_state()) {           
           it.printf(64, 10, id(orbitronb), TextAlign::TOP_CENTER, "%.0f", id(co2temp).state);
           }
           it.print(55, 50, id(bebas), "°C");
#      - id: page3
#        lambda: |-
#          it.strftime(0, 10,id(slk),TextAlign::BASELINE_LEFT,"%d-%m-%Y",id(time1).now());
#          it.strftime(0, 48, id(bebas20), TextAlign::BASELINE_LEFT, "%H:%M", id(time1).now());
#      - id: page4
#        lambda: |-
#            it.printf(0,15,id(slk),TextAlign::BASELINE_LEFT,"Lux: %s",id(espip).state.c_str());
#            if (id(wifisignal).has_state()) {
#              it.printf(0,63,id(bebas),TextAlign::BASELINE_LEFT, "%.0f dB",id(wifisignal).state);
#            }

interval:
  - interval: 4s
    then:
      - display.page.show_next: my_display
      - component.update: my_display
```

### Anything in the logs that might be useful for us?

```txt
See error in issue
```

### Additional information

Never had this issue before. After error I need to upload compiled code via USB again. OTA failes with this error.

Tokn59, 2021-12-17:
Any news on this please ?

closed 2022-05-01 by github-actions[bot]

Nibb31, 2022-07-10:
Same problem here. The exact same script worked fine on previous versions and now no longer uploads.

danielscheidler, 2022-07-28:
same problem here. any solutions?