esphome/issues #2796: on_boot and custom component void setup() does not truly run on power up exclusively.

SparkyDan555, 2021-12-08:
### The problem

I am using `on_boot` and `void setup()` of a custom output component to start a stepper motor moving towards a homing switch. My issue is that on_boot and setup() does not mean on startup. Therefore whenever the ESP reboots due to bad WiFi etc, the motor starts to move unexpectedly. It does not allow the user to run a snippet of code once **and only once** on power up of the device. In my opinion on_boot should not run after a reboot especially when said reboot was not initiated by the user.

### Which version of ESPHome has the issue?

ESPHome version 2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

on_boot

### Example YAML snippet

```yaml
esphome:
  name: ${devicename}
  comment: ${device_description}
  platform: ESP8266
  board: nodemcuv2
  platformio_options:
    board_build.f_cpu: 160000000L
  includes:
    - uart_read_line_sensor.h
    - projectorlift.h
  libraries:
    - "waspinator/AccelStepper"
  on_boot:
    priority: -100
    # ...
    then:
      - remote_transmitter.transmit_raw:
          code: [-690, 208, -691, 208, -693, 209, -686, 660, -242, 659, -244, 206, -692, 658, -244, 658, -243, 658, -245, 204, -694, 206, -692, 207, -693, 206, -692, 657, -244, 658, -244, 206, -692, 207, -692, 206, -693, 206, -693, 657, -244, 206, -693, 206, -6969, 657, -243, 207, -691, 208, -690, 208, -691, 208, -691, 208, -689, 660, -243, 659, -243, 207, -691, 659, -243, 658, -244, 658, -244, 206, -693, 206, -693, 206, -693, 205, -692, 658, -245, 657, -244, 206, -693, 207, -691, 206, -693, 206, -693, 657, -244, 206, -693, 205, -6968, 659, -242, 208, -692, 207, -691, 208, -691, 207, -692, 207, -689, 660, -243, 659, -243, 207, -692, 658, -244, 658, -243, 658, -244, 206, -694, 205, -692, 207, -692, 206, -693, 657, -244, 658, -243, 207, -692, 207, -692, 206, -694, 205, -692, 658, -244, 206, -693, 206, -6968, 659, -242, 208, -690, 208, -691, 208, -690, 208, -690, 209, -690, 660, -242, 659, -242, 208, -691, 659, -243, 659, -244, 657, -244, 206, -693, 206, -692, 207, -692, 207, -692, 658, -243, 658, -244, 206, -693, 206, -692, 207, -695, 203, -693, 658, -243, 206, -693, 206, -6967, 660, -242, 208, -691, 208, -690, 208, -690, 209, -694, 204, -690, 660, -242, 659, -243, 208, -691, 659, -244, 661, -242, 656, -244, 206, -692, 207, -691, 208, -691, 207, -691, 659, -245, 656, -244, 206, -693, 206, -692, 207, -693, 206, -693, 656, -245, 206, -693, 206, -6967, 660, -242, 208, -690, 209, -690, 208, -690, 209, -690, 208, -690, 664, -242, 655, -243, 208, -691, 659, -243, 658, -244, 657, -245, 206, -692, 207, -692, 206, -692, 207, -692, 658, -244, 657, -244, 207, -692, 207, -692, 206, -692, 207, -692, 658, -244, 206, -693, 206, -6968, 659, -242, 208, -690, 209, -689, 209, -690, 209, -689, 210, -690, 659, -242, 660, -242, 208, -691, 662, -242, 657, -245, 656, -244, 207, -690, 208, -691, 208, -691, 208, -692, 657, -244, 658, -244, 206, -692, 207, -692, 207, -691, 207, -692, 658, -244, 207, -692, 206, -6968, 660, -242, 208, -689, 209, -690, 209, -690, 209, -690, 208, -690, 660, -242, 659, -243, 208, -690, 660, -243, 658, -244, 658, -244, 206, -692, 207, -692, 207, -691, 207, -691, 659, -244, 658, -243, 207, -692, 207, -692, 206, -692, 207, -691, 659, -244, 206, -693, 206, -6968, 659, -242, 209, -689, 209, -690, 209, -689, 213, -686, 209, -690, 660, -242, 659, -243, 208, -690, 660, -245, 656, -243, 659, -254, 196, -691, 208, -691, 208, -690, 208, -691, 659, -243, 659, -243, 207, -691, 208, -691, 208, -692, 206, -692, 658, -244, 206, -693, 206, -6968, 660, -242, 208, -690, 209, -690, 208, -690, 209, -690, 209, -690, 659, -242, 660, -242, 208, -691, 659, -243, 659, -243, 658, -243, 207, -692, 207, -692, 207, -691, 208, -691, 658, -244, 658, -247, 204, -691, 207, -692, 207, -691, 212, -687, 659, -244, 206, -693, 206, -6967, 660, -243, 208, -690, 208, -690, 209, -690, 209, -689, 209, -691, 659, -244, 658, -241, 209, -690, 660, -243, 659, -243, 658, -244, 207, -692, 206, -692, 207, -691, 208, -691, 659, -243, 658, -244, 207, -692, 207, -692, 207, -691, 207, -691, 659, -244, 206, -693, 206, -6969, 659, -242, 208, -690, 209, -690, 209, -689, 209, -691, 208, -689, 662, -243, 657, -242, 208, -691, 659, -243, 659, -244, 658, -244, 206, -691, 208, -691, 208, -691, 207, -692, 658, -244, 658, -244, 206, -692, 207, -692, 207, -692, 207, -691, 658, -245, 206, -693, 206, -6968, 659, -242, 209, -690, 209, -691, 207, -691, 208, -691, 208, -689, 660, -242, 660, -244, 206, -690, 660, -244, 658, -243, 658, -244, 207, -691, 208, -692, 206, -691, 208, -692, 658, -244, 658, -243, 207, -692, 207, -691, 208, -691, 207, -692, 660, -241, 208, -692, 206, -6968, 660, -243, 207, -691, 208, -690, 209, -689, 210, -689, 209, -689, 661, -242, 660, -242, 208, -690, 660, -243, 659, -243, 658, -244, 207, -691, 207, -692, 207, -692, 207, -692, 658, -243, 659, -244, 206, -692, 207, -692, 207, -691, 207, -692, 658, -245, 205, -692, 207, -6972, 656, -243, 208, -689, 209, -690, 209, -690, 209, -690, 208, -690, 660, -242, 660, -242, 208, -691, 659, -244, 658, -242, 660, -243, 207, -695, 204, -691, 208, -691, 208, -690, 660, -242, 661, -241, 209, -690, 208, -691, 208, -690, 208, -692, 658, -244, 206, -693, 206, -6968, 660, -242, 208, -690, 209, -690, 209, -689, 209, -690, 209, -690, 660, -242, 660, -241, 209, -690, 660, -243, 659, -242, 659, -244, 207, -691, 208, -691, 207, -692, 207, -691, 659, -244, 658, -244, 206, -692, 207, -692, 207, -691, 208, -691, 659, -243, 207, -691, 208, -6969, 658, -244, 207, -690, 208, -691, 208, -690, 209, -690, 209, -690, 659, -243, 659, -243, 208, -691, 658, -244, 658, -244, 658, -244, 206, -693, 206, -694, 205, -692, 207, -693, 657, -244, 657, -244, 207, -692, 207, -692, 207, -691, 207, -692, 658, -245, 205, -692, 207, -6968, 661, -244, 206, -693, 206, -690, 208, -691, 208, -691, 208, -689, 661, -242, 659, -243, 208, -691, 659, -243, 658, -243, 659, -244, 206, -695, 205, -691, 207, -692, 207, -691, 659, -244, 658, -243, 207, -692, 207, -691, 208, -691, 208, -691, 659, -243, 207, -692, 207, -6968, 660, -242, 208, -690, 209, -691, 208, -690, 209, -689, 209, -690, 660, -242, 660, -243, 207, -690, 660, -243, 659, -243, 662, -242, 205, -692, 207, -691, 208, -690, 209, -691, 658, -244, 658, -244, 207, -691, 207, -692, 207, -692, 207, -692, 658, -245, 205, -692, 207, -6968, 661, -241, 209, -690, 209, -690, 208, -690, 209, -690, 209, -690, 661, -241, 660, -243, 207, -690, 660, -243, 661, -243, 656, -244, 207, -691, 208, -691, 208, -691, 208, -690, 660, -243, 658, -244, 207, -693, 206, -692, 207, -691, 208, -693, 657, -243, 207, -692, 207, -6969, 660, -242, 208]
          repeat:
            times: 2
            wait_time: 50ms
```

### Additional information

[Archive.zip](https://github.com/esphome/issues/files/7680921/Archive.zip)

oxan, 2021-12-09:
A reboot is also a boot, so it doesn't make sense to may to not run `on_boot` then (besides that it'll break the setup of lots of users). Aside from being unexpected, not running `setup()` on reboots is technically impossible as well, since state gets reset on reboots and most components depend on their state being initialized correctly in `setup()`.

You can make your automations by querying the reset reason through `system_get_rst_info`.

SparkyDan555, 2021-12-09:
My issue is with the motor moving at random times due to random reboots from WiFi etc after it is already setup. I think it would be beneficial to have a configuration option to ignore soft resets and only allow on_boot to run on a true hardware boot.

closed 2022-04-23 by github-actions[bot]