esphome/issues #2798: climate_action_to_string() can't be used in a text sensor

MrHaroldA, 2021-12-10:
### The problem

When using the string representation of the current action of a climate component, you'll get an error compiling it:

```
text_sensor:
  - platform: template
    name: "${dev_name} HVAC state"
    update_interval: 10s
    lambda: return climate_action_to_string(id(${dev_machine_name}_climate).action);
```

Result:
```
Compiling /data/cellar/.pioenvs/cellar/src/main.cpp.o
/config/esphome/cellar.yaml: In lambda function:
/config/esphome/cellar.yaml:70:61: error: could not convert 'esphome::climate::climate_action_to_string(((int)cellar_climate->esphome::thermostat::ThermostatClimate::<anonymous>.esphome::climate::Climate::action))' from 'const esphome::LogString*' to 'esphome::optional<std::basic_string<char> >'
     lambda: return climate_action_to_string(id(${dev_machine_name}_climate).action);
                                                             ^
/config/esphome/cellar.yaml:71:3: warning: control reaches end of non-void function [-Wreturn-type]
     
   ^
*** [/data/cellar/.pioenvs/cellar/src/main.cpp.o] Error 1
```

This used to work up till a few versions ago.

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

Sonoff TH10

### Component causing the issue

climate

### Example YAML snippet

```yaml
substitutions:
  dev_name: "Cellar"
  dev_machine_name: cellar

dallas:
  - pin: GPIO14

sensor:
  - platform: dallas
    name: "${dev_name} inside"
    index: 0
    id: inside

switch:
  - platform: gpio
    pin: GPIO12
    id: relay
    restore_mode: RESTORE_DEFAULT_OFF
    internal: yes

text_sensor:
  - platform: template
    name: "${dev_name} HVAC state"
    update_interval: 10s
    lambda: return climate_action_to_string(id(${dev_machine_name}_climate).action);
    
climate:
  - platform: thermostat
    id: "${dev_machine_name}_climate"
    name: "${dev_name}"
    min_idle_time: 60s
    min_cooling_off_time: 60s
    min_cooling_run_time: 30s

    default_target_temperature_high: 12 °C

    visual:
      min_temperature: 8 °C
      max_temperature: 16 °C
      temperature_step: 0.5 °C
    sensor: inside

    idle_action:
      - switch.turn_off: relay

    cool_action:
      - switch.turn_on: relay
```

### Anything in the logs that might be useful for us?

```txt
Compiling /data/cellar/.pioenvs/cellar/src/main.cpp.o
/config/esphome/cellar.yaml: In lambda function:
/config/esphome/cellar.yaml:70:61: error: could not convert 'esphome::climate::climate_action_to_string(((int)cellar_climate->esphome::thermostat::ThermostatClimate::<anonymous>.esphome::climate::Climate::action))' from 'const esphome::LogString*' to 'esphome::optional<std::basic_string<char> >'
     lambda: return climate_action_to_string(id(${dev_machine_name}_climate).action);
                                                             ^
/config/esphome/cellar.yaml:71:3: warning: control reaches end of non-void function [-Wreturn-type]
     
   ^
*** [/data/cellar/.pioenvs/cellar/src/main.cpp.o] Error 1
```

MrHaroldA, 2021-12-10:
The magic combination to get it to compile, as suggested by ssieb in the ESPHome Discord server:

```return std::string(LOG_STR_ARG(climate_action_to_string(id(climate_id).action)));```

oxan, 2021-12-13:
That function is meant to only be a logging helper. Why do you want to publish it? The action should already be available as an attribute on the `climate` entity in HA.

MrHaroldA, 2021-12-13:
> That function is meant to only be a logging helper. Why do you want to publish it? The action should already be available as an attribute on the `climate` entity in HA.

I would like the HVAC state (Idle, cool, or heat) to be visible like this:

[image]

The ESPHome Climate component does not expose this to Home Assistant, so I had to add a text sensor for that which worked nice up till a few versions ago.

oxan, 2021-12-14:
> The ESPHome Climate component does not expose this to Home Assistant

It should, as an attribute on the climate entity.

MrHaroldA, 2021-12-14:
The climate entity only has the temperature sensor (cellar_inside), the climate entity itself (which shows target and current temp), and my _now unavailable_ text sensor.

[image]

In the Entities configuration, only a device tracker entity is mentioned as extra, and no (apparent) way to display the current action in a history graph.

My Nest climate does have the HVAC state, which I tried to copy to my ESPHome fridge controller.

oxan, 2021-12-14:
> The climate entity only has the temperature sensor (cellar_inside), the climate entity itself (which shows target and current temp), and my _now unavailable_ text sensor.

Those are the entities of the climate device, I'm refering to an [attribute](https://www.home-assistant.io/docs/configuration/templating/#attributes) on the climate entity (`climate.cellar`). You can convert this into an entity using a template sensor in HA by itself.

> `return std::string(LOG_STR_ARG(climate_action_to_string(id(climate_id).action)));`

This crashes because the message strings are stored in flash, and require aligned reads (using e.g. `strcpy_P` or the Arduino `String`) class.

MrHaroldA, 2021-12-14:
> You can convert this into an entity using a template sensor in HA by itself.

I was hoping for a simple text sensor in ESPHome, which requires no further HA configuration.

I'll revert my one-liner to the switch case I was using a couple of months ago.

```text_sensor:
  - platform: template
    name: "${dev_name} HVAC state"
    update_interval: 10s
    lambda: |-
      switch (id(climate_id).action) {
        case CLIMATE_ACTION_OFF:
          return (std::string) "OFF";
        case CLIMATE_ACTION_COOLING:
          return (std::string) "COOLING";
        case CLIMATE_ACTION_HEATING:
         return (std::string) "HEATING";
        case CLIMATE_ACTION_IDLE:
          return (std::string) "IDLE";
        default:
          return (std::string) "UNKNOWN";
      }```

closed 2022-04-28 by github-actions[bot]