esphome/issues #2799: Thermostat with fan only mode giving temperature high keyerror

Th3M1k3y, 2021-12-10:
### The problem

When adding `fan_only_action` along with the needed `min_fanning_off_time` and `min_fanning_run_time` I am getting `KeyError: 'default_target_temperature_high'` when trying to compile.

Removing those makes it compile successfully.

### Which version of ESPHome has the issue?

2021.10.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

Wemos D1 ESP32

### Component causing the issue

thermostat

### Example YAML snippet

```yaml
substitutions:
  # General variables
  device_name: mobile-heater
  friendly_name: 'Mobile heater'
  device_description: 'Create warmth'
  

esphome:
  name: '${device_name}'
  comment: '${device_description}'
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret iot_ssid
  password: !secret iot_pass

# Enable logging
logger:
  level: WARN

# Enable Home Assistant API
api:
  password: !secret esphome_api_password

ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

web_server:
  port: 80
  auth:
    username: !secret esphome_web_username
    password: !secret esphome_web_password

# Sensors for ESP version and WIFI information
text_sensor:
  - platform: version
    hide_timestamp: true
    name: "${friendly_name} - ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} - IP Address"
      icon: mdi:wifi
    ssid:
      name: "${friendly_name} - Connected SSID"
      icon: mdi:wifi-strength-2
      
      
switch:
  - platform: gpio
    name: "${friendly_name} - Fan"
    pin: GPIO12
    id: output_fan
    
  - platform: gpio
    name: "${friendly_name} - Heater 1"
    pin: GPIO12
    id: output_heat1
    
  - platform: gpio
    name: "${friendly_name} - Heater 2"
    pin: GPIO12
    id: output_heat2
    
  - platform: gpio
    name: "${friendly_name} - Heater 3"
    pin: GPIO12
    id: output_heat3

dallas:
  - pin: GPIO2
  
sensor:
  - platform: dallas
    address: 0xE8011438D052AA28
    id: sensor_temp
    name: "${friendly_name} - Temperature"
    
climate:
  - platform: thermostat
    id: mobile_heater
    name: "Thermostat Climate Controller"
    default_mode: "off"
    sensor: sensor_temp
    default_target_temperature_low: 20 °C
    heat_deadband: 0.5 °C
    heat_overrun: 0.5 °C
    
    visual:
      min_temperature: 15 °C
      max_temperature: 25 °C
      temperature_step: 0.5 °C
    
    min_fanning_off_time: 30s
    min_fanning_run_time: 30s
    
    min_heating_off_time: 60s
    min_heating_run_time: 60s
    
    min_idle_time: 30s
    
    max_heating_run_time: 30s # supplemental_heating_action will be called repeatedly at an interval defined by this parameter
    supplemental_heating_delta: 1 °C # when the temperature difference between the lower set point and the current temperature exceeds this value, supplemental_heating_action will be called immediately
    
    heat_action:
      - switch.turn_on: output_fan
      - switch.turn_on: output_heat1
          
    supplemental_heating_action:
      lambda: !lambda |-
          if (id(mobile_heater).current_temperature < (id(mobile_heater).target_temperature-1))
          {
            id(output_heat2).turn_on();
          }
          else
          {
            id(output_heat2).turn_off();
          }
          
          if (id(mobile_heater).current_temperature < (id(mobile_heater).target_temperature-3))
          {
            id(output_heat3).turn_on();
          }
          else
          {
            id(output_heat3).turn_off();
          }
      
    idle_action:
      - switch.turn_off: output_fan
      - switch.turn_off: output_heat1
      - switch.turn_off: output_heat2
      - switch.turn_off: output_heat3
      - logger.log: "Idle"
      
    fan_only_action: 
      - switch.turn_on: output_fan
```

### Anything in the logs that might be useful for us?

```txt
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 817, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 804, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 324, in command_compile
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 156, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 168, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 148, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/thermostat/climate.py", line 455, in to_code
    config[CONF_DEFAULT_TARGET_TEMPERATURE_HIGH],
KeyError: 'default_target_temperature_high'
```

ssieb, 2021-12-11:
The problem appears to be at https://github.com/esphome/esphome/blob/dev/esphome/components/thermostat/climate.py#L433  It shouldn't be two points unless cooling is enabled for the fan.

kbx81, 2021-12-11:
Yuuuuppp...confirmed that's the problem. PR incoming...

closed 2021-12-11 by jesserockz