esphome/issues #2800: esp32 nodemcu-32s "Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)"

Levenale, 2021-12-10:
### The problem

I'm a new user to esphome, doing a fresh install and trying to get a basic environment on a Nodemcu-32s board so I can continue building a sensor.

I setup the basic config (see below), built and downloaded with esphome dashboard, and it fails.

_Specifics_
1. Using esphome dashboard, I'm connected to the nodemcu32s via a USB cable for downloading.
2. I created a basic config with my wifi information in a secrets file
3. I verified the config (success)
4. I Installed the config
5. Code built (with callouts below), device was flashed and booted, then failed with stack trace (below)
6. I updated the planformio package and tried again; same results. During that install, it seemed to be identifying my wifi network, however the reload / reboot failed. I have pasted the first log since there appeared to be a baud-rate mismatch on the second log during the stack trace which garbled some of the output.

Specific callouts that I saw in the logs that may be concerning:
1. The following two warnings:

```
/home/xxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/home/xxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/home/xxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

```
2. Two warnings regarding dependency failures for Hash and ESP8266WiFi:

```
Library Manager: Installing Hash
Library Manager: Warning! Could not install dependency {'name': 'Hash', 'platforms': ['espressif8266']} for package 'ESPAsyncWebServer-esphome'
Library Manager: Installing ESP8266WiFi
Library Manager: Warning! Could not install dependency {'name': 'ESP8266WiFi', 'platforms': ['espressif8266']} for package 'ESPAsyncWebServer-esphome'

```

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

nodemcu esp32s v1.1

### Component causing the issue

appears to be the wifi component as fault occurs immediately after.

### Example YAML snippet

```yaml
esphome:
  name: powermon1
  platform: ESP32
  board: nodemcu-32s

# Enable logging
[... 397 lines cut ...]
[13:01:20]A14      0x00000020  A15      0x00000000  SAR      0x00000000  EXCCAUSE 0x00000005  
[13:01:20]EXCVADDR 0x00000000  LBEG     0x00000000  LEND     0x00000000  LCOUNT   0x00000000  
[13:01:20]
[13:01:20]ELF file SHA256: 000000000000000
[13:01:20]
[13:01:20]Backtrace:0x40086f2:0x3fb1600x4014c03:0x3fb180 0x4008ba:0x3fb1a 0x40089c2:0x3fb1c
[13:01:20]
[13:01:20]ore 1 register dump:
[13:01:20]PC       0x4000c71  PS       0x00060034  A0       0x8008af  A1       0x3fbc0  
[13:01:20]A2       0x3fb058  A3       0x3f17780  A4       0x00000014  A5       0x3fbe0  
[13:01:20]A6       0x3fbe8  A7       0x00000001  A8       0x00000001  A9       0x3f17781  
[13:01:20]A10      0x000000a  A11      0x3fb060  A12      0x8008bb  A13      0x3fbe0  
[13:01:20]A14      0x00000008  A15      0x00000001  SAR      0x00000012  EXCCAUSE 0x00000005  
[13:01:20]EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff  
[13:01:20]
[13:01:20]ELF file SHA256: 000000000000000
[13:01:20]
[13:01:20]Backtrace:0x4000c71:0x3fbc00x4008af:0x3fbc00x4008ca:0x3fbc00x4008c5a0x4008456a
[13:01:20]
[13:01:20]Rebooting...
```

### Additional information

This does not appear to be the same as issues #1276 or #2302 since they are different boards.

Levenale, 2021-12-11:
**_Summary_**: I looked at release notes for similar changes, and then walked backward on releases until I was successfully able to load and execute the basic yaml on 1.18.0.

**Details**:
=
Did some looking around in the release notes and noticed:
* https://github.com/esphome/esphome/pull/2441

which went into the 2021.10.4 release.

I reverted esphome==2021.9.3

Tagged a clean build and attempted a build / install.

Got the following error:
```
"Library Manager: Installing Hash
Warning! Could not find the package with 'Hash' requirements for your system 'linux_x86_64'
```

Result was to still have the panic, but did not get the ESP8266WiFi error.

```
Hard resetting via RTS pin...
========================= [SUCCESS] Took 58.08 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:02:43][I][logger:174]: Log initialized
[19:02:43][C][ota:379]: There have been 0 suspected unsuccessful boot attempts.
[19:02:43][I][app:029]: Running through setup()...
[19:02:43][C][wifi:037]: Setting up WiFi...
[19:02:43][D][wifi:382]: Starting scan...
[19:02:44]Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU1)
[19:02:44]Core 1 register dump:
[19:02:44]PC       0x4000c71  PS       0x00060034  A0       0x8008ac  A1       0x3fbb0  
[19:02:44]A2       0x3fb058  A3       0x3f17fc A4       0x00000014  A5       0x3fbe0  
[19:02:44]A6       0x3fbe8  A7       0x00000001  A8       0x00000001  A9       0x3f17fd 
[19:02:44]A10      0x000000a  A11      0x00000000  A12      0x8008b77  A13      0x3fbe0  
[19:02:44]A14      0x00000008  A15      0x00000001  SAR      0x0000001e EXCCAUSE 0x00000006  
[19:02:44]EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff  
[19:02:44]
[19:02:44]ELF file SHA256: 000000000000000
[19:02:44]
[19:02:44]Backtrace:0x4000c71:0x3fbb00x4008ab:0x3fbb00x4008c68:0x3fbb00x4008c1e0x000004a
[19:02:44]
[19:02:44]ore 0 register dump:
[19:02:44]PC       0x40086f6  PS       0x00060d4  A0       0x80089c9  A1       0x3fbf0  
[19:02:44]A2       0x3fbc8  A3       0x3fb208  A4       0x00000000  A5       0x0000001a 
[19:02:44]A6       0x40083f8  A7       0x3fc45c A8       0x3f0cc  A9       0x3fbf0  
[19:02:44]A10      0x00000017  A11      0x00000000  A12      0x3fbf4  A13      0x00000040  
[19:02:44]A14      0x00000000  A15      0x3fb204  SAR      0x0000001e EXCCAUSE 0x00000006  
[19:02:44]EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff  
[19:02:44]
[19:02:44]ELF file SHA256: 000000000000000
[19:02:44]
[19:02:44]Backtrace:0x40086f6:0x3fbf00x40089c6:0x3fbf0
[19:02:44]
[19:02:44]Rebooting...
```

Next, reverted esphome==2021.8.2

received same package error and panic.

Next reverted esphome==1.20.4

same library error for Hash, same panic.

Next reverted to esphome==1.19.4

same library error for hash, same panic, but better output:
[... 359 lines cut ...]
Writing at 0x00084000... (100 %)
Wrote 872512 bytes (487065 compressed) at 0x00010000 in 42.9 seconds (effective 162.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================= [SUCCESS] Took 48.49 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:45:46][I][logger:170]: Log initialized
[19:45:46][C][ota:366]: There have been 6 suspected unsuccessful boot attempts.
[19:45:46][I][app:029]: Running through setup()...
[19:45:46][C][wifi:033]: Setting up WiFi...
[19:45:48][D][wifi:324]: Starting scan...x98\x9ef\x80\x98\x80[D][wifi:339]: Found networks:
[19:45:48][I][wifi:385]: - 'xxx' [redacted]▂▄▆█
[19:45:48][D][wifi:386]:     Channel: 1
[19:45:48]xf3\x98xx\x98\x9e\xe6\x9e`\xe6f\xaeW˥\xe9387]:     RSSI: -68 dB
[19:45:48][I][wifi:194]: WiFi Connecting to 'xxx'...
[19:45:50][W][wifi_esp32:409]: Event: Disconnected ssid='xxx' bssid=

[redacted] reason='Auth Expired'
[19:45:50][W][wifi:485]: Error while connecting to network.
[19:45:50][W][wifi:522]: Restarting WiFi adapter...
[19:45:50][I][wifi:194]: WiFi Connecting to 'xxx'...
```

jamiebegin, 2022-02-15:
I'm experiencing the same issue with the same hardware. Thank you for sharing your research.