esphome/issues #2805: Shutdown "switch" is not supported by the new button component

lanrat, 2021-12-11:
### The problem

With the release of ESPHome `2021.12.0` the new [Button Component](https://esphome.io/components/button/index.html) was added, which can now be used for things like the [Restart Button](https://esphome.io/components/button/restart.html), however it does not yet work with the [Shutdown Switch](https://esphome.io/components/switch/shutdown.html).

Can a Shutdown Button component be added?

Since shutting down, like restarting, does not track an on/off state, it would make sense to make a Shutdown Button component as well.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.0

### What platform are you using?

ESP32

### Board

NA

### Component causing the issue

Shutdown Switch

lanrat, 2021-12-11:
It would likely also be worthwhile to make the [Safe Mode Switch](https://esphome.io/components/switch/safe_mode.html) a button as well.

oxan, 2021-12-13:
Yes, I think it makes sense to convert these switches to buttons. Pull Requests welcome ;)

jesserockz, 2021-12-13:
Yes I plan to do them, but this should be a feature request, not an issue.

closed 2021-12-15 by oxan

jsuanet, 2021-12-15:
@oxan Hi, I'm working on a PR for this, is this no longer needed?

oxan, 2021-12-15:
@jsuanet No, still welcome, I closed it because it's not an issue but a feature request.