esphome/issues #2808: Buttons doesn't work via MQTT

definitio, 2021-12-12:
### The problem

Sending `PRESS` via MQTT doesn't work.

### Which version of ESPHome has the issue?

2021.12

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12

### What platform are you using?

ESP32

### Component causing the issue

button

### Example YAML snippet

```yaml
button:
  - platform: restart
    name: $device_name restart
```
```

### Anything in the logs that might be useful for us?

```txt
[W][mqtt.button:023]: 'restart': Received unknown status payload: PRESS
```

sliwma, 2021-12-13:
Here's some info from HA MQTT maintainer:
https://github.com/home-assistant/core/issues/61666#issuecomment-992445801

berfenger, 2021-12-14:
I can confirm this bug.
Home Assistant sends `PRESS` to `device_topic/button/button_id/command`, but the button does not get actually pressed. A message is logged instead: "Received unknown status payload: PRESS"
If I manually send "press" (notice the lower case) to "device_topic/button/button_id/command", the button gets actually pressed.

TL;DR
A MQTT message containing `PRESS` does not work, but `press` does work.

oxan, 2021-12-15:
Could any of you try esphome/esphome#2917?

I'll add that the MQTT integration in ESPHome is in desperate need of a maintainer. None of the currently active devs actually use MQTT nor have a test setup, so while we try our best to keep it working, mistakes like these routinely slip through the cracks.

berfenger, 2021-12-15:
This fix (https://github.com/esphome/esphome/pull/2917) works as expected for me. Thanks!

closed 2021-12-20 by jesserockz