esphome/issues #2809: Df player wont wake up

scrumpydc, 2021-12-12:
### The problem

when the dfplayer enters sleep mode there is no way of waking it . using dfplayer_set_device_tf dose nothing stays in sleep mode reset same thing stays in sleep mode need to add the always on 0x4A and/or normal working 0x0B i have also tried making a uart switch that sends the commands but cannot get it to wright to uart  either by calling a service or as a switch still wont wake hope there is a way to solve it as the only way i have found to wake it is to unplug the sd and replug or manually play by connecting to ground 

### Which version of ESPHome has the issue?

 2021.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.1

### What platform are you using?

ESP32

### Board

esp32-wroom-32d

### Component causing the issue

dfplayer

scrumpydc, 2021-12-27:
so done more testing and had it on for days and still not able to wake it and having it send a command  0x9 arg 0x02 every 5 min will not keep it awake

ssieb, 2022-01-07:
I can't reproduce this.  Sleep and wake are working fine for me.

scrumpydc, 2022-01-09:
so it seems most chips that are purchased stating that it is a dfplayer are false and that the only ones that work as suspected are the ones with the chip YX5200 i ordered from multiple suppliers and all were fakes with chips that do not run serial communication and seem to enter a low power mode and have been unable to wake any of the (16) df players i have tested to anyone who reads this i think that only the YX5200 chip is one that works so be super careful what you get here is a link to others findings regarding the different boards [https://www.thebackshed.com/forum/ViewTopic.php?TID=11977&P=1](url)