esphome/issues #2813: Onboard LED nest to antenna ESP8266 (status LED) always ON

bruno-dogancic, 2021-12-12:
### The problem

After hooking up and configuring the RGB (common cathode) LED in ESPHome, the onboard (blue) LED (next to to the antenna) stays ON.
If I change color to anything from PINK to RED to YELLOW to WHITE (right third of the color wheel) and at certain high enough brightness the onboard LED turns off, for any other color and if RGB LED is turned off the onboard LED stays ON.

I guess it has something to do with onboard LED being on same GPIO pin or something like that, but I cannot seem to figure it out... I have tried with status LED component without any success as well.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.10.7

### What platform are you using?

ESP8266

### Board

nodemcu (AZ Delivery V2)

### Component causing the issue

LIGHT RGB

### Example YAML snippet

```yaml
esphome:
  name: multisensor-andrija

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Multisensor-Andrija"
    password: "fh5KXhQp3TTF"

captive_portal:

sensor:
  # Temp and humidity
  - platform: dht
    pin: D7
    model: DHT22
    temperature:
      name: "Multisensor Andrija Temperature"
    humidity:
      name: "Multisensor Andrija Humidity"
    update_interval: 30s
  
  # Lux
  - platform: tsl2561
    name: "Multisensor Andrija LUX"
    address: 0x39
    update_interval: 30s

binary_sensor:
  - platform: gpio
    pin: D6
    name: "Multisensor Andrija Motion"
    device_class: motion

light:
# Circle "neopixel" like LED
  - platform: fastled_clockless
    chipset: WS2812
    pin: D3
    num_leds: 8
    rgb_order: GRB
    name: "Multisensor Andrija Circle LED"
    effects:
      - pulse:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
# Common cathode RGB LED
  - platform: rgb
    name: "Multisensor Andrija RGB LED"
    red: output_red1
    green: output_green1
    blue: output_blue1

output:
  - platform: esp8266_pwm
    id: output_red1
    pin: D4
  - platform: esp8266_pwm
    id: output_green1
    pin: D5
  - platform: esp8266_pwm
    id: output_blue1
    pin: D8

#status_led:
#  pin:
#    number: D0
#    inverted: true

i2c:
  sda: D2
  scl: D1
  scan: false
```

### Anything in the logs that might be useful for us?

```txt
I don't see any logs regarding the status LED.
```

martgras, 2021-12-12:
D4 (GPIO02) is connected to the internal LED

bruno-dogancic, 2021-12-12:
Shame on me. I didn't properly do my homework. I have hasted while reading and noted that GPIO00 is connected to an LED. But it turns out nodemcu has two of them... https://lowvoltage.github.io/2017/07/09/Onboard-LEDs-NodeMCU-Got-Two

Thank you for pointing that out to me @martgras .

I am closing this as it works when I connect to a non-LED GPIO pin.

closed 2021-12-12 by bruno-dogancic