esphome/issues #2816: 2021.12 breaks ESP32-S2 build

misery, 2021-12-13:
### The problem

Seems https://github.com/esphome/esphome/pull/2846 breaks the single core ESP32-S2. 2021.11.x works without problems.

```
src/esphome/components/esp32/core.cpp: In function 'void esphome::arch_init()':
src/esphome/components/esp32/core.cpp:50:3: error: 'disableCore1WDT' was not declared in this scope
   disableCore1WDT();
   ^~~~~~~~~~~~~~~
src/esphome/components/esp32/core.cpp:50:3: note: suggested alternative: 'disableCore0WDT'
   disableCore1WDT();
   ^~~~~~~~~~~~~~~
   disableCore0WDT
```

### Which version of ESPHome has the issue?

2021.12

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12

### What platform are you using?

ESP32

### Board

esp32-s2-kaluga-1

### Component causing the issue

esp32/core

### Example YAML snippet

```yaml
esp32:
  board: esp32-s2-kaluga-1

esphome:
  name: test
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```
```

### Additional information

I know that esp-idf is recommended for ESP32-S2. I need components like MQTT that is arduino only at the moment.

misery, 2021-12-13:
Maybe ```#ifndef CONFIG_FREERTOS_UNICORE``` could help.

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L109

oxan, 2021-12-13:
This seems more like a bug in Arduino, they shouldn't set `CONFIG_ARDUINO_RUNNING_CORE` to `1` for a device that only has core 0. 

I'll see if I can create a workaround, but please report this to Arduino as well.

martgras, 2021-12-15:
Can you please doublecheck that you have the latest version of the source code 
I tried to reproduce the issue on the latest dev version and I don't see the error 

https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/core.cpp#L49 also checks `CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1` which is not set for me
(Try deleting the .esphome/build directory to make sure there no stale content)

misery, 2021-12-15:
I tried ```2021.12``` release only. I checkout out current dev branch now and it works. Thanks, seems https://github.com/esphome/esphome/pull/2856 already fixed it. Could this be merged for ```2021.12.1```?

oxan, 2021-12-15:
Ah yes, that one coincidentally also fixes this issue. I've scheduled it for a bugfix release.

misery, 2021-12-17:
@oxan They don't want to change it. Maybe you want to comment it: https://github.com/espressif/arduino-esp32/issues/6019

oxan, 2021-12-20:
Since 2021.12.2 with esphome/esphome#2856 is (almost) released, and Arduino has merged a fix, I think this is resolved. Feel free to reopen if it's still persisting.

closed 2021-12-20 by oxan