esphome/issues #2817: Permission denied

eximo84, 2021-12-13:
### The problem

updating my ESP devices, now get Permission Denied after compile.
Reading appears to be an issue with execute permissions on the file system?  I mapped a new folder inside my SMB shared folder where the current configuration exists called "/cache" and gave all files and folders 777 permissions however still getting errors.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP32

### Board

nodemcu

### Anything in the logs that might be useful for us?

```txt
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <Wire> 1.0.1
|-- <ESPmDNS> 1.0
Compiling .pioenvs/esp32_dev_03/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: Permission denied
Compiling .pioenvs/esp32_dev_03/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp32_dev_03/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: Permission denied
*** [.pioenvs/esp32_dev_03/src/esphome/components/api/api_connection.cpp.o] Error 126
*** [.pioenvs/esp32_dev_03/src/esphome/components/api/api_frame_helper.cpp.o] Error 126
sh: 1: xtensa-esp32-elf-g++: Permission denied
Compiling .pioenvs/esp32_dev_03/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/esp32_dev_03/src/esphome/components/api/api_pb2.cpp.o] Error 126
sh: 1: xtensa-esp32-elf-g++: Permission denied
*** [.pioenvs/esp32_dev_03/src/esphome/components/api/api_pb2_service.cpp.o] Error 126
```

LarkeMikko, 2021-12-22:
I have had similar issue past 5 days (from december 17) and haven't found solution yet.. I have tried re installing, updating and upgrading pretty much everything related to esphome and platformio with no change.. I'm running esphome with debian if that information helps for finding solutions

coax, 2021-12-30:
It's a cache folder issue, explained here: https://github.com/esphome/issues/issues/2594#issuecomment-949419636

horodchukanton, 2022-01-10:
@coax Thanks! I've got my permissions problem fixed by adding the 'exec' option to the /etc/fstab entry for the drive that has the docker mount point on.

u20p17, 2022-03-06:
> @coax Thanks! I've got my permissions problem fixed by adding the 'exec' option to the /etc/fstab entry for the drive that has the docker mount point on.

Hi,
I have the same problem and tried your suggestion. I opened the terminal on unraid, then cd /etc and after that nano fstab... Here I added the exec option to my hard drive and saved it.

```
 GNU nano 5.3                                                                                fstab                                                                                          
/dev/disk/by-label/UNRAID  /boot         vfat      auto,rw,flush,noatime,nodiratime,dmask=77,fmask=177,shortname=mixed,exec  0  1
/boot/bzmodules            /lib/modules  squashfs  ro,defaults  0  2
/boot/bzfirmware           /lib/firmware squashfs  ro,defaults  0  2
tmpfs                      /dev/shm      tmpfs     defaults     0  0
hugetlbfs                  /hugetlbfs    hugetlbfs defaults     0  0
```

But the problem stayed there. Did I something wrong? (Unraid 6.9.2, ESPHome, 2022.02.06 in a docker)

Thanks!

ssieb, 2022-08-07:
`/boot` is not the right filesystem.  Also, your root filesystem, whatever it is, will already have execute permission or nothing would work at all.  You need to find out where your container `/config` points to or add a mapping to the container for `/cache` that points to local storage.

u20p17, 2022-10-04:
@ssieb: My /config points to /mnt/user/appdata/ESPHome. What do I have to do in order to fix the problem?

Thanks!

horodchukanton, 2022-10-04:
@ssieb Your folder should have execution permission for the user which owns the ESPHome process (docker container in your case)

In my case the problem was in the mount attributes (I had `noexec` specified)

If that is not your case you should fix the specific folder permissions.
Use `ls -l /mnt/user/appdata/ESPHome` to check the permissions on the folder.
Use `ps ax | grep esphome` to see the owner of the process.
Adjust the permissions with `chown`, `chmod` or user by specifying the --user attribute for the docker command.

u20p17, 2022-10-04:
@horodchukanton : I did the first two steps you wrote and this is the result:
[image]

The user who should get permissions is called u20p17 - how would the command look like in this case? Thank you in advance!

horodchukanton, 2022-10-04:
@ssieb I'm sorry, but the commands were wrong )
The `ls` should have `-d` to show permissions on the folder itself and the `ps` should have `u` to show the owner of the process (already updated the commands in previous message).

To make `u20p17` owner of the folder, simply call the
 `chown -R u20p17:root /mnt/user/appdata/ESPHome`
 
Also, you probably have the execution bit missing according to the file permissions. You can add it with
`chmod u+x /mnt/user/appdata/ESPHome`

u20p17, 2022-10-04:
[image]

horodchukanton, 2022-10-04:
@u20p17 That's not what I meant. You're mounting the *remote* share without the `exec` option, and as the result the *local* OS does not allow you to execute files.
You should add the `,exec` option to your `/etc/fstab` line

horodchukanton, 2022-10-04:
@u20p17 Please refer to the SSHFS documentation on how to specify the permanent mount

u20p17, 2022-10-07:
@horodchukanton: I nopw tried to add the mount to the fstab. I added the following:

`UUID=301d42ac-f4eb-4e55-8792-9ad57d1ef60f  /mnt/user/appdata/ESPHome  xfs  defaults,exec  0  0`

After saving the fstab I worte `mount --all` in the cmd and after that I tried to compile again with ESPHome... But the permission error is still there - why?

horodchukanton, 2022-10-08:
@u20p17 Im sorry, but I am not able to help you with mounts, as I know two little about your configuration. Please look for help in specific Linux community.

horodchukanton, 2022-10-08:
@u20p17 Im sorry, but I am not able to help you with mounts, as I know too little about your configuration. Please look for help in specific Linux community.

u20p17, 2022-10-11:
I think I fixed the issue now. It's a bit strange, but what I did is the following:
1.)   `ls -ld /mnt/user/appdata/ESPHome`
       Output:  ` drwxrwxrwx 1 nobody users 90 Mar  3  2022 /mnt/user/appdata/ESPHome/`
So it seemed that the every user/group should have read/write/execute rights on this folder. Nevertheless I did the following:

2.) `chmod -R 777 /mnt/user/appdata/ESPHome`
I recursive changed the rights for all users/groups to read/wrtie/execute and now it is working again... Strange - but maybe it helps other people...

/mnt/user/appdata/ESPHome points to the /config folder of ESPHome...