esphome/issues #2818: CUSTOM UART

BSD31, 2021-12-13:
### The problem

Hello.
I am using a custom UART Device which no longer works since today (update 2021.12 this morning)
Typically, I receive a message on RX (56: 62: 02: AC: 00: 00) and I acknowledge receipt via TX (2F: 33: 0A)

Here is my .h file
```
#include "esphome.h"
class Read_Data : public PollingComponent, public UARTDevice {
	public:
		Read_Data(UARTComponent *parent) : UARTDevice(parent) {}
 	int pos;
	int data[7];
	void setup() override {
	}
	void update() override{
		pos = 0;
		while(available()) {
 			pos++;
			data[pos]  = read();
			if (pos == 6) {
				write(0x2F);
				write(0x33);
				write(0x0A);
			}
		}
	}
};
```

and my .yaml file
```
uart:
      - id: uart_bus
        tx_pin: GPIO14
        rx_pin: GPIO12
        baud_rate: 9600
        debug:
    
    custom_component:
      - lambda: |-
          auto Data = new Read_Data(id(uart_bus));
          return {Data};
```

The logs do not give any error :
[15:04:37][D][uart_debug:114]: <<< 56:62:02:AC:00:00
[15:04:37][D][uart_debug:114]: >>> 2F:33:0A
[15:04:40][D][uart_debug:114]: <<< 56:62:02:AC:00:00
[15:04:40][D][uart_debug:114]: >>> 2F:33:0A
[15:04:46][D][uart_debug:114]: <<< 56:62:02:AC:00:00
[15:04:46][D][uart_debug:114]: >>> 2F:33:0A
[15:04:49][D][uart_debug:114]: <<< 56:62:02:AC:00:00
[15:04:49][D][uart_debug:114]: >>> 2F:33:0A
---/---

But in reality there is nothing on the TX

I added a switch in my .YAML to send my acknowledgment manually and it works ...

```
    switch:
      - platform: uart
        name: "0x2F:0x33:0x0A"
        data: [0x2F, 0x33, 0x0A]
```

[15:11:07][D][switch:021]: '0x2F:0x33:0x0A' Toggling ON.
[15:11:07][D][switch:037]: '0x2F:0x33:0x0A': Sending state ON
[15:11:07][D][uart.switch:020]: '0x2F:0x33:0x0A': Sending data...
[15:11:07][D][switch:037]: '0x2F:0x33:0x0A': Sending state OFF

What happened ?
Am I doing something wrong?

Best regards

### Which version of ESPHome has the issue?

v2021.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

HA-OS core-2021.12.1 in Synology VMM

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

CUSTOM UART

BSD31, 2021-12-14:
Same result with an ESP01, ESP07, ESP12 AND ESP32 ...
I also tried with the 'Dev' version of ESPHome.
The logs are correct but no real signal on TX.
Is the problem only with me ?
Thank you for your help

BSD31, 2021-12-14:
So is there no one who can analyze my simple little piece of code ?
No one to say if this is a big mistake on my part or a little bug from the last update ?
Please guide me.

mmakaay, 2021-12-15:
If it were a simple error, I'm sure somebody would have responded by now. Looking at the code I see no obvious mistake. The debug logs tell us that there is data being sent to the UART component, because it's the UART component that is logging the bytes on their way out.

What would help is if you would try with older versions, to find the last version for which things still worked. That would make it a lot easier to look at what might have caused the issue for you.

You say "But in reality there is nothing on the TX". How did you check that? Did you connect the TX to a serial logger to verify this?

BSD31, 2021-12-15:
Oh thank you for your analysis.
I feel less alone.
I have not connected a laboratory serial logger but the device which is connected to my UART port lights a connection icon if the acknowledgment is received and valid.
This icon has two different states:
     0x2F: 0x32: 0x0A -> The icon blink.
     0x2F: 0x33: 0x0A -> The icon is On.
I created two switches in my .YAML that write directly to the UART port.
When I click manually the icon lights up in the two different states.
On the other hand, automatic writing to the UART port via the .h file does not light up the icon in any case.
Finally, my code was working before the last update.
I do not really know what else to do...

BSD31, 2021-12-15:
I will try to find an oscilloscope to visualize the signal...

BSD31, 2021-12-19:
Eureka!
I found the reason for my bug.
With the oscilloscope, I noticed that when I write from the .h file :

```
    .../...
    write (0x2F);
    write (0x33);
    write (0x0A);
    .../...
```

this is not equal to what I write from the .yaml

```
    switch:
        - platform: uart
           name: "TEST (0x2F: 0x33: 0x0A)"
           data: [0x2F, 0x33, 0x0A]
```

In the .h file the data is sent independently of each other (on different lines) while in the .yaml file the data is sent one after the other (on the same line ).
It seems to me that this problem did not exist before the December update !
How now to write to the same line from the .h file because [0x2F, 0x33, 0x0A] does not work ?
Thank you for your help.

[image]

netsnakecq, 2022-01-07:
using write_array ?

BSD31, 2022-02-03:
Exactly !
Using 'write_array' it works perfectly.