esphome/issues #2820: Substitutions in secret name not compiling

EmnaX, 2021-12-13:
### The problem

Hello!

I recently updated Home Assistant to 2021.12.0 and ESPHome to 2021.12.0. Prior to the update, my yaml files for a NodeMCU would compile fine with substitutions used in a variable name for a secret. After the update, I get an error. My system also crashed during the update and I had to restore from a backup so it's possible something else is going on here. Anyway, anyone else using substitutions in secret names successfully??  

My NodeMCU yaml file (frame-b-status.yaml):  
```
# This is a NodeMCU used to monitor the program run status for Frame B.

substitutions:
  frame: 'Frame B'
  frame_strict: 'frame-b' # must be [a-z] [0-9] or - or _

########################
## Unique Node Config ##
########################

# Enable Home Assistant API
api:
  password: !secret ${frame_strict}_api-ota_password

ota:
  password: !secret ${frame_strict}_api-ota_password

wifi:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret ${frame_strict}_fallback_hotspot_password

########################
## Common Node Config ##
########################

<<: !include common_config-V1-3.yaml
# Note, I haven't shown the contents of the common config since I don't
# think it is relevant. I can certainly add it if necessary.
```

My secrets.yaml file:
```
# All the passwords for my ESPHome devices

wifi_ssid_password: "monkey"

frame-b_api-ota_password: "monkey2"
frame-b_fallback_hotspot_password: "monkey3"
```

And the error I'm getting:
```
INFO Reading configuration /config/esphome/frame-b-status.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Secret '${frame_strict}_api-ota_password' not defined
  in "/config/esphome/frame-b-status.yaml", line 14, column 13:
      password: !secret ${frame_strict}_api-ot ... 
                ^
```

So it seems like the variable `${frame_strict}` isn't getting replaced with the value `frame-b` before the search for the secret name takes place.  

The thing is, this compiled fine yesterday and I have the code running on a NodeMCU... could this be a bug in the ESPHome 2021.12.0 release?

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.0

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

secrets

### Anything in the logs that might be useful for us?

```txt
I would be happy to include more logging info if you let me know where to look or what you need.
```

### Additional information

I also posted over on the Home Assistant Community:
https://community.home-assistant.io/t/esphome-using-secrets-with-substitution/366253

EmnaX, 2021-12-20:
Update:
I just rolled back to a backup from 2021-11-29 and the issue remains... so nothing broke in the update I guess. I'm not sure how I was able to compile before... maybe I added the substitution into the secret name after?? Not sure. Anyway, this would be a nice feature to add so I don't have to use the same secrets for all the nodes. Not a big deal though.

For completeness, here are the versions I was using when I went back to my 2021-11-29 backup:
core-2021.11.4
supervisor-2021.12.2
Home Assistant OS 7.0
ESPHome v2021.11.4

jesserockz, 2021-12-22:
secrets are parsed as part of the yaml preprocessing code, and substitutions are done after that. Not sure how it would have ever worked.

closed 2022-05-07 by github-actions[bot]