esphome/issues #2822: Error while compiling ESP32 -->nodemcu-32s

ashrafmn, 2021-12-14:
### The problem

I was able to continuously compile and upgrade to latest ESPHome version.
Only since the last two updates I'm not able to compile my only ESP32 device. Unlike all my other devices which are using ESP8266/esp01_1m boards are being successfully compiled with no issues.

Please note, this sane yaml file was working fine with no issues at all with previous ESPHome versions

### Which version of ESPHome has the issue?

Current version: 2021.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.12.1

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

sh: 1: xtensa-esp32-elf-g++: not found

### Example YAML snippet

```yaml
esphome:
   name: 'esp32sweather'
   platform: ESP32
   board: nodemcu-32s

    
    
wifi:
  ssid: 'yyyyyyyy'
  password: 'xxxxxxxxxx'
  fast_connect: on
  manual_ip:
   static_ip: 192.168.xx.xxx
   gateway: 192.168.xx.xxx
   subnet: 255.255.255.0
   dns1: 192.168.xx.xxx
  ap:
    ssid: 'esp32sweather'
    password: "12345678"

captive_portal:

api:
web_server:
  port: 80
logger:
  level: verbose

ota:

time:
  - platform: sntp
    id: sntp_time
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org

sensor:
    
  - platform: dht
    model:  AM2302
    pin: GPIO25
    temperature:
      name: "esp32sweather DHT11 Temperature"
      id: dht11temp
    humidity:
      name: "esp32sweather DHT11 Humidity"
      id: dht11humid
    update_interval: 60s
    
    
  - platform: wifi_signal
    name: 'esp32sweather WiFi Signal'
    update_interval: 60s

switch:
  - platform: restart
    name: "esp32sweather Restart"

text_sensor:
  - platform: template
    name: "esp32sweather_MAC_Address"
    lambda: 'return {WiFi.macAddress().c_str()};'
    icon: mdi:fingerprint
    update_interval: 500s
    
  - platform: template
    name: "esp32sweather IP Address"
    lambda: 'return {WiFi.localIP().toString().c_str()};'
    icon: mdi:fingerprint
    update_interval: 500s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp32S_weather.yaml...
INFO Detected timezone 'Asia/Riyadh'
INFO Generating C++ source...
INFO Compiling app...
Processing esp32sweather (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 5.22 seconds ==========================
```

### Additional information

I'm using pi 64bit device booting through USB.

dooziedan, 2021-12-20:
I can confirm that behavior. I solved it by shutting down my nginx add-on and downloading the compiled code manually in esphome.

ashrafmn, 2021-12-20:
Good news,, 
So how do I get/deploy the fix??

ashrafmn, 2022-01-02:
hello dears,

how do use the fix to recompile my device? as i mentioned, i'm using HassOS v7

ashrafmn, 2022-02-25:
hello,

can someone help here,, im still having the same issue, even-though ESPHome version is now updated to 2022.2.5

i'm not able to compile  my ESP 32 chip!!

ashrafmn, 2022-05-02:
Hello
Anyone can help here please!!
I'm still not able to compile my ESP32 chip on hassio for ESPhome!!

ashrafmn, 2022-05-21:
@dooziedan 
Hi,,

any chance to help here,, 
I'm still getting errors while complaining my ESP32

I'm now on :
Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.2
Home Assistant OS 8.1
Kernel version 5.15.32-v8
Agent version 1.2.1

and ESPHome 2022.3.1

and the issue still the same :-(

below is the output when compiling:

```
INFO Reading configuration /config/esphome/esp32S_weather.yaml...
INFO Detected timezone 'Asia/Riyadh'
INFO Generating C++ source...
INFO Compiling app...
Processing esp32sweather (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
|-- <ArduinoJson> 6.18.5
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/esp32sweather/.pioenvs/esp32sweather/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 7.63 seconds ==========================
```