esphome/issues #2823: pulse_meter reports values twice or not as expected

rmeissn, 2021-12-14:
### The problem

I'm trying to translate rotations of an old Ferraris energy meter to current energy flow (in W). I've set up a esp8266 with a [TRCT5000](https://www.az-delivery.de/products/linienfolger-modul-mit-tcrt5000-und-analog-ausgang) and attached the digital signal pin to an IO-pin of the esp. I've managed to get a clear signal (one pulse per rotation, no flicker) from this setup (see output of "Pin" in the log).

So I configured a pulse_meter as (see yaml-config below), but it puts out strange values - one value when switching from off to on, one value when switching from on to off - see logs. Actually it should report just one value, when switching from off to on. I managed to get only one value by increasing the time for the internal_timeout to like 10 seconds, but that's not the intended use case (at least according to the [documentation](https://esphome.io/components/sensor/pulse_meter.html)).

According to the documentation pulse_meter "measures the time between rising edges on a pin, for each pulse it outputs the frequency in pulses/min.". So one output for one pulse, not two (in case the pulse won't flicker). 
"internal_filter (Optional, Time): If a pulse shorter than this time is detected, it’s discarded and no pulse is counted.". So by setting this to e.g. 250ms, it filters flickering pulses shorter than 250ms (which isn't the case for me; see logs (no rapid on/off))

See this picture for a graphical explanation:
[image]

I'm not sure whether the implementation is buggy, the documentation needs to be improved or whether my setup is misconfigured.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

d1mini

### Component causing the issue

sensor/pulse_meter

### Example YAML snippet

```yaml
esphome:
  name: zaehler
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  password: "Hidden"

wifi:
  ssid: "Hidden"
  password: "Hidden"
  power_save_mode: high
  output_power: 10
    
binary_sensor:
  - platform: gpio
    pin:
      number: D2
    name: "Pin"

sensor:
  - platform: pulse_meter
    name: "Current Energy Flow"
    unit_of_measurement: 'W'
    device_class: 'power'
    state_class: 'measurement'
    pin:
      number: D2
    timeout: 5min
    internal_filter: 250ms
    filters:
      - multiply: 800 # (60s / 75 pulses per kWh) * 1000 for W
      - filter_out: nan
```

### Anything in the logs that might be useful for us?

```txt
[12:55:16][D][binary_sensor:036]: 'Pin': Sending state ON
[12:55:16][D][sensor:113]: 'Current Energy Flow': Sending state 638.22998 W with 2 decimals of accuracy
[12:55:19][D][binary_sensor:036]: 'Pin': Sending state OFF
[12:55:19][D][sensor:113]: 'Current Energy Flow': Sending state 12920.59180 W with 2 decimals of accuracy
[12:56:36][D][binary_sensor:036]: 'Pin': Sending state ON
[12:56:36][D][sensor:113]: 'Current Energy Flow': Sending state 622.26135 W with 2 decimals of accuracy
[12:56:40][D][binary_sensor:036]: 'Pin': Sending state OFF
[12:56:40][D][sensor:113]: 'Current Energy Flow': Sending state 13844.82227 W with 2 decimals of accuracy
[12:57:56][D][binary_sensor:036]: 'Pin': Sending state ON
[12:57:56][D][sensor:113]: 'Current Energy Flow': Sending state 627.96814 W with 2 decimals of accuracy
[12:58:00][D][binary_sensor:036]: 'Pin': Sending state OFF
[12:58:00][D][sensor:113]: 'Current Energy Flow': Sending state 12789.76855 W with 2 decimals of accuracy
```

rmeissn, 2021-12-14:
This comment/issue might be related and states the filter function to work differently than documented: https://github.com/esphome/issues/issues/2022#issuecomment-834574644
Comment: "This is the length of time after a pulse that we will ignore any further pulses, ..."
Documentation: "If a pulse shorter than this time is detected, it’s discarded and no pulse is counted."

Within the same issue, my own problem seems to be noticed too as of https://github.com/esphome/issues/issues/2022#issuecomment-834767896

rmeissn, 2022-01-12:
I've tested it again with esphome 2021.12.3, but still got the same (log) output as in my first comment.
I've managed to get rid of the second (higher) published state by using a filter `- lambda: if (id(binaryPin).state) { return x; } else { return {}; }` which only reports on high state of the binary sensor, so after a rising edge occurred. But I guess this won't work for higher frequencies?!
I can only guess that the reported value is slightly incorrect (too high), as it is calculated based on the time between the last edge something was reported (so the falling edge of a pulse) and the next rising edge (rising edge of the next pulse).

dtgeorge, 2022-01-20:
Got same issue here. It seems they track both rising and falling edges of the pulse. Internal_filter could help, but requires very low pulse duty cycle OR nearly fixed pulse frequency. That way we can "cover" second pulse edge with internal_filter interval.

dtgeorge, 2022-01-20:
I think I found it)
Here: https://github.com/esphome/esphome/blob/dev/esphome/components/pulse_meter/pulse_meter_sensor.cpp
This part:
`void PulseMeterSensor::setup() {
  this->pin_->setup();
  this->isr_pin_ = pin_->to_isr();
  this->pin_->attach_interrupt(PulseMeterSensor::gpio_intr, this, **gpio::INTERRUPT_ANY_EDGE**);

  this->last_detected_edge_us_ = 0;
  this->last_valid_edge_us_ = 0;
}`

gpio::INTERRUPT_ANY_EDGE should be replaced to something like "leading edge"

dtgeorge, 2022-01-20:
Yo!

Got walkaround!!! It's a bit ugly, but...

So idea is that pulse_meter component tracks both rising and falling edges of the pulse (meanwhile doc says that only rising one). Thus, while not fixed, we need some pulse divider to divide sensor pulse count by two and put result as pulse_meter input. We can make divider hardware with couple of triggers or software  (if you have enough free pins)

Here is my config:

```
binary_sensor:
  - platform: gpio
    id: water_count_raw
    pin: D0 #connect you water, electricity, etc. counter here
    on_release: # pulse edge doesn't matter, but only one of two !!!
      then:
        - switch.toggle: ipin

switch:
  - platform: gpio
    id: ipin
    pin: D1
    
sensor:
  - platform: pulse_meter
    pin: D2
    name: 'Water Speed Indoor'

```
Connect sensor output to DO and shorten D1 to D2. That's it!

rmeissn, 2022-01-20:
@dtgeorge Your workaround is quite smart! Thanks for sharing it =) I guess you could also use:

```
switch:
  - platform: gpio
    pin: D1
    id: ipin
    on_turn_on:
    - delay: 50ms
    - switch.turn_off: ipin
```

and set the pulse_meters internal filter to 50-100ms. So it should work similar to a S0 port, but both implementations will break when this issue gets fixed. Yours because the falling edge isn't recognized anymore and mine because the pulse will get filtered.

dtgeorge, 2022-01-20:
> @dtgeorge Your workaround is quite smart! Thanks for sharing it =) I guess you could also use:
>
> ```
> switch:
>   - platform: gpio
>     pin: D1
>     id: ipin
>     on_turn_on:
>     - delay: 50ms
>     - [...]

Not sure yours are as good as mine ) As far as I know using any kind of delay is a bad practice. Depending on realization, delay could block execution for other threads. So, unless you are using some wrappers for particular implementation (like JavaScript setTimeout(), etc.), using pause, delays, etc should be avoided...

jesserockz, 2022-01-20:
> As far as I know using any kind of delay is a bad practice

@dtgeorge 
A delay in YAML like that is non-blocking and is basically like `setTimeout` in JS.

jesserockz, 2022-01-20:
Also

> So idea is that pulse_meter component tracks both rising and falling edges of the pulse (meanwhile doc says that only rising one).

If you look at the code for the pulse meter interrupt, it attaches to any edge, but then it will do:
```c++
  // We only look at rising edges
  if (!sensor->isr_pin_.digital_read()) {
    return;
  }
```

This is because you could `invert` the pin in config based on your needs

dtgeorge, 2022-01-20:
> Also
>
>
> If you look at the code for the pulse meter interrupt, it attaches to any edge, but then it will do:
>
> ```c++
>   // We only look at rising edges
>   if (!sensor->isr_pin_.digital_read()) { [...]

Oops...
"So what?"
Then why pulse_sensor emit increment to each pulse edge??????

stevebaxter, 2022-01-20:
So we should definitely only count the rising edges. One possible problem is that you're getting noise on your falling edge, so we're seeing it as a second rising edge. You probably won't see this without putting an oscilloscope on the input though - `binary_sensor` may not see it as it isn't fast enough (it's just a polling sensor I think, it's not interrupt based).

The solution here should be to use the internal filter option, this is exactly what it's for. In your case the water meter has a very slow pulse time, I would probably set the filter to 10s so you get the first rising edge and then ignore the rest of the pulse including any noise on the falling edge.

dtgeorge, 2022-01-20:
> So we should definitely only count the rising edges. One possible problem is that you're getting noise on your falling edge, so we're seeing it as a second rising edge. You probably won't see this [...]

Anyway, there are something wrong in your code... Let's say (but not the only) these:
`const uint32_t pulse_width_ms = this->pulse_width_us_ / 1000`

Are you sure that 1ms == 1 us / 1000 ????)

dtgeorge, 2022-01-20:
> So we should definitely only count the rising edges. One possible problem is that you're getting noise on your falling edge, so we're seeing it as a second rising edge. You probably won't see this [...]

I used oscilloscope - nothing suspicious. May be it's esp8266 issue. And your digital_read() gives wrong results... But definitely pulse_meter emits updates on every!!! edge... 
internal_fiter doesn't solve issue, cause my pulses duty cycle too high (it's magnet stocked to gear, sometimes counter can even stop while magnet is over hall sensor).

dtgeorge, 2022-01-21:
Made some tests... 
I can confirm that config bellows 100% working, as pulse_meter total match mechanical counter delta (tested 130 liters with 0.3 liter inaccuracy):

```
switch:
  - platform: gpio
    id: ipin
    pin: D1

binary_sensor:
  - platform: gpio
    name: water_count_raw_indoor
    pin:
      number: D0
      inverted: true
    on_press:
      then:
        - switch.turn_on: ipin
    on_release:
      then:
        - switch.turn_off: ipin    
sensor:
  - platform: pulse_meter
    pin: D2
    unit_of_measurement: 'L/min'
    name: 'Water Speed Indoor'
    timeout: 60s
    accuracy_decimals: 3
    filters:
      - multiply: 0.310344 # 9/29
          
    total:
      name: "Water Total Indoor"
      accuracy_decimals: 3
      unit_of_measurement: "L"
      filters:
        - multiply: 0.310344 # 9/29

```

So pulse_meter obviously tracks (trying to) rising edge, but due to esp8266 issue tracks both edges. Adding RC chain to input doesn't help, so this is not about bounce!

dtgeorge, 2022-01-21:
Here is my counter:

[image]

rmeissn, 2022-01-21:
@stevebaxter According to the documentation, I'd expect a binary_sensor to also being interrupt based, but I haven't looked at the code. See https://esphome.io/components/binary_sensor/gpio.html#debouncing-values

So, if I understand it right, the implementation of the internal_filter is not debouncing. So it's not doing: ignore flicker; only count a rising edge if the whole pulse (low - high - low) is longer than the e.g. 100ms.
Instead, it does:
1. Is there a rising edge? Count, even if it is from a flickering signal
2. ignore any signal changes after this rising edge for the specified time
3. if the signal is still high and there is any flicker at the falling edge: Count again
This should be made clear in the docs.

Is it possible to introduce a debouncing filter, which is ignoring pulses (low - high - low) shorter than a specified time?
So this filter would be supposed to do (counts at the rising edge):
rising edge - note time as time1, set timer
falling edge - note time as time2, reset timer if `time2 - time1 <= specified time` 
timer elapsed - signal still high? note a pulse

More simplistic version (counts at the falling edge):
rising edge - note time as time1
falling edge - note time as time2, note a pulse if `time2 - time1 > specified time` 

Both proposed implementations allow to debounce a signal with software instead of hardware.

dtgeorge, 2022-01-21:
> @stevebaxter According to the documentation, I'd expect a binary_sensor to also being interrupt based, but I haven't looked at the code. See https://esphome.io/components/binary_sensor/gpio.html#debouncing-values

Why not just copy debounce filter implementation from `binary_seonsor `component code? I guess it should be there, at least switching hall sensor to `binary_sensor` works great!

dtgeorge, 2022-01-21:
> Looks like someone has just done that, see [esphome/feature-requests#1354](https://github.com/esphome/feature-requests/issues/1354) and [esphome/esphome#3082](https://github.com/esphome/esphome/pull/3082).

I am confused, can you explain me? @stevebaxter 

I am using `binary_sensor` as a walkaround by proxying hall sensor output to `pulse_meter`. You can find working config few posts above. That `binary_sensor` has no any debounce settings (delays, etc), works ok.

If bounce/noise would be present then afaik binary sensor wouldn't work (at least without some filtering), as it's the same wire connected to same pin. But it works somehow. The only reason may be - `binary_sensor` has some embedded default bounce filtering? Or it's more complicated reasons...

So the question is how could that be? I am not sure that filtering each edge separately could help, as it looks like problem may not be bounce/noise related...

stevebaxter, 2022-01-21:
If you look at the issue and PR above, it looks like exactly your issue, and it looks like @cstaahl has provided the solution you've just asked for (to add an option to debounce the rising edge).

dtgeorge, 2022-01-25:
Got another walkaround )))
It's available with core-2021.12.10 

Idea is simple - count pulses and all staff outside ESP.
ESP used just to fix pulses with binary sensor, here we can filter edges, etc.

So, here is esp node config:
```
binary_sensor:
  - platform: gpio
    name: "indoor water counter raw"
    pin:
      number: D1
      inverted: true
    filters:
      - delayed_on: 50ms
      - delayed_off: 50ms
```

And here is main configuration.yml:
```
counter:
  indoor_water_counter_pulses:

automation:
  action:
    service: counter.increment
    data:
      entity_id: counter.indoor_water_counter_pulses

template:
  - sensor:
      - name: "Indoor Water Total"
        unit_of_measurement: l
        state: >
          {{ states('counter.indoor_water_counter_pulses') | int * 0.310344 }}
  - sensor:
      - name: "Indoor Water Speed"
        unit_of_measurement: "l/min"
        state: >
          {%- if as_timestamp(now()) - as_timestamp(states.sensor.indoor_water_total.last_updated) > 30 -%}
          0
          {%- else -%}
          {{ states('sensor.indoor_water_speed_per_sec') | float(default=0) * 60 | round(2) }}
          {%- endif -%}

sensor:
  - platform: statistics
    name: "Indoor Water Speed per sec"
    entity_id: sensor.indoor_water_total
    precision: 4
    state_characteristic: change_second #these on require update core-2021.12.10
    sampling_size: 2
```

TommySharpNZ, 2022-02-17:
I’m definitely having this issue with my pulse counter… Do we know if there is a standard way to resolve this now? Or should I be trying some of the workarounds listed?

netweaver1970, 2022-02-17:
I think I also have this issue on 2 of my 3 ESPHome delivered pulse_meter readings, the water and gasmeter. 

The Electricity sensor uses an S0 port, those meter values are very accurate although are very spikey -high- when pumps or other inductive loads start in the house. Which might be correct after all, knowing the effects. 
Still, short peaks of 3 kWh consumption when the pump in  a central heating gasboiler switches on is a lot. 

The water and gas sensors are wired up with reed contact swtich, so probably noisy as hell. 
They consistently indicate way more than actual consumption. 
Should I wait for the official solution (the FILTER_MODE thing) or start hacking in one of the above workarounds?

rmeissn, 2022-02-21:
https://github.com/esphome/esphome/pull/3082 was merged recently and will probably be published as of the 2022.3 release. I'll test this new filter mode as soon as this new release happens. Until then, I'll stick to [this alternative project](https://github.com/Eisbaeeer/Ferraris_MQTT_Energy_Counter_Meter_TCRT5000) to get correct readings.

rmeissn, 2022-03-21:
The merged functionality, released with esphome 2022.3, works perfectly fine for me and solves this issue. Simply use the new internal_filter_mode option PULSE ( https://esphome.io/components/sensor/pulse_meter.html ) with a filter time of ~150ms (maybe less) for Ferraris energy meters.

closed 2022-03-21 by rmeissn

HausnerR, 2022-03-23:
Hello, looking on new pulse_meter code I'm worried that it cannot properly count pulses shorter than `loop()` interval.
Loop interval is about 15ms, so if we assume:

internal_filter_mode: PULSE,
internal_filter: DEFAULT (13us),
and we want to count pulse of 50us which will be between loop executions.

then look here: https://github.com/esphome/esphome/blob/dev/esphome/components/pulse_meter/pulse_meter_sensor.cpp#L34

First filtering `if` will be true, but second `if` and `else if` never execute because pulse is between loops. We never get
`!this->sensor_is_high_ && this->isr_pin_.digital_read()` or `this->sensor_is_high_ && !this->isr_pin_.digital_read()` as true. So pulse will be ommited.

cstaahl, 2022-03-23:
> Hello, looking on new pulse_meter code I'm worried that it cannot properly count pulses shorter than `loop()` interval. Loop interval is about 15ms, so if we assume:
>
> internal_filter_mode: PULSE, [...]

You are right. Furthermore it seems to cause the issue described in https://github.com/esphome/issues/issues/3143 when pulses are fired during the loop.
I've created a PR that should address both the linked issue and your concern: https://github.com/esphome/esphome/pull/3321