esphome/issues #2825: Multi-Click - At Least timing

torianironfist, 2021-12-14:
### The problem

I have a standard toggle (not momentary) switch at the top of my basement stairs with a Shelly 1PM installed.  I would like to use this switch to switch the relay normally when switched once and turn off all the basement lights when switched twice quickly.  I wrote some yaml for a binary sensor using on_multi_click.

It works with a couple problems, the first switch after booting does not trigger an action and the next switch after the "double click" action also does not trigger.

As far as I can tell, the issue is starting a multi-click timing with the "at least" grammar.  When first booted or flashed, the timing is not initialized, so it cannot time "at least" until the switch has been toggled once and it appears a similar thing happens after the "double click", it no longer keeps track of the timing between the last switch.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12.1

### What platform are you using?

ESP8266

### Board

Shelly 1PM

### Component causing the issue

Binary Sensor

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO4
    id: switchid
    on_multi_click:
      - timing:
          - ON for at most 0.3s
          - OFF for at least 0.2s
        then:
          - homeassistant.service:
              service: light.turn_off
              data:
                entity_id: light.group
          - logger.log: "Double-Click - Off to On to Off"
      - timing:
          - OFF for at most 0.3s
          - ON for at least 0.2s
        then:
          - homeassistant.service:
              service: light.turn_off
              data:
                entity_id: light.group
          - logger.log: "Double-Click - On to Off to On"
      - timing:
          - ON for at least 0.5s
          - OFF for at least 0.5s
        then:
          - switch.toggle: shelly_relay
          - logger.log: "Single-Click - On to Off"
      - timing:
          - OFF for at least 0.5s
          - ON for at least 0.5s
        then:
          - switch.toggle: shelly_relay
[... 70 lines cut ...]
[14:06:08][V][binary_sensor.automation:025]: START min=0 max=300
[14:06:08][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[14:06:08][V][binary_sensor.automation:035]: Multi Click: action not started because first level does not match!
[14:06:08][V][binary_sensor.automation:025]: START min=500 max=4294967294
[14:06:08][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[14:06:08][V][binary_sensor.automation:095]: Multi Click: You waited too long to PRESS.
[14:06:08][V][binary_sensor.automation:071]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[14:06:09][V][binary_sensor.automation:089]: Multi Click: You can now PRESS the button.
[14:06:09][V][binary_sensor.automation:074]: Multi Click: Cooldown ended, matching is now enabled again.
[14:06:13][D][binary_sensor:036]: 'switchid': Sending state ON
[14:06:13][V][binary_sensor.automation:025]: START min=0 max=300
[14:06:13][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[14:06:13][V][binary_sensor.automation:035]: Multi Click: action not started because first level does not match!
[14:06:13][V][binary_sensor.automation:025]: START min=500 max=4294967294
[14:06:13][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[14:06:13][V][binary_sensor.automation:062]: C i=1 min=500
[14:06:13][V][binary_sensor.automation:095]: Multi Click: You waited too long to RELEASE.
[14:06:13][V][binary_sensor.automation:071]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[14:06:14][V][binary_sensor.automation:089]: Multi Click: You can now RELEASE the button.
[14:06:14][V][binary_sensor.automation:101]: Multi Click: Hooray, multi click is valid. Triggering!
[14:06:14][D][switch:021]: 'Light' Toggling OFF.
[14:06:14][D][switch:037]: 'Light': Sending state OFF
[14:06:14][D][main:414]: Single-Click - Off to On
[14:06:14][V][binary_sensor.automation:074]: Multi Click: Cooldown ended, matching is now enabled again.
```

torianironfist, 2022-01-23:
If anyone else runs into this, I was able to create a workaround with lambdas.  This is only set up to distinguish between single and double clicks, but could be extrapolated further if more complex patterns were needed.

`esphome:
  on_boot:
    then:
      - lambda: |-
          id(multi_click_time) = millis();
    
globals:
  - id: multi_click_time
    type: int
    restore_value: no
  - id: single_click
    type: bool
    restore_value: no
	
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO5
    id: switch
    on_state:
      then:
        - if: 
            condition:
              lambda: 'return millis() - id(multi_click_time) > 500;'
            then:
              - lambda: 'id(multi_click_time) = millis(); id(single_click) = true;'
            else: 
              - lambda: 'id(multi_click_time) = millis();'
              - logger.log: "Double Click Detected!"
              # Double click action here
        - delay: .5s
        - if: 
            condition:
              lambda: 'return id(single_click) == true && millis() - id(multi_click_time) > 500;'
            then:
              - lambda: 'id(single_click) = false;'
              - logger.log: "Single Click Detected!"
              # Single click action here
            else: 
              - lambda: 'id(single_click) = false;'`

jesserockz, 2022-01-23:
Something I have done to get around this was to make a template binary sensor, and turn that binary sensor on, delay 50ms and turn back off again, so basically turning every state change (off to on or on to off) into a click and doing the timing logic in that template binary sensor.

jesserockz, 2022-01-23:
```yaml
- platform: gpio
  id: raw_switch
  pin:
    number: 0
    mode: INPUT_PULLUP
  on_state:
    then:
      - binary_sensor.template.publish:
          id: switch_clicker
          state: ON
      - delay: 100ms
      - binary_sensor.template.publish:
          id: switch_clicker
          state: OFF

- platform: template
  id: switch_clicker
  name: ${friendly_name}
  on_multi_click:
  - timing:
    - ON for at most 0.2s
    - OFF for at least 0.5s
    then:
      # single "click" actions - up or down once only
  - timing:
    - ON for at most 0.2s
    - OFF for at most 0.5s
    - ON for at most 0.2s
    - OFF for at least 0.5s
    then:
      # double click actions - up down, or down up
    ```

torianironfist, 2022-01-23:
Making the template into a faux momentary switch.  Brilliant.  That is much easier to read than my code.  Thanks very much.

jaddel, 2022-01-31:
this is great, thx!

jaddel, 2022-02-13:
For those looking for a complete example:

```
binary_sensor:

# Wall Swtich 1
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT
    id: sensorid1
    filters:
      - delayed_on_off: 80ms # Debounce filter to eliminate random false switching. 
    on_state:
      then:
        - binary_sensor.template.publish:
            id: switch_clicker
            state: ON
        - delay: 100ms
        - binary_sensor.template.publish:
            id: switch_clicker
            state: OFF

  - platform: template
    id: switch_clicker
    on_multi_click:
      - timing:
        - ON for at most 0.5s
        - OFF for at least 0.5s
        then:
          - logger.log: "Single Short Click"
          - lambda: |-
              if (id(template_sensor_1).state) {
              id(template_sensor_1).publish_state(false);
              } else { 
              id(template_sensor_1).publish_state(true);
              }

      - timing:
          - ON for at most 0.5s
          - OFF for at most 0.5s
          - ON for at most 0.5s
          - OFF for at least 0.5s
        then:
          - logger.log: "Double Click"
          - lambda: |-
              if (id(template_sensor_2).state) {
              id(template_sensor_2).publish_state(false);
              } else {
              id(template_sensor_2).publish_state(true); 
              }

      - timing:
          - ON for at most 0.5s
          - OFF for at most 0.5s
          - ON for at most 0.5s
          - OFF for at most 0.5s
          - ON for at most 0.5s
          - OFF for at least 0.2s
        then:
          - logger.log: "Tripple Click"
          - lambda: |-
              if (id(template_sensor_3).state) {
              id(template_sensor_3).publish_state(false);
              } else {
              id(template_sensor_3).publish_state(true); 
              }

  - platform: template
    name: Remote 1 $friendly_name
    id: template_sensor_1
  - platform: template
    name: Remote 2 $friendly_name
    id: template_sensor_2
  - platform: template
    name: Remote 3 $friendly_name
    id: template_sensor_3
```

SmoothSilence, 2022-05-31:
Hi guys. I came across the same problem but I use the mult-click for a LED strip.
The idea was a single press turns it on or off,
double click goes to max-min brightness of the LED strip
and hold slowly dims up or down the strip till you let go.
Besides physical push buttons I use HA too. On both sides HA and wall buttons I have that error and it turns off the LED strip
Could you point me to where I messed up here `?`
`esphome:
  name: yard

esp32:
  board: esp32dev

# Enable Home Assistant API
api:
  #password: !secret api_password
  
ota:
  password: "d38824f1345be0b6847b1085aa7929da"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Yard Fallback Hotspot"
    password: "075664688s"
captive_portal:

status_led:
  pin:
    number: 2
    inverted: True

# Enable logging
debug:

logger:
  level: VERBOSE #Change the log level as required; during testing verbose is helpful. 
  baud_rate: 0

globals:
  # Za Tenda led strip globals
  - id: light_tenda_min_brightness  #min brighness for Tenda led
    type: float
    restore_value: no
    initial_value: '0.05'
  - id: light_tenda_max_brightness   #max brightness za Tenda LED
    type: float
    restore_value: no
    initial_value: '1.0'
  - id: dim_direction_tenda   #nasoka (pojachuva ili stishuva led lentata) dirrection going up or down
    type: int
    restore_value: no
    initial_value: '1'
  - id: TENDA_loop_stop # kraj chitanje taster na najmal ili najgolem brightness (stops loop if at max or min)
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: step #dimmer steps
    type: float
    restore_value: no
    initial_value: '0.02'

# sensor to report wifi RSSI

binary_sensor:
  # Tenda taster: button for LED strip
  - platform: gpio
    pin:
[... 328 lines cut ...]
[17:19:13][D][light:050]:   Brightness: 32%
[17:19:13][D][light:084]:   Transition length: 1.0s
[17:19:43][D][binary_sensor:036]: 'Tenda Taster': Sending state ON
[17:19:43][V][binary_sensor.automation:025]: START min=0 max=1000
[17:19:43][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[17:19:43][V][binary_sensor.automation:025]: START min=0 max=1000
[17:19:43][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[17:19:43][V][binary_sensor.automation:025]: START min=2000 max=4294967294
[17:19:43][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[17:19:43][V][component:199]: Component gpio.binary_sensor took a long time for an operation (0.06 s).
[17:19:43][V][component:200]: Components should block for at most 20-30ms.
[17:19:43][D][binary_sensor:036]: 'Tenda Taster': Sending state OFF
[17:19:43][V][binary_sensor.automation:054]: A i=1 min=0 max=1000
[17:19:43][V][binary_sensor.automation:062]: C i=1 min=500
[17:19:43][V][binary_sensor.automation:071]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[17:19:43][D][esp32.preferences:114]: Saving preferences to flash...
[17:19:43][V][binary_sensor.automation:101]: Multi Click: Hooray, multi click is valid. Triggering!
[17:19:43][D][main:131]: on_multi_click: pali ili gasi eden stisok
[17:19:43][D][light:035]: 'Tenda' Setting:
[17:19:43][D][light:046]:   State: OFF
[17:19:43][D][light:084]:   Transition length: 1.0s
[17:19:44][V][binary_sensor.automation:095]: Multi Click: You waited too long to PRESS.
[17:19:44][V][binary_sensor.automation:071]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[17:19:44][V][binary_sensor.automation:074]: Multi Click: Cooldown ended, matching is now enabled again.
[17:19:45][V][binary_sensor.automation:074]: Multi Click: Cooldown ended, matching is now enabled again.`

SmoothSilence, 2022-05-31:
Hi guys. I came across the same problem but I use the mult-click for a LED strip.
The idea was a single press turns it on or off,
double click goes to max-min brightness of the LED strip
and hold slowly dims up or down the strip till you let go.
Besides physical push buttons I use HA too. On both sides HA and wall buttons I have that error and it turns off the LED strip
Could you point me to where I messed up here ?

```
esphome:
  name: yard

esp32:
  board: esp32dev

# Enable Home Assistant API
api:
  #password: !secret api_password
  
ota:
  password: "d38824f1345be0b6847b1085aa7929da"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Yard Fallback Hotspot"
    password: "075664688s"
captive_portal:

status_led:
  pin:
    number: 2
    inverted: True

# Enable logging
debug:

logger:
  level: VERBOSE #Change the log level as required; during testing verbose is helpful. 
  baud_rate: 0

globals:
  # Za Tenda led strip globals
  - id: light_tenda_min_brightness  #min brighness for Tenda led
    type: float
    restore_value: no
    initial_value: '0.05'
  - id: light_tenda_max_brightness   #max brightness za Tenda LED
    type: float
    restore_value: no
    initial_value: '1.0'
  - id: dim_direction_tenda   #nasoka (pojachuva ili stishuva led lentata) dirrection going up or down
    type: int
    restore_value: no
    initial_value: '1'
  - id: TENDA_loop_stop # kraj chitanje taster na najmal ili najgolem brightness (stops loop if at max or min)
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: step #dimmer steps
    type: float
    restore_value: no
    initial_value: '0.02'

# sensor to report wifi RSSI

binary_sensor:
  # Tenda taster: button for LED strip
[... 321 lines cut ...]
[18:10:40][D][main:152]: pijachuva ili stishuva
[18:10:46][D][binary_sensor:036]: 'Tenda Taster': Sending state ON
[18:10:46][V][binary_sensor.automation:025]: START min=0 max=1000
[18:10:46][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[18:10:46][V][binary_sensor.automation:025]: START min=0 max=1000
[18:10:46][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[18:10:46][V][binary_sensor.automation:025]: START min=2000 max=4294967294
[18:10:46][V][binary_sensor.automation:026]: Multi Click: Starting multi click action!
[18:10:46][V][component:199]: Component gpio.binary_sensor took a long time for an operation (0.06 s).
[18:10:46][V][component:200]: Components should block for at most 20-30ms.
[18:10:46][D][binary_sensor:036]: 'Tenda Taster': Sending state OFF
[18:10:46][V][binary_sensor.automation:054]: A i=1 min=0 max=1000
[18:10:46][V][binary_sensor.automation:062]: C i=1 min=500
[18:10:46][V][binary_sensor.automation:071]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[18:10:46][V][binary_sensor.automation:101]: Multi Click: Hooray, multi click is valid. Triggering!
[18:10:46][D][main:130]: on_multi_click: pali ili gasi eden stisok
[18:10:46][D][light:035]: 'Tenda' Setting:
[18:10:46][D][light:046]:   State: OFF
[18:10:46][D][light:084]:   Transition length: 1.0s
[18:10:47][V][binary_sensor.automation:095]: Multi Click: You waited too long to PRESS.
[18:10:47][V][binary_sensor.automation:071]: Multi Click: Invalid length of press, starting cooldown of 1000 ms...
[18:10:47][V][binary_sensor.automation:074]: Multi Click: Cooldown ended, matching is now enabled again.
[18:10:48][V][binary_sensor.automation:074]: Multi Click: Cooldown ended, matching is now enabled again.
[18:11:27][D][esp32.preferences:114]: Saving preferences to flash...
`