esphome/issues #2826: SonOff Basic Rev1 not working any more with 2021.12.0

BenPru, 2021-12-14:
### The problem

After compiling 2021.12.0 and uploading the firmware the device is offline and the fallback wlan is not available.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

sonoff basic rev1

### Example YAML snippet

```yaml
substitutions:
  device_name: "terrassenbeleuchtung"
  upper_device_name: "TerrassenBeleuchtung"
  # Green LED (inverted)
  status_led: GPIO13
  # Button (inverted)
  button_state: GPIO0
  # Relay and Red LED
  button_switch: GPIO12

esphome:
  name: $device_name
  platform: ESP8266
  board: esp8285

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: !secret esphome_api_password

# Enable over the air updates
ota:
  safe_mode: true
  reboot_timeout: 10min
  num_attempts: 5

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: on
  use_address: 10.0.17.101

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name}"
    password: !secret fallback_password

# see: https://esphome.io/components/time.html
time:
  - platform: homeassistant
    id: homeassistant_time

# Enable Web server
web_server:
  port: 80
  auth:
    username: !secret esphome_web_username
    password: !secret esphome_web_password

text_sensor:
  - platform: version
    # entity_id: "${device_name}_version"
    name: "${device_name} - Version"
    icon: mdi:cube-outline

binary_sensor:
  - platform: status
    name: "${device_name} - Status"
    device_class: connectivity

  # toggle relay on/off
  - platform: gpio
    pin:
      number: ${button_state}
      mode: INPUT_PULLUP
      inverted: True
    id: "${device_name}_button_state"
    on_press:
      - switch.toggle: button_switch

sensor:
  - platform: wifi_signal
    name: "${device_name} - WLAN Signal"
    update_interval: 60s
    icon: mdi:wifi

  - platform: uptime
    name: "${device_name} - Uptime"
    update_interval: 60s
    icon: mdi:clock-outline

status_led:
  pin:
    number: ${status_led}
    inverted: True
  id: led_green

switch:
  - platform: gpio
    pin: ${button_switch}
    id: relay
  - platform: template
    name: "${upper_device_name}"
    icon: mdi:power
    optimistic: true
    id: button_switch
    turn_on_action:
      - switch.turn_on: relay
      # - output.turn_on: led_green
    turn_off_action:
      - switch.turn_off: relay
      # - output.turn_off: led_green
```

ajfriesen, 2022-01-04:
Having the same issue with some of my sonoff basics as well.
I have 2 broken sonoffs basics now.

One sonoff basic was able to upgrade though.

randybb, 2022-01-04:
Please provide logs...
@ajfriesen Just guessing that you got something like _ID 'button' conflicts with the name of an esphome integration, please use another ID name._?

ajfriesen, 2022-01-04:
@randybb 
I do not think this has anything to do with the problem.

I have another sonoff basic with exactly the same config except the name and it is just working fine, even after the update.

smoke007, 2022-01-22:
@BenPru,
See my post here https://github.com/esphome/issues/issues/2964#issuecomment-1019006987  I think this might help you.