esphome/issues #2828: Home Assistant - M5STACK Timer Cam - [esp32_camera:096]:   Setup Failed: ERROR

benoitdevries, 2021-12-15:
### The problem

Hello Experts !

The device is detected by esphome, and the initialisation works well.

Then I follow the configuration for this devices, [thanks to the Esphome documentation](https://esphome.io/components/esp32_camera.html#configuration-for-m5stack-camera).
The code is validated, so I upload it.
Then in the logs, I can read [esp32_camera:096]:   Setup Failed: ERROR

So I search the internet to find if someone has the same issue. No match.

I found [the documentation of the M5Starck Timer Cam](https://docs.m5stack.com/en/unit/timercam_x), and tried to adapt the pins.
I stil get the same error code.

Moreover, this entity is discovered by Home Assistant. I can add it to the Lovelace Dashboard.
Unfortunately, the image is not displayed.

Can you please help me to make it work in Home Assistant ?
Thanks in advance.

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.5 / supervisor-2021.12.1 / Home Assistant OS 7.0

### What platform are you using?

ESP8266

### Board

M5STACK Timer Camera X

### Example YAML snippet

```yaml
esphome:
  name: camera-m5
  platform: ESP32
  board: m5stack-timer-cam

# Enable logging
logger:
  level: VERBOSE
  
# Enable Home Assistant API
api:

ota:
  password: "mine"

wifi:
  ssid: "mine"
  password: "mine"

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Camera-M5 Fallback Hotspot"
    password: "vuAEzdfde0Lw"

esp32_camera:
  name: camera-m5
  external_clock:
    pin: GPIO23 
    frequency: 20MHz
  i2c_pins:
    sda: GPIO25
    scl: GPIO23
  data_pins: [GPIO32, GPIO35, GPIO34, GPIO5, GPIO39, GPIO18, GPIO36, GPIO19]
  vsync_pin: GPIO22
  href_pin: GPIO26
  pixel_clock_pin: GPIO21
  resolution: XGA
  jpeg_quality: 10
  max_framerate: 8 fps
  idle_framerate: 0 fps
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/camera-m5.yaml...
INFO Starting log output from camera-m5.local using esphome API
INFO Successfully connected to camera-m5.local
[09:06:49][I][app:102]: ESPHome version 2021.12.0 compiled on Dec 12 2021, 20:33:16
[09:06:49][C][wifi:488]: WiFi:
[09:06:49][C][wifi:350]:   Local MAC: 08:3A:F2:6A:4A:98
[09:06:49][C][wifi:351]:   SSID: [redacted]
[09:06:49][C][wifi:352]:   IP Address: 192.168.1.20
[09:06:49][C][wifi:354]:   BSSID: [redacted]
[09:06:49][C][wifi:355]:   Hostname: 'camera-m5'
[09:06:49][C][wifi:357]:   Signal strength: -53 dB ▂▄▆█
[09:06:49][V][wifi:359]:   Priority: 0.0
[09:06:49][C][wifi:361]:   Channel: 11
[09:06:49][C][wifi:362]:   Subnet: 255.255.255.0
[09:06:49][C][wifi:363]:   Gateway: 192.168.1.254
[09:06:49][C][wifi:364]:   DNS1: 192.168.1.254
[09:06:49][C][wifi:365]:   DNS2: 0.0.0.0
[09:06:49][C][logger:233]: Logger:
[09:06:49][C][logger:234]:   Level: VERBOSE
[09:06:49][C][logger:235]:   Log Baud Rate: 115200
[09:06:49][C][logger:236]:   Hardware UART: UART0
[09:06:49][C][esp32_camera:046]: ESP32 Camera:
[09:06:49][C][esp32_camera:047]:   Name: camera-m5
[09:06:49][C][esp32_camera:048]:   Internal: NO
[09:06:49][C][esp32_camera:050]:   Board Has PSRAM: YES
[09:06:49][C][esp32_camera:053]:   Data Pins: D0:32 D1:35 D2:34 D3:5 D4:39 D5:18 D6:36 D7:19
[09:06:49][C][esp32_camera:054]:   VSYNC Pin: 22
[09:06:49][C][esp32_camera:055]:   HREF Pin: 26
[09:06:49][C][esp32_camera:056]:   Pixel Clock Pin: 21
[09:06:49][C][esp32_camera:057]:   External Clock: Pin:23 Frequency:20000000
[09:06:49][C][esp32_camera:058]:   I2C Pins: SDA:25 SCL:23
[09:06:49][C][esp32_camera:059]:   Reset Pin: -1
[09:06:49][C][esp32_camera:083]:   Resolution: 1024x768 (XGA)
[09:06:49][E][esp32_camera:096]:   Setup Failed: ERROR
[09:06:49][C][ota:082]: Over-The-Air Updates:
[09:06:49][C][ota:083]:   Address: camera-m5.local:3232
[09:06:50][C][ota:086]:   Using Password.
[09:06:50][C][api:134]: API Server:
[09:06:50][C][api:135]:   Address: camera-m5.local:6053
[09:06:50][C][api:139]:   Using noise encryption: NO
[09:06:50][C][mdns:084]: mDNS:
[09:06:50][C][mdns:085]:   Hostname: camera-m5
[09:06:50][V][mdns:086]:   Services:
[09:06:50][V][mdns:088]:   - _esphomelib, _tcp, 6053
[09:06:50][V][mdns:090]:     TXT: version = 2021.12.0
[09:06:50][V][mdns:090]:     TXT: mac = 083af26a4a98
[09:06:50][V][mdns:090]:     TXT: platform = ESP32
[09:06:50][V][mdns:090]:     TXT: board = m5stack-timer-cam
```

### Additional information

[image]
[image]

markusg1234, 2021-12-15:
Hi,

I have the same camera and it works correctly. 
Looks like you have put GPIO23 as the external clock. It should be GPIO27.
Regards,

Markus

benoitdevries, 2021-12-16:
Hi Markus,

thanks a lot for replying !
I changed the External clock pin from GPIO23 to GPIO27, and I worked very fine !
Very fine, but just for 2 minutes. 
I paste here the entire log, from Compilation to the Error message after showing an image.

(In a second time, I will improve my code with your kind battery management proposal)

INFO Reading configuration /config/esphome/camera-m5.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing camera-m5 (board: m5stack-timer-cam; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <WiFi> 1.0
|-- <Update> 1.0
|-- <ESPmDNS> 1.0
Compiling /data/camera-m5/.pioenvs/camera-m5/src/esphome/components/api/api_connection.cpp.o
Compiling /data/camera-m5/.pioenvs/camera-m5/src/esphome/components/api/api_server.cpp.o
Compiling /data/camera-m5/.pioenvs/camera-m5/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/camera-m5/.pioenvs/camera-m5/src/esphome/core/application.cpp.o
Compiling /data/camera-m5/.pioenvs/camera-m5/src/esphome/core/util.cpp.o
Compiling /data/camera-m5/.pioenvs/camera-m5/src/main.cpp.o
Linking /data/camera-m5/.pioenvs/camera-m5/firmware.elf
RAM:   [=         ]  13.9% (used 45700 bytes from 327680 bytes)
Flash: [=====     ]  52.9% (used 971162 bytes from 1835008 bytes)
Building /data/camera-m5/.pioenvs/camera-m5/firmware.bin
========================= [SUCCESS] Took 81.25 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of camera-m5.local
INFO  -> 192.168.1.20
INFO Uploading /data/camera-m5/.pioenvs/camera-m5/firmware.bin (971264 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from camera-m5.local using esphome API
**WARNING Can't connect to ESPHome API for camera-m5.local: Timeout while connecting to ('192.168.1.20', 6053)**
INFO Trying to reconnect to camera-m5.local in the background
INFO Successfully connected to camera-m5.local
[11:36:48][I][app:102]: ESPHome version 2021.12.1 compiled on Dec 16 2021, 11:34:46
[11:36:48][C][wifi:488]: WiFi:
[11:36:48][C][wifi:350]:   Local MAC: 08:3A:F2:6A:4A:98
[11:36:48][C][wifi:351]:   SSID: [redacted]
[11:36:48][C][wifi:352]:   IP Address: 192.168.1.20
[11:36:48][C][wifi:354]:   BSSID: [redacted]
[11:36:48][C][wifi:355]:   Hostname: 'camera-m5'
[11:36:48][C][wifi:357]:   Signal strength: -67 dB ▂▄▆█
[11:36:48][V][wifi:359]:   Priority: 0.0
[11:36:48][C][wifi:361]:   Channel: 11
[11:36:48][C][wifi:362]:   Subnet: 255.255.255.0
[11:36:48][C][wifi:363]:   Gateway: 192.168.1.254
[11:36:48][C][wifi:364]:   DNS1: 192.168.1.254
[11:36:48][C][wifi:365]:   DNS2: 0.0.0.0
[11:36:48][C][logger:233]: Logger:
[11:36:48][C][logger:234]:   Level: VERBOSE
[11:36:48][C][logger:235]:   Log Baud Rate: 115200
[11:36:48][C][logger:236]:   Hardware UART: UART0
[11:36:48][C][esp32_camera:046]: ESP32 Camera:
[11:36:48][C][esp32_camera:047]:   Name: camera-m5
[11:36:48][C][esp32_camera:048]:   Internal: NO
[11:36:48][C][esp32_camera:050]:   Board Has PSRAM: YES
[11:36:48][C][esp32_camera:053]:   Data Pins: D0:32 D1:35 D2:34 D3:5 D4:39 D5:18 D6:36 D7:19
[11:36:48][C][esp32_camera:054]:   VSYNC Pin: 22
[11:36:48][C][esp32_camera:055]:   HREF Pin: 26
[11:36:48][C][esp32_camera:056]:   Pixel Clock Pin: 21
[11:36:48][C][esp32_camera:057]:   External Clock: Pin:27 Frequency:20000000
[... 25 lines cut ...]
[11:37:18][D][esp32_camera:162]: Got Image: len=32074
[11:37:28][D][esp32_camera:162]: Got Image: len=32430
[11:37:38][D][esp32_camera:162]: Got Image: len=32201
[11:37:48][D][esp32_camera:162]: Got Image: len=32326
[11:37:58][D][esp32_camera:162]: Got Image: len=31312
[11:38:08][D][esp32_camera:162]: Got Image: len=31766
[11:38:18][D][esp32_camera:162]: Got Image: len=31986
[11:38:28][D][esp32_camera:162]: Got Image: len=37512
[11:38:38][D][esp32_camera:162]: Got Image: len=36812
[11:38:48][D][esp32_camera:162]: Got Image: len=37037
[11:38:58][D][esp32_camera:162]: Got Image: len=38480
[11:39:00][D][esp32_camera:162]: Got Image: len=39233
[11:39:01][D][esp32_camera:162]: Got Image: len=39229
[11:39:02][D][esp32_camera:162]: Got Image: len=39291
[11:39:03][D][esp32_camera:162]: Got Image: len=39420
[11:39:04][D][esp32_camera:162]: Got Image: len=39189
[11:39:05][D][esp32_camera:162]: Got Image: len=39101
[11:39:06][D][esp32_camera:162]: Got Image: len=39388
[11:39:07][D][esp32_camera:162]: Got Image: len=39266
[11:39:08][D][esp32_camera:162]: Got Image: len=39342
**INFO camera-m5.local: Ping timed out!
INFO Disconnected from ESPHome API for camera-m5.local
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for camera-m5.local: Error connecting to ('192.168.1.20', 6053): [Errno 111] Connect call failed ('192.168.1.20', 6053)
INFO Trying to reconnect to camera-m5.local in the background**

markusg1234, 2021-12-16:
Hi,
I did some tests with your settings and found when I set the resolution to XGA it caused the ESP32 to reboot with errors.

I normally have mine set to SXGA. Not sure if XGA is supported on the camera or a bug. 

Can you try SXGA.

Regards,
Markus.

benoitdevries, 2021-12-17:
Works damn well !
Thanks a lot @markusg1234  !!!

closed 2021-12-17 by benoitdevries

markusg1234, 2021-12-18:
forgot to mention if you would like to monitor the battery voltage add.

sensor:
  - platform: adc
    pin: GPIO38
    name: "Battery Voltage"
    filters:
      - multiply: 3.3

tiimsvk, 2021-12-24:
Hi me this problem pops up too.
Modul: esp32 cam ai thinker
Esphome version:2021.12.1
Home Assistant version: 2021.12.5

Yaml config:
`# Author by airijia.com
# ESP32-CAM
# Lovelace camera card https://github.com/custom-cards/camera-card
# Video tutorial https://youtu.be/-towjof_ixU
substitutions:
  # Modify variables based on your settings
  hostname: 'esp32-cam1'
  ssid: 'DreamLoveHome'
  password: 'ZelenyDom2'
esphome:
  name: $hostname
  platform: ESP32
  board: esp32dev
wifi:
  ssid: $ssid
  password: $password
  fast_connect: True
api:
  reboot_timeout: 0s
ota:
logger:

# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: bedroom
  # ... Frame Settings
  resolution: 1024x768
  jpeg_quality: 10
  max_framerate: 20fps
  idle_framerate: 1fps
  vertical_flip: false
  horizontal_mirror: false

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: $hostname light
`
Log:

INFO Reading configuration /config/esphome/esp32-cam1.yaml...
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from esp32-cam1.local using esphome API
INFO Successfully connected to esp32-cam1.local
[07:53:16][I][app:102]: ESPHome version 2021.12.1 compiled on Dec 24 2021, 07:51:21
[07:53:16][C][wifi:488]: WiFi:
[07:53:16][C][wifi:350]:   Local MAC: C8:C9:A3:F9:50:D8
[07:53:16][C][wifi:351]:   SSID: [redacted]
[07:53:16][C][wifi:352]:   IP Address: 192.168.31.12
[07:53:16][C][wifi:354]:   BSSID: [redacted]
[07:53:16][C][wifi:355]:   Hostname: 'esp32-cam1'
[07:53:16][C][wifi:357]:   Signal strength: -40 dB ▂▄▆█
[07:53:16][C][wifi:361]:   Channel: 1
[07:53:16][C][wifi:362]:   Subnet: 255.255.255.0
[07:53:16][C][wifi:363]:   Gateway: 192.168.31.1
[07:53:16][C][wifi:364]:   DNS1: 192.168.31.1
[07:53:16][C][wifi:365]:   DNS2: 0.0.0.0
[07:53:16][C][logger:233]: Logger:
[07:53:16][C][logger:234]:   Level: DEBUG
[07:53:16][C][logger:235]:   Log Baud Rate: 115200
[07:53:16][C][logger:236]:   Hardware UART: UART0
[07:53:16][C][gpio.output:010]: GPIO Binary Output:
[07:53:16][C][gpio.output:011]:   Pin: GPIO4
[07:53:16][C][light:098]: Light 'esp32-cam1 light'
[07:53:16][C][esp32_camera:046]: ESP32 Camera:
[07:53:16][C][esp32_camera:047]:   Name: bedroom
[07:53:16][C][esp32_camera:048]:   Internal: NO
[07:53:16][C][esp32_camera:050]:   Board Has PSRAM: YES
[07:53:16][C][esp32_camera:053]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[07:53:16][C][esp32_camera:054]:   VSYNC Pin: 25
[07:53:16][C][esp32_camera:055]:   HREF Pin: 23
[07:53:16][C][esp32_camera:056]:   Pixel Clock Pin: 22
[07:53:16][C][esp32_camera:057]:   External Clock: Pin:0 Frequency:20000000
[07:53:16][C][esp32_camera:058]:   I2C Pins: SDA:26 SCL:27
[07:53:16][C][esp32_camera:059]:   Reset Pin: -1
[07:53:16][C][esp32_camera:086]:   Resolution: 1280x1024 (SXGA)
[07:53:16][E][esp32_camera:096]:   Setup Failed: ERROR
[07:53:16][C][ota:082]: Over-The-Air Updates:
[07:53:16][C][ota:083]:   Address: esp32-cam1.local:3232
[07:53:16][C][api:134]: API Server:
[07:53:16][C][api:135]:   Address: esp32-cam1.local:6053
[07:53:16][C][api:139]:   Using noise encryption: NO
[07:53:16][C][mdns:084]: mDNS:
[07:53:16][C][mdns:085]:   Hostname: esp32-cam1

Thanks for answer

markusg1234, 2021-12-24:
Hi, 
try setting the board to esp32cam instead of esp32dev to see if that makes a difference.
Regards,
Markusg

markusg1234, 2021-12-24:
Hi, your settings seem ok. Normally I set the board to esp32cam but that shouldn't stop it working.

I would suggest checking that the camera locking lever is fully tight as this can cause setup failed errors.
Regards,
Markus

tiimsvk, 2021-12-24:
I do not think so.  I have two cameras.  They stopped working after this monthly update

markusg1234, 2021-12-24:
Fair enough.
Running the same version with your settings seems to work here with a esp32cam. 
Esphome version:2021.12.1
Home Assistant version: 2021.12.5

kobold81, 2021-12-28:
With AI-Tinker:

With the December release, camera setup would fail if I used "board: esp-wrover-kit". When changed to "board: esp32cam" the camera works. 
If the following is added tothe esp32_camera component, setup will also fail "setup_priority: -200"

Esphome version:2021.12.1
Home Assistant version: 2021.12.5

kc9rqi, 2022-02-15:
@tiimsvk, did you ever get this working?  I'm gettin the same error.

kc9rqi, 2022-02-15:
Had the same issues, was able to get mine going using the following YAML

esphome:
  name: device name
  platform: ESP32
  board: esp32cam
  arduino_version: 1.0.6 # updated on feb /2022 using Home Assistant OS 7.4 core-2022.2.6