esphome/issues #2831: Error Using Custom Global Type

RebbePod, 2021-12-15:
### The problem

When including a header file with a custom class, the global type can't be set to that class.

This is the custom class I currently have in my header file:
```cpp
    class CustomTime : public time::RealTimeClock {
    public:
    void setup() override;
    void update() override;
    void set_epoch_time(uint32_t epoch) { this->synchronize_epoch_(epoch); }

    protected:
    time_t time_;
    };
```

The error I get is shown below.

### Which version of ESPHome has the issue?

2021.12.1 

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

globals:

### Example YAML snippet

```yaml
globals:
  - id: custom_global
    type: CustomTime
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp:125:27: error: 'CustomTime' was not declared in this scope
 globals::GlobalsComponent<CustomTime> *custom_global;
                           ^
src/main.cpp:125:37: error: template argument 1 is invalid
 globals::GlobalsComponent<CustomTime> *custom_global;
```

OttoWinter, 2022-01-18:
How are you including the header file? Did you declare it in esphome->includes?

RebbePod, 2022-01-18:
> How are you including the header file? Did you declare it in esphome->includes?

It was added in the YAML includes. I have other functions that work just fine, it's only in the globals that it wasn't recognizing it.

```yaml
esphome:
  includes: shared/header.h
```

rpatel3001, 2022-01-24:
Ran into the same issue today. The problem seems to be that in the generated src/main.cpp file the globals are inserted before the includes.

jesserockz, 2022-01-24:
> Ran into the same issue today. The problem seems to be that in the generated src/main.cpp file the globals are inserted before the includes.

I was going to reply the same thing, most of the time the custom headers want to access something from main.cpp, not the other way around and I suspect this is why it is in this order.

rpatel3001, 2022-01-24:
A possible solution is to add a custom_globals or post_header_globals key or something along those lines to control placement. I had this issue when developing https://github.com/rpatel3001/BleDistance. I suspect it's probably more appropriate in my case to turn anything that needs this functionality into a proper custom sensor instead of hacking away with lambdas and adding complexity to the config file.

rpatel3001, 2022-01-27:
My workaround was to add a global vector of the custom object type and add global functions for adding and getting members from the vector. This requires adding some sort of identifier to the constructor call for your objects that the getter can use to find the right object in the vector. An example of this approach is in the repo I posted above.

leoshusar, 2022-10-29:
I have the same problem. I wanted to create custom enum class:
```cpp
enum class CalibrationState { none, notify, ok, error };
```
to use in `globals` and then I realized that I have no way to include it, because it's included at the very end, like this:
```cpp
globals::GlobalsComponent<CalibrationState> *calibration_state;
...
#include "calibration_state.h";
```

paco3346, 2023-03-03:
I found this in the NabuCasa forums and it's working great

_(rewritten to reference the OP in this github issue)_

>  I was playing around and found another maybe simpler way of accomplishing this by setting up your class variable `custom_global` as global instead of localized under the custom component instantiation.
>
> What you do is add this line after your class definition in your header file:
> ```C++
> # existing code here
> CustomTime* custom_global;
> ```
>
> Then, rather than creating a global variable, you instantiate the existing definition in a custom_component.
>
> ```yaml
> custom_component:
>  - lambda: |-
>      custom_global = new CustomTime();
>      return {custom_global};
> ```
>
> So what that means then, is in your other lambda functions you are able to access your class functions using:
> ```C++
> custom_global->set_epoch_time();
> ```
>
> The main downside is that it's now more difficult to reference the variable from yaml references but that's probably not something you'd be doing anyway.
>

Credit goes to @Dilbert66
https://community.home-assistant.io/t/calling-custom-functions/358121/9

Dilbert66, 2023-03-03:
> > The main downside is that it's now more difficult to reference the variable from yaml references but that's probably not something you'd be doing anyway.

You can add an "id" tag  to the custom component and you can then access it that way if needed.

paco3346, 2023-03-03:
> > > The main downside is that it's now more difficult to reference the variable from yaml references but that's probably not something you'd be doing anyway.
> 
> You can add an "id" tag to the custom component and you can then access it that way if needed.

Oh good point. I forgot that the global we're setting in the header file is the same object we're registering in the component.

yanncharlou, 2024-08-30:
A fix or another workaround would be great as custom_component is deprecated : https://esphome.io/custom/custom_component.html#generic-custom-component

Dilbert66, 2024-08-30:
What I do for an external component is to create a global pointer before the class and within the class instantation, I assign the class to the pointer ie classPtr=this. In the header file I assign an extern for that ptr. Since the ptr exists in the global scope and is assigned to the created class, it will be easily accessible.  This is ok since the class is only ever instantiated once.  

FYI in the customclass.cpp
```
CustomClass*  customClassPtr;

CustomClass::CustomClass (....) {

   customClassPtr=this;

}
 ```
and in the customclass.h
```
Class CustomClass(...) {
}

extern CustomClass*  customClassPtr;

```

DragRedSim, 2026-10-01:
I came across this while troubleshooting my own install, which uses an enum type of which there's a couple of instances in globals.
If the include string is just the path to the header file, ESPHome will include it late in the file. If, however, the include string is wrapped in square brackets, it will be included early, before the globals are defined.
```
esphome:
  includes: shared/header.h
```
becomes
```
esphome:
  includes: <shared/header.h>
```

ssieb, 2026-10-01:
Those are actually "angle" brackets and that's documented, although the description there doesn't explain why you would want to use it.
https://esphome.io/components/esphome/#includes

closed 2026-10-01 by ssieb