esphome/issues #2832: SCD41 (Sensirion CO2, Temperature, and Humidity Sensor) periodically stops reporting measurements

Jsakkos, 2021-12-15:
### The problem

The sensor will occasionally become unresponsive and send the message "Data not ready yet". Power cycling the device has no effect.

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

Wemos D1 mini 4mb

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  name: air-quality-1
  platform: ESP8266
  board: d1_mini
i2c:
  sda: 4
  scl: 5
  scan: true
  id: bus_a
sensor:
  - platform: scd4x
    co2:
      name: "CO2"
      id: "co2"
    temperature:
      name: "Temperature"
      id: "temperature"
    humidity:
      name: "Humidity"
      id: "humidity"
    update_interval: 30s
```

### Anything in the logs that might be useful for us?

```txt
[17:51:01][W][scd4x:175]: Data not ready yet!
```

### Additional information

Seems similar to this issue with the SCD30: https://github.com/esphome/issues/issues/987

sjtrny, 2022-01-18:
Any chance you can test with an ESP32 instead?

Jsakkos, 2022-01-18:
@sjtrny I unfortunately don't have any ESP32 devices. One thing I will add is that I have 4 of these setup, and only one has this issue. I've tried swapping out the SCD41 itself, reflashing the device OTA, and swapping out the ESP8266 itself. The strange thing is that after a reflash, it will work properly for ~ 30 mins before displaying this message.

agrueing, 2022-02-03:
Can confirm on an ESP32

tbailey1712, 2022-02-14:
On an esp8266 nodemcu, having the same issue.  The sensor initializes and displays its serial number.  The response is "data not ready yet" each time it is polled.  

I don't believe the board is defective as it has did output data one time for me.  Sensor is an SCD41 from Adafruit.

nicolasff, 2022-02-22:
This looks very similar to what I'm seeing, I've just asked about it [in the Adafruit forums](https://forums.adafruit.com/viewtopic.php?f=19&t=188704&p=913299). Mine isn't using ESPHome though, it's connected to an [ESP32-S2](https://unexpectedmaker.com/feathers2-neo) running CircuitPython.

When I detect that it's been too long without data (> 5 min), I stop the periodic measurements, re-initialize the sensor, and start measuring again:

```python
if time.monotonic() - last_co2_read_time > 300:
    print('Stopping periodic measurements')
    scd4x.stop_periodic_measurement()
    time.sleep(1)
    print('Reinitializing CO2 sensor')
    scd4x.reinit()
    time.sleep(1)
    print('Starting periodic measurements again')
    scd4x.start_periodic_measurement()
```

I don't know enough about ESPHome to tell, but is it possible to program in this kind of workaround with the YAML config? Ideally the best would be to avoid such hacks and have the sensor keep working.

This workaround is usually enough to receive some data again, but it usually doesn't last long and soon enough I'm having the issue once more. In some cases calling `reinit` doesn't fix it, so I have some code to detect this and restart the whole board; after a reboot I've always seen it produce data again.

I've tried using the sensor both in its regular periodic measurement mode and its low-power mode, and this has made no difference. In both cases I eventually end up stuck on `data_ready == False` forever. It was said above that it can take ~30 minutes for it to get stuck; for me it's completely unpredictable. Sometimes it takes just 5 minutes, other times 12+ hours.

Since I can use CircuitPython to send commands to the sensor, is there anything I could try to help find what's causing this issue?

V4l3n10, 2022-02-22:
@Jsakkos I am trying to do the same as you are. I will be using an ESP32-S2 Feather. Would you share your code please. Thanks

Freyert, 2022-08-17:
Connecting this to a forum post:

* [Pimoroni SCD41 Communication Failed on ESP32](https://community.home-assistant.io/t/pimoroni-scd41-communication-failed-on-esp32/400569)

nicolasff, 2022-08-17:
I should have updated this GitHub issue after I updated my Adafruit forum post. I am no longer encountering this issue, at all.

I had originally connected my SCD-41 in a _chain_ of sensors all connected with STEMMA QT/Qwiic cables with one end being a FeatherS2 Neo  and the SCD-41 being the 3rd device in that chain. It went like this:

FeatherS2 Neo → PMSA0003I → MCP9808 → SCD-41 → multiplexer → (2 × SSD1327)

The problem was likely with the amount of power that could be carried by the STEMMA QT port on the Feather to this entire chain. I changed the first link of this chain from a STEMMA QT cable to 4 Dupont cables connected to the board's (5V, GND, SCL, SDA) and going to the PMSA0003I sensor ([photo of the device after making this change](https://imgur.com/SM4evVS)). I left the rest of the chain as it was, and I never encountered this issue again.

So I would recommend making sure that you are supplying **enough power** to the SCD-41 sensor. Mine used to get into this "no data" state several times a day, and since making this change it hasn't happened even once.

Freyert, 2022-08-17:
@nicolasff thanks for the update! I moved the 3v3 to the 5v pin and it started working again. 🙏 thanks so much!

Freyert, 2022-08-20:
I've tried setting it to lower and lower power modes. Seems to do better that way.

The data sheet seems to indicate it needs 11-13 mA when running with periodic measurements, and 115-137 mA for high performance at 5V.

I found this ESPHome documentation that indicates the default "drive strength" is about 20 mA.
https://esphome.io/guides/configuration-types.html#pin-schema

[image]

svde, 2023-11-27:
Managed to fix this with the [factory_reset Action](https://esphome.io/components/sensor/scd4x.html#factory-reset-action).

martinanton, 2023-12-13:
I can't find my scd41.
could you please post your yaml? 

Communication failed! Is the sensor connected?

`[17:23:32][I][i2c.arduino:069]: Results from i2c bus scan:
[17:23:32][I][i2c.arduino:071]: Found no i2c devices!
[17:23:32][C][scd4x:094]: scd4x:
[17:23:32][C][scd4x:095]:   Address: 0x62
[17:23:32][W][scd4x:099]: Communication failed! Is the sensor connected?
[17:23:32][C][scd4x:112]:   Automatic self calibration: ON
[17:23:32][C][scd4x:121]:   Ambient pressure compensation disabled
[17:23:32][C][scd4x:122]:   Altitude compensation: 0m
[17:23:32][C][scd4x:127]:   Measurement mode: periodic (5s)
[17:23:32][C][scd4x:139]:   Temperature offset: 4.00 °C
[17:23:32][C][scd4x:140]:   Update Interval: 30.0s
[17:23:32][C][scd4x:141]:   CO2 'CO2wohnen'
[17:23:32][C][scd4x:141]:     Device Class: 'carbon_dioxide'
[17:23:32][C][scd4x:141]:     State Class: 'measurement'
[17:23:32][C][scd4x:141]:     Unit of Measurement: 'ppm'
[17:23:32][C][scd4x:141]:     Accuracy Decimals: 0
[17:23:32][C][scd4x:141]:     Icon: 'mdi:molecule-co2'
[17:23:32][C][scd4x:142]:   Temperature 'tempBaumhaus'
[17:23:32][C][scd4x:142]:     Device Class: 'temperature'
[17:23:32][C][scd4x:142]:     State Class: 'measurement'
[17:23:32][C][scd4x:142]:     Unit of Measurement: '°C'
[17:23:32][C][scd4x:142]:     Accuracy Decimals: 2
[17:23:32][C][scd4x:142]:     Icon: 'mdi:thermometer'
[17:23:32][C][scd4x:143]:   Humidity 'humidityBaumhaus'
[17:23:32][C][scd4x:143]:     Device Class: 'humidity'
[17:23:32][C][scd4x:143]:     State Class: 'measurement'
[17:23:32][C][scd4x:143]:     Unit of Measurement: '%'
[17:23:32][C][captive_portal:088]: Captive Portal:`

thanxs to all! ;-)

R1DEN, 2024-01-08:
Every disturbance in power delivery results in the sensor not working, until I restart the ESP. After restart it can work for more than a day without any issues.

I've just set up everything on a breadboard, the module is just hanging. And every time I move it - it's not reporting after.
Maybe an auto re-init should be implemented?

pgwipeout, 2025-01-01:
I don't know if it is the source of any of the problems described here, but I recently received a counterfeit SCD41 from the big A. It behaved strangely on the bus before finally dying. Look for the mandatory laser etched QR code on the canister, as described in the data sheet section 4.6 "Traceability and Identification".

waarsenburg, 2025-01-19:
I had the same problem. What finally worked for me was putting a 220uF capacitor across the 3V3 supply. It worked ever since.

Thynix, 2025-01-22:
I also notice

```
[W][component:157]: Component scd4x.sensor set Warning flag: unspecified
```

just before the first `Data not ready yet!`

@waarsenburg How did you arrive at the 220uF value? I don't see a difference in behavior with 100uF, and I could go up to 1000uF, but I didn't want to flail without reason. I have a separate 3.3v supply running off the 5v, so I don't think it's a problem with insufficient amperage coming from the Huzzah ESP8266.

I wonder if there are different classes of problems here? It sounds like for some it works after restart until there's a power disturbance, and for others (myself included) it immediately fails to give measurements. Mine is an SCD40 from Adafruit; are the behavior differences explained by vendor/integrator differences?

waarsenburg, 2025-01-23:
The only reason I used 220uF because it was the capacitor lying closest to me on the workbench. No theory behind it... What also seems to cause reboots, is bad Wifi. I have a display defined in the YAML code, see https://www.waarsenburg.com/20250123_095356.jpg
Sometimes I see the graph restarted some time ago. The only reason I can imagine is bad Wifi (which is absolutely true in the room the sensor resides). But since placing the capacitor, its not rebooting anymore.