esphome/issues #2833: monochromatic light OFF state 100% brightness

obrador, 2021-12-15:
### The problem

Hi, I don't know if that's an issue or if it's the normal behaviour. I've set up a monochromatic light with a PWM output, it works fine, but when I set brightness to 0%, the state turns automatically to "OFF" and the brightness become 100%. Light is off, it works in fact, but I would like brightness to be 0%. Is there a way to achieve that?

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

No HA. I use OpenHab

### What platform are you using?

ESP8266

### Board

D1

### Component causing the issue

monochromatic light

### Example YAML snippet

```yaml
light:
  - platform: monochromatic
    name: "White LED strip"
    output: wled_strip

output:
  - platform: esp8266_pwm
    id: wled_strip
    pin: GPIO13
```

### Anything in the logs that might be useful for us?

```txt
No errors
```

oxan, 2021-12-16:
No. Why do you want that?

obrador, 2021-12-16:
Openhab reads only brightness as the state of the light, so when light is OFF, it shows as ON because brightness is 100%. But I'm sure I can solve this inconvenient through Openhab settings.

Thanks a lot.

oxan, 2021-12-16:
How does OpenHAB connect to ESPHome? through MQTT?

I think it'd be sensible to change the MQTT interface to report 0% brightness when a light is off.

obrador, 2021-12-16:
Hi, yeah through MQTT. Openhab can do a transformation on the input MQTT message, so I can map OFF state to 0% brightness. Surely will work.

Thanks!

oxan, 2021-12-20:
Can you try esphome/esphome#2934?

ahpohl, 2023-03-18:
The OpenHAB MQTT dimmer channel doesn't have a power switch so I like to control the lamp power state purely with the brightness setting. Of course one could add another power channel to the thing, but that would add another control just for one lamp. Here is what I would like to achieve:

brightness = 0 --> state: OFF
brightness > 0 --> state ON

[image]

The patch below implements exactly this behavior and fixes the 100 % brightness issue:
```
diff --git a/esphome/components/light/light_call.cpp b/esphome/components/light/light_call.cpp
index fb4e45b3..0b1af889 100644
--- a/esphome/components/light/light_call.cpp
+++ b/esphome/components/light/light_call.cpp
@@ -231,7 +231,9 @@ LightColorValues LightCall::validate_() {
   // Turn off when brightness is set to zero, and reset brightness (so that it has nonzero brightness when turned on).
   if (this->brightness_.has_value() && *this->brightness_ == 0.0f) {
     this->state_ = optional<float>(false);
-    this->brightness_ = optional<float>(1.0f);
+    //this->brightness_ = optional<float>(1.0f);
+  } else if (this->brightness_.has_value()) {
+       this->state_ = optional<float>(true);
   }
 
   // Set color brightness to 100% if currently zero and a color is set.
```

The PR https://github.com/esphome/esphome/pull/2934 @oxan is already closed so I commented here instead.

ahpohl, 2023-03-18:
Created a new PR https://github.com/esphome/esphome/pull/4586 which implements the changes posted above.