esphome/issues #2835: Partition light generates 2 entities, one being non responsive

jarrettv, 2021-12-16:
### The problem

Problem was discussed on discord thread here: 920746883125743617

I'm using 15 of 16 ledc channels on an esp32 for my stairs. I have 15 steps. I want to treat the stairs as one addressable monochromatic light.

First attempt:
```  - platform: partition
    id: step_lights
    name: ${display_name}
    #color_mode: brightness
    effects:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
    segments:
      - single_light_id: step0
      - single_light_id: step1
      - single_light_id: step2
      - single_light_id: step3
      - single_light_id: step4
      - single_light_id: step5
      - single_light_id: step6
      - single_light_id: step7
      - single_light_id: step8
      - single_light_id: step9
      - single_light_id: step10
      - single_light_id: step11
      - single_light_id: step12
      - single_light_id: step13
      - single_light_id: step14```

This resulted in two entities, neither of which would properly control the lights. Both entities were non-responsive.

My next attempt:
```  - platform: partition
    id: step_lights
    name: ${display_name}
    effects:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
    segments:
      - single_light_id: step_0_1_2
      - single_light_id: step_3_4_5
      - single_light_id: step_6_7_8
      - single_light_id: step_9_10_11
      - single_light_id: step_12_13_14```
This also resulted in two entities and the "step_lights" entity worked fine. The additional entity "esphome_stairslight" entity did not work.

I'm not sure why two entities are being created. That seems like a bug.
My full config source is available here: https://github.com/jarrettv/esphome-config/blob/master/stairs.yaml

### Which version of ESPHome has the issue?

2021.12

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12

### What platform are you using?

ESP32

### Board

NodeMCU

### Component causing the issue

light partition

### Example YAML snippet

```yaml
See above
```

oli-f, 2022-01-04:
I have had the same issue (one correct entity and one non-functional) and I think I have tracked it down a little bit:

```yaml
  - platform: partition
    name: "Beacon Light"
    segments:
      - single_light_id: light_red
```
produces the following code:
```cpp
  light_addressablelightwrapper = new light::AddressableLightWrapper(light_red);
  light_lightstate = new light::LightState("", light_addressablelightwrapper);
  light_lightstate->set_component_source("partition.light");
  App.register_component(light_lightstate); // <-- this line produces the issue
  App.register_light(light_lightstate);
```

In the WebUI for each segment an entity is created without a name and in Home Assistant a single entity (appart from the correct one) named "light.esphome_beacon_lightlight" appears, so i guess the reason for Home Assistant only showing one is some fallback due to the missing name.

If I remove `cg.add(cg.App.register_light(light_state))` from [https://github.com/esphome/esphome/blob/dev/esphome/components/partition/light.py#L107](url) the config works as intended. I am not sure if registering the AddressableLightWrapper as a light is necessary for other components, but from my understanding the registering is not consistent with addressable segments.

ssieb, 2022-01-28:
I think this is fixed with https://github.com/esphome/esphome/pull/3092