esphome/issues #2836: UART doc page description of UART is misleading

glyndon, 2021-12-17:
### The problem

The document page describing UART may be misleading/inaccurate in its description of what a UART is/does.
(I can't edit the page, and not accustomed to creating pull requests, so offering this info here)

UART is a serial data transciever; it's a device, not a bus nor a protocol.  The acronym seems to have fallen into common vernacular as something more than it is, but its heritage is much deeper than that usage. 
It stands for Universal Asynchronous Receiver Transmitter - just a device that emits or receives bits serially. The serial format for single characters on the wire is configured by the settings for the number of start bits, stop bits, and parity bit polarity.
The physical link is not really a bus because it is limited to just two devices that are wired to send/receive with each other - there can't be more than the 2.

The docs might say something like: "The UART provides a serial data link to another device which has a UART configured to the same data format and speed." (or more simply "between two devices")

(RS-232, another very old datalink standard, refers to the voltages of the signals between two UARTs, and so many years ago the term 'RS-232' used to be commonly when referring to such a UART-link combo.)

https://en.wikipedia.org/wiki/Universal_asynchronous_receiver-transmitter

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

UART

oxan, 2021-12-17:
While you're technically correct, I think it's helpful for the docs to stick to the common usage of the term even if it's technically not fully correct, to not unnecessarily confuse people that aren't familiar with it. The distinction is seldom relevant.

glyndon, 2021-12-17:
Which may be easier for everyone, with the unfortunate side-effect of further deteriorating their knowledge about technology.
To mis-know something is normal, to propagate that error erodes the collective skill of everyone.
Pebbles in an avalanche.

oxan, 2021-12-17:
Where you see deterioration of knowledge, I just see language and terminology changing. It happens all the time, e.g. tty used to primarily refer to a physical teleprinter, but nowadays it's primarily used to refer to (emulated) terminals. But if you value it, I'm happy to consider a PR to use the correct terminology, as long as the documentation remains clear for people not accustomed to the terminology.

glyndon, 2021-12-17:
Thanks for looking into this.
Yes, TTY is a good example of another one of those 'antique' terms still in use, and it's true that most of today's IT people probably have no idea where it came from. 
Consider, however, that in other areas such as automotive hobby pursuits, respect for and knowledge of the past technologies is widely valued. Sure, plenty of hobbyists just want their car to go faster, but the ones who know how it got to where it is may be in a better position to take it further. History is very important, and accurate history even moreso.

glyndon, 2021-12-18:
Here's an example of why I'm passionate about not losing legacy knowledge and understanding about how and why things worked and where they came from.
https://www.youtube.com/watch?v=B1J2RMorJXM

closed 2022-05-02 by github-actions[bot]