esphome/issues #2837: Unexpected error: Error while retrieving firmware file

hanzch, 2021-12-17:
### The problem

Hello, everyone, I have tried for three days but I can’t flash the firmware for ESP32. The display error is as follows (it seems that the GITHUB link is invalid?) I look forward to everyone taking the time to reply, thanks! ~

[image]

> Using 'COM4' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Connecting....

>Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 30:83:98:00:CB:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
>Unexpected error: Error while retrieving firmware file 'https://raw.githubusercontent.com/espressif/arduino-esp32/1.0.4/tools/sdk/bin/bootloader_dio_40m.bin': HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/arduino-esp32/1.0.4/tools/sdk/bin/bootloader_dio_40m.bin (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x000002800325F4C0>: Failed to establish a new connection: [Errno 11004] getaddrinfo failed'))

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

oxan, 2021-12-17:
> it seems that the github link is invalid

The link itself is fine, but DNS isn't working correctly on your system.

hanzch, 2021-12-17:
> The link itself is fine, but DNS isn't working correctly on your system.

> https://raw.githubusercontent.com/espressif/arduino-esp32/1.0.4/tools/sdk/bin/bootloader_dio_40m.bin

Hello, I have no problem accessing other github resources here, but when I download this resource, the save fails. Are there any problems with your test?

oxan, 2021-12-17:
> Are there any problems with your test?

No, it works fine. The error message you get already says `getaddrinfo` fails, indicating DNS failure. Check your Docker networking.

hanzch, 2021-12-17:
> No, it works fine. The error message you get already says `getaddrinfo` fails, indicating DNS failure. Check your Docker networking.

Thanks, I changed the DNS on another computer and it can be downloaded normally. Thank you for your help.

closed 2021-12-17 by hanzch