esphome/issues #2838: ESP32 S2 config file

homonto, 2021-12-17:
### The problem

I have bunch of ESP32 S2 but I am not able to get them working with ESPHome - probably I don't know how to configure the yaml file. I am trying this one:

`
esp32:
  board: esp32dev
  variant: ESP32S2
  framework:
    type: esp-idf
    version: recommended
`
but then I get an error:
`Processing esp32s2113stairsled (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.80400.0
Error: Could not find the package with 'platformio/toolchain-xtensa32 @ ~2.80400.0' requirements for your system 'linux_aarch64'`

any help please? I am sure S2 is supported ;-)

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.1

### What platform are you using?

ESP32-IDF

### Board

ESP32 S2

### Component causing the issue

compiler

### Example YAML snippet

```yaml
esphome:
  name: esp32s2113stairsled

esp32:
  board: esp32dev
  variant: ESP32S2
  framework:
    type: esp-idf
    version: recommended
```

### Anything in the logs that might be useful for us?

```txt
Processing esp32s2113stairsled (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.80400.0
Error: Could not find the package with 'platformio/toolchain-xtensa32 @ ~2.80400.0' requirements for your system 'linux_aarch64'
```

oxan, 2021-12-17:
The toolchain for this variant isn't available on ARM64 yet. Try it on something running x86.

homonto, 2021-12-17:
thank you - my only ESPHome is on my Raspberry Pi that hosts home assistant ;-(

jwillikers, 2022-01-10:
> > @homonto I've got an example ESPHome project for an ESP32-S2 board here: https://github.com/jwillikers/esphome-funhouse
> > You can see ESPHome is currently building that YAML file in CI on GItHub.
> > Perhaps you could see if that builds for you?
> 
> but do you compile it on Raspberry Pi? the issue is: no build for arm

It looks like perhaps they are still working on support for aarch64 in esp-idf. See issue espressif/esp-idf#6432.

I currently have to build on an `x86_64` machine specifically for ESP32-S2 and ESP32-C3 boards.

mmakaay, 2022-06-23:
Duplicate of #2722