esphome/issues #2839: Problems with the I2C bus on C3

Joerny, 2021-12-17:
### The problem

I try to add I2C bus sensors to a new ESP32 C3 (NODEMCU ESP32-C3S KIT). Unfortunately, it is not working as the sensors will not be recognized. I tried it with two different BME280 and a CCS811 which work fine on my ESP8266. The main issue is that there will be the message `Recovery failed: SCL is held LOW on the I2C bus`.

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

NODEMCU ESP32-C3S KIT

### Component causing the issue

I2C

### Example YAML snippet

```yaml
esphome:
  name: weather-station
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    platform_packages:
      - framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.1
    board_build.variant: esp32c3
    board_build.f_cpu: 160000000L
    board_build.f_flash: 40000000L
    upload_protocol: esptool
    board_build.flash_mode: dio

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "Random SSID"
  password: "RandomPassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Weather Station Fallback Hotspot"
    password: "5czXpDsSa3tj"

i2c:
  sda: 5
  scl: 4
```

### Anything in the logs that might be useful for us?

```txt
'''
[11:57:03]Build:Feb  7 2021
[11:57:03]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[11:57:03]SPIWP:0xee
[11:57:03]mode:DIO, clock div:2
[11:57:03]load:0x3fcd6100,len:0x3f0
[11:57:03]load:0x403ce000,len:0x6a8
[11:57:03]load:0x403d0000,len:0x2398
[11:57:03]SHA-256 comparison failed:
[11:57:03]Calculated: 5938b3b84f3ab0568f933494a361b2efa6540586e4a954817becdf9719baa9e3
[11:57:03]Expected: f3e046e98d213e66c1ba230da56bf5d5fdb3d903e22d6d6137f691b758d25ed7
[11:57:03]Attempting to boot anyway...
[11:57:03]entry 0x403ce000
[11:57:03][I][logger:214]: Log initialized
[11:57:03][C][ota:458]: There have been 2 suspected unsuccessful boot attempts.
[11:57:03][D][esp32.preferences:114]: Saving preferences to flash...
[11:57:03][I][app:029]: Running through setup()...
[11:57:03][I][i2c.arduino:156]: Performing I2C bus recovery
[11:57:03][E][i2c.arduino:172]: Recovery failed: SCL is held LOW on the I2C bus
[11:57:03][   255][E][esp32-hal-i2c-slave.c:573] i2c_slave_check_line_state(): Bus Invalid State, Can't init sda=0, scl=1
[11:57:03][   261][E][esp32-hal-i2c-slave.c:309] i2cSlaveInit(): bad pin state
[11:57:03][   267][E][Wire.cpp:167] begin(): Slave Init ERROR
[11:57:03][   272][E][esp32-hal-i2c.c:251] i2cSetClock(): could not acquire lock
[11:57:03][C][wifi:037]: Setting up WiFi...
[11:57:03][D][wifi:370]: Starting scan...
[11:57:08][D][wifi:385]: Found networks:
[11:57:08][I][wifi:429]: - 'Random SSID' (38:10:D5:3F:19:81) ▂▄▆█
[11:57:08][D][wifi:430]:     Channel: 11
[11:57:08][D][wifi:431]:     RSSI: -47 dB
[11:57:08][I][wifi:429]: - 'Random SSID' (7C:FF:4D:05:45:7D) ▂▄▆█
[11:57:08][D][wifi:430]:     Channel: 11
[11:57:08][D][wifi:431]:     RSSI: -73 dB
[11:57:08][I][wifi:245]: WiFi Connecting to 'Random SSID'...
[11:57:09][I][wifi:502]: WiFi Connected!
[11:57:09][C][wifi:350]:   Local MAC: 34:B4:72:4C:46:20
[11:57:09][C][wifi:351]:   SSID: 'Random SSID'
[11:57:09][C][wifi:352]:   IP Address: 192.168.1.48
[11:57:09][C][wifi:354]:   BSSID: 38:10:D5:3F:19:81
[11:57:09][C][wifi:355]:   Hostname: 'weather-station'
[11:57:09][C][wifi:357]:   Signal strength: -45 dB ▂▄▆█
[11:57:09][C][wifi:361]:   Channel: 11
[11:57:09][C][wifi:362]:   Subnet: 255.255.255.0
[11:57:09][C][wifi:363]:   Gateway: 192.168.1.1
[11:57:09][C][wifi:364]:   DNS1: 192.168.1.1
[11:57:09][C][wifi:365]:   DNS2: 0.0.0.0
[11:57:09][D][wifi:511]: Disabling AP...
[11:57:09][C][ota:082]: Over-The-Air Updates:
[11:57:09][C][ota:083]:   Address: weather-station.local:3232
[11:57:09][W][ota:092]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[11:57:09][C][api:025]: Setting up Home Assistant API server...
[11:57:09][I][app:062]: setup() finished successfully!
[11:57:09][I][app:102]: ESPHome version 2021.12.1 compiled on Dec 17 2021, 11:56:35
[11:57:09][C][wifi:488]: WiFi:
[11:57:09][C][wifi:350]:   Local MAC: 34:B4:72:4C:46:20
[11:57:09][C][wifi:351]:   SSID: 'Random SSID'
[11:57:09][C][wifi:352]:   IP Address: 192.168.1.48
[11:57:09][C][wifi:354]:   BSSID: 38:10:D5:3F:19:81
[11:57:09][C][wifi:355]:   Hostname: 'weather-station'
[11:57:09][C][wifi:357]:   Signal strength: -45 dB ▂▄▆█
[11:57:09][C][wifi:361]:   Channel: 11
[11:57:09][C][wifi:362]:   Subnet: 255.255.255.0
[11:57:09][C][wifi:363]:   Gateway: 192.168.1.1
[11:57:09][C][wifi:364]:   DNS1: 192.168.1.1
[11:57:09][C][wifi:365]:   DNS2: 0.0.0.0
[11:57:09][C][logger:233]: Logger:
[11:57:09][C][logger:234]:   Level: DEBUG
[11:57:09][C][logger:235]:   Log Baud Rate: 115200
[11:57:09][C][logger:236]:   Hardware UART: UART0
[11:57:09][C][i2c.arduino:032]: I2C Bus:
[11:57:09][C][i2c.arduino:033]:   SDA Pin: GPIO7
[11:57:09][C][i2c.arduino:034]:   SCL Pin: GPIO6
[11:57:09][C][i2c.arduino:035]:   Frequency: 50000 Hz
[11:57:09][C][i2c.arduino:041]:   Recovery: failed, SCL is held low on the bus
[11:57:09][I][i2c.arduino:048]: Scanning i2c bus for active devices...
[11:57:10][  6516][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock
[11:57:10][  6520][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock
[11:57:10][  6526][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock
[11:57:10][  6532][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock

[... snipp ...]

[11:57:10][  7161][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock
[11:57:10][  7167][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock
[11:57:10][  7173][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock
[11:57:10][  7179][E][esp32-hal-i2c.c:139] i2cWrite(): could not acquire lock
[11:57:10][I][i2c.arduino:060]: Found no i2c devices!
[11:57:10][C][ota:082]: Over-The-Air Updates:
[11:57:10][C][ota:083]:   Address: weather-station.local:3232
[11:57:10][W][ota:092]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[11:57:10][C][api:134]: API Server:
[11:57:10][C][api:135]:   Address: weather-station.local:6053
[11:57:10][C][api:139]:   Using noise encryption: NO
[11:57:10][C][mdns:084]: mDNS:
[11:57:10][C][mdns:085]:   Hostname: weather-station
'''
```

jckkk, 2021-12-21:
Perhaps try with different pins. Pins 6 and 7 are used by internal flash. See https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Joerny, 2021-12-22:
The NODEMCU ESP32-C3S KIT has a different pin layout. See https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-32s-kit-v1.0_specification.pdf
However, I tried several combinations with no luck.

jckkk, 2021-12-22:
I'm no expert and can be wrong, but it seems to me that they are still used to communicate with the integrated flash (FSPICLK, FSPID)

Joerny, 2021-12-22:
I am no expert, either. Therefore, I tried several combinations of different pins (0, 1, 3, 4, 5, ...). For reasons I only focused on one side of the device. Now I tried pins 18 & 19. The message changed (`Bus is in Slave Mode`) but the device is still not found:

```
[19:55:23]Build:Feb  7 2021
[19:55:23]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[19:55:23]SPIWP:0xee
[19:55:23]mode:DIO, clock div:2
[19:55:23]load:0x3fcd6100,len:0x3f0
[19:55:23]load:0x403ce000,len:0x6a8
[19:55:23]load:0x403d0000,len:0x2398
[19:55:23]SHA-256 comparison failed:
[19:55:23]Calculated: 5938b3b84f3ab0568f933494a361b2efa6540586e4a954817becdf9719baa9e3
[19:55:23]Expected: f3e046e98d213e66c1ba230da56bf5d5fdb3d903e22d6d6137f691b758d25ed7
[19:55:23]Attempting to boot anyway...
[19:55:23]entry 0x403ce000
[19:55:23][I][logger:214]: Log initialized
[19:55:23][C][ota:458]: There have been 1 suspected unsuccessful boot attempts.
[19:55:23][D][esp32.preferences:114]: Saving preferences to flash...
[19:55:23][I][app:029]: Running through setup()...
[19:55:23][I][i2c.arduino:156]: Performing I2C bus recovery
[19:55:23][   256][E][Wire.cpp:283] setClock(): Bus is in Slave Mode
[19:55:23][C][wifi:037]: Setting up WiFi...
[19:55:23][D][wifi:370]: Starting scan...
[19:55:27][D][wifi:385]: Found networks:
[19:55:27][I][wifi:429]: - 'Random SSID' (38:10:D5:3F:19:81) ▂▄▆█
[19:55:28][D][wifi:430]:     Channel: 1
[19:55:28][D][wifi:431]:     RSSI: -41 dB
[19:55:28][I][wifi:429]: - 'Random SSID' (7C:FF:4D:05:45:7D) ▂▄▆█
[19:55:28][D][wifi:430]:     Channel: 1
[19:55:28][D][wifi:431]:     RSSI: -76 dB
19:55:28][I][wifi:245]: WiFi Connecting to 'Random SSID'...
[19:55:29][I][wifi:502]: WiFi Connected!
[19:55:29][C][wifi:350]:   Local MAC: 34:B4:72:4C:46:20
[19:55:29][C][wifi:351]:   SSID: 'Random SSID'
[19:55:29][C][wifi:352]:   IP Address: 192.168.1.48
[19:55:29][C][wifi:354]:   BSSID: 38:10:D5:3F:19:81
[19:55:29][C][wifi:355]:   Hostname: 'weather-station'
[19:55:29][C][wifi:357]:   Signal strength: -43 dB ▂▄▆█
[19:55:29][C][wifi:361]:   Channel: 1
[19:55:29][C][wifi:362]:   Subnet: 255.255.255.0
[19:55:29][C][wifi:363]:   Gateway: 192.168.1.1
[19:55:29][C][wifi:364]:   DNS1: 192.168.1.1
[19:55:29][C][wifi:365]:   DNS2: 0.0.0.0
[19:55:29][D][wifi:511]: Disabling AP...
[19:55:29][C][ota:082]: Over-The-Air Updates:
[19:55:29][C][ota:083]:   Address: weather-station.local:3232
[19:55:29][C][api:025]: Setting up Home Assistant API server...
[19:55:29][I][app:062]: setup() finished successfully!
[19:55:29][I][app:102]: ESPHome version 2021.12.1 compiled on Dec 22 2021, 19:54:59
[19:55:29][C][wifi:488]: WiFi:
[19:55:29][C][wifi:350]:   Local MAC: 34:B4:72:4C:46:20
[19:55:29][C][wifi:351]:   SSID: 'Random SSID'
[19:55:29][C][wifi:352]:   IP Address: 192.168.1.48
[19:55:29][C][wifi:354]:   BSSID: 38:10:D5:3F:19:81
[19:55:29][C][wifi:355]:   Hostname: 'weather-station'
[19:55:29][C][wifi:357]:   Signal strength: -43 dB ▂▄▆█
[19:55:29][C][wifi:361]:   Channel: 1
[19:55:29][C][wifi:362]:   Subnet: 255.255.255.0
[19:55:29][C][wifi:363]:   Gateway: 192.168.1.1
[19:55:29][C][wifi:364]:   DNS1: 192.168.1.1
[19:55:29][C][wifi:365]:   DNS2: 0.0.0.0
[19:55:29][C][logger:233]: Logger:
[19:55:29][C][logger:234]:   Level: DEBUG
[19:55:29][C][logger:235]:   Log Baud Rate: 115200
[19:55:29][C][logger:236]:   Hardware UART: UART0
[19:55:29][C][i2c.arduino:032]: I2C Bus:
[19:55:29][C][i2c.arduino:033]:   SDA Pin: GPIO18
[19:55:29][C][i2c.arduino:034]:   SCL Pin: GPIO19
[19:55:29][C][i2c.arduino:035]:   Frequency: 50000 Hz
[19:55:29][C][i2c.arduino:038]:   Recovery: bus successfully recovered
[... 18 lines cut ...]
[19:55:29][I][i2c.arduino:056]: Unknown error at address 0x0D
[19:55:29][  6582][E][Wire.cpp:308] beginTransmission(): Bus is in Slave Mode
[19:55:29][  6583][E][Wire.cpp:331] endTransmission(): Bus is in Slave Mode
[19:55:29][I][i2c.arduino:056]: Unknown error at address 0x0E

[...snipp...]

[19:55:31][  8370][E][Wire.cpp:308] beginTransmission(): Bus is in Slave Mode
[19:55:31][  8371][E][Wire.cpp:331] endTransmission(): Bus is in Slave Mode
[19:55:31][I][i2c.arduino:056]: Unknown error at address 0x75
[19:55:31][  8387][E][Wire.cpp:308] beginTransmission(): Bus is in Slave Mode
[19:55:31][  8389][E][Wire.cpp:331] endTransmission(): Bus is in Slave Mode
[19:55:31][I][i2c.arduino:056]: Unknown error at address 0x76
[19:55:31][  8405][E][Wire.cpp:308] beginTransmission(): Bus is in Slave Mode
[19:55:31][  8406][E][Wire.cpp:331] endTransmission(): Bus is in Slave Mode
[19:55:31][I][i2c.arduino:056]: Unknown error at address 0x77
[19:55:31][I][i2c.arduino:060]: Found no i2c devices!
[19:55:31][C][ota:082]: Over-The-Air Updates:
[19:55:31][C][ota:083]:   Address: weather-station.local:3232
[19:55:31][C][api:134]: API Server:
[19:55:31][C][api:135]:   Address: weather-station.local:6053
[19:55:31][C][api:139]:   Using noise encryption: NO
[19:55:31][C][mdns:084]: mDNS:
[19:55:31][C][mdns:085]:   Hostname: weather-station
```

closed 2022-01-03 by oxan

societyofrobots, 2022-02-23:
> Recovery failed: SCL is held LOW on the I2C bus

Do you have pull-up resistors on SDA and SCL?

Joerny, 2022-02-25:
After the update to 2022.1 everything works as expected...