esphome/issues #2846: Teleinfo issue

76MPaul, 2021-12-18:
### The problem

Hi everyone, 

I'm trying to retrieve a Linky informations with the Teleinfo Integration and I got the errors bellow.
I can't get the logs in ESPHome even when the log from the usb say it's connected on wifi.

What I tried :
 - no integration-> work
 - Only the UART config -> work
 - Teleinfo integration - > not working
 - second ESP -> not working
 - Esphome dev version -> same results with above tests

Any idea ? 

### Which version of ESPHome has the issue?

2021.12.1 & 2021.12.1-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.3 & supervisor-2021.12.2

### What platform are you using?

ESP32

### Board

esp32-devkit

### Component causing the issue

Teleinfo 

### Example YAML snippet

```yaml
esphome:
  name: teleinfo
  platform: ESP32
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "********"

wifi:
  ssid: "********"
  password: "********"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Teleinfo Fallback Hotspot"
    password: "********"

captive_portal:

uart:
  id: uart_bus
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 1200
  parity: EVEN
  data_bits: 7

teleinfo:
  id: myteleinfo
  update_interval: 30s
  historical_mode: true

sensor:
  - platform: teleinfo
    tag_name: "HCHC"
    name: "hchc"
    unit_of_measurement: "Wh"
    icon: mdi:flash
    teleinfo_id: myteleinfo
  - platform: teleinfo
    tag_name: "HCHP"
    name: "hchp"
    unit_of_measurement: "Wh"
    icon: mdi:flash
    teleinfo_id: myteleinfo
  - platform: teleinfo
    tag_name: "PAPP"
    name: "papp"
    unit_of_measurement: "VA"
    icon: mdi:flash
    teleinfo_id: myteleinfo

text_sensor:
  - platform: teleinfo
    tag_name: "OPTARIF"
    name: "optarif"
    teleinfo_id: myteleinfo
```

### Anything in the logs that might be useful for us?

```txt
Plateform informations from esphome install :
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0

USB Logs from ESP :
[23:11:09][E][esp32-hal-misc.c:111] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
[21:01:13][D][esp32.preferences:114]: Saving preferences to [21:01:13]
[21:01:13]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:01:13]configsip: 0, SPIWP:0xee
[21:01:13]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:01:13]mode:DIO, clock div:2
[21:01:13]load:0x3fff0018,len:4
[21:01:13]load:0x3fff001c,len:1044
[21:01:13]load:0x40078000,len:8896
[21:01:13]load:0x40080400,len:5828
[21:01:13]entry 0x400806ac
[21:01:14][I][logger:214]: Log initialized
[21:01:14][C][ota:458]: There have been 10 suspected unsuccessful boot attempts.
[21:01:14][D][esp32.preferences:114]: Saving preferences to flash...
[21:01:14][E][ota:464]: Boot loop detected. Proceeding to safe mode.
[21:01:14][I][app:029]: Running through setup()...
[21:01:14][C][wifi:037]: Setting up WiFi...
[21:01:14][D][wifi:370]: Starting scan...
[21:01:16][D][wifi:385]: Found networks:
[21:01:16][I][wifi:429]: - '*******' (*******) ▂▄▆█
[21:01:16][D][wifi:430]:     Channel: 2
[21:01:16][D][wifi:431]:     RSSI: -64 dB
[21:01:16][D][wifi:434]: - '*******' (*******) ▂▄▆█
[21:01:16][D][wifi:434]: - '*******' (*******) ▂▄▆█
[21:01:16][D][wifi:434]: - '*******' (*******) ▂▄▆█
[21:01:16][D][wifi:434]: - '*******' (*******) ▂▄▆█
[21:01:16][I][wifi:245]: WiFi Connecting to '*******'...
[21:01:17][I][wifi:502]: WiFi Connected!
[21:01:17][C][wifi:350]:   Local MAC: *******
[21:01:17][C][wifi:351]:   SSID: '*******'
[21:01:17][C][wifi:352]:   IP Address: 192.168.1.22
[21:01:17][C][wifi:354]:   BSSID: *******
[21:01:17][C][wifi:355]:   Hostname: 'teleinfo'
[21:01:17][C][wifi:357]:   Signal strength: -64 dB ▂▄▆█
[21:01:17][C][wifi:361]:   Channel: 2
[21:01:17][C][wifi:362]:   Subnet: 255.255.255.0
[21:01:17][C][wifi:363]:   Gateway: 192.168.1.1
[21:01:17][C][wifi:364]:   DNS1: 192.168.1.1
[21:01:17][C][wifi:365]:   DNS2: 0.0.0.0
[21:01:17][D][wifi:511]: Disabling AP...
[21:01:17][C][ota:082]: Over-The-Air Updates:
[21:01:17][C][ota:083]:   Address: teleinfo.local:3232
[21:01:17][C][ota:086]:   Using Password.
[21:01:17][W][ota:092]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[21:01:17][I][app:062]: setup() finished successfully!
[21:01:17][I][ota:474]: Waiting for OTA attempt.
[21:01:17][I][app:102]: ESPHome version 2022.1.0-dev compiled on Dec 18 2021, 18:42:54
[21:01:17][C][wifi:488]: WiFi:
[21:01:17][C][wifi:350]:   Local MAC: *******
[21:01:17][C][wifi:351]:   SSID: '*******'
[21:01:17][C][wifi:352]:   IP Address: 192.168.1.22
[21:01:17][C][wifi:354]:   BSSID: *******
[21:01:17][C][wifi:355]:   Hostname: 'teleinfo'
[21:01:17][C][wifi:357]:   Signal strength: -63 dB ▂▄▆█
[21:01:17][C][wifi:361]:   Channel: 2
[21:01:17][C][wifi:362]:   Subnet: 255.255.255.0
[21:01:17][C][wifi:363]:   Gateway: 192.168.1.1
[21:01:17][C][wifi:364]:   DNS1: 192.168.1.1
[21:01:17][C][wifi:365]:   DNS2: 0.0.0.0
[21:01:17][C][logger:233]: Logger:
[21:01:17][C][logger:234]:   Level: DEBUG
[21:01:17][C][logger:235]:   Log Baud Rate: 115200
[21:01:17][C][logger:236]:   Hardware UART: UART0
[21:01:17][C][captive_portal:144]: Captive Portal:
[21:01:17][C][ota:082]: Over-The-Air Updates:
[21:01:17][C][ota:083]:   Address: teleinfo.local:3232
[21:01:17][C][ota:086]:   Using Password.
[21:01:18][W][ota:092]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
```

0hax, 2021-12-20:
Hi,
It looks like the Logger is using `UART0` for logging output (`GPIO3` and `GPIO1`).
And you are declaring the same UART to be used by the teleinfo component.
Can you tell us on which pins you plugged the optocoupler?

76MPaul, 2021-12-20:
Hi, 

Thanks for your answer, 
I tried with and without the optocoupler plugged for the same result. 
If I remembered correctly I plugged the optocoupler on the RX so the GPIO3.

Is the error happening only cause I try to log the ESP on the computer as the same time (with esphomeFlasher) ? 

I will try this evening to get results without logging the esp by USB.

Thanks, 

76MPaul

0hax, 2021-12-20:
Feel free to send a schematic explaining exactly how things are plugged if you need further help

76MPaul, 2021-12-20:
Here is the schematic :
[image]

0hax, 2021-12-21:
I could be wrong but I think TXDO and RXDO are already used by the USB uart.
If that's true, you need to check with another UART.

jesserockz, 2021-12-22:
Yes especially when you need different uart settings for the teleinfo uart.

ESP32 can use hardware UART on any available pin, so best would be to just pick another pin for the teleinfo uart.

76MPaul, 2021-12-22:
Thanks a lot ! 
That was the issue. It's know solved !

It may be interesting to change or advertise on the documentation ?

Thanks again for your help !

jesserockz, 2021-12-22:
There is info around uart [here](https://esphome.io/components/uart.html#hardware-uarts)

closed 2021-12-22 by jesserockz