esphome/issues #2847: Arduino port expander cannot compile

cicinovec, 2021-12-19:
### The problem

When trying to compile code with arduino port expander, it cannot compile

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.3

### What platform are you using?

ESP8266

### Board

ESP8266 + Arduino Nano

### Component causing the issue

arduino port expander

### Example YAML snippet

```yaml
esphome:
  name: bedroom_plants_and_led
  platform: ESP8266
  board: nodemcu
  includes:
  - arduino_port_expander.h

wifi:
  ssid: "Devices"
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom Plants"
    password: !secret wifi_password
    
captive_portal:

# Enable Home Assistant API
api:

ota:

i2c:
  id: i2c_bus

logger:
  level: DEBUG

# define the port expander hub, here we define one with id 'expander1',
# but you can define many
custom_component:
  - id: expander1
    lambda: |-
      auto expander = new ArduinoPortExpander((i2c_bus), 0x08, true);
      return {expander};
      
# define analog sensors
sensor:
[... 172 lines cut ...]
In file included from src/main.cpp:48:0:
src/arduino_port_expander.h:105:7: note: ArduinoPortExpander::ArduinoPortExpander()
 class ArduinoPortExpander : public Component, public I2CDevice
       ^
src/arduino_port_expander.h:105:7: note:   candidate expects 0 arguments, 3 provided
src/arduino_port_expander.h:105:7: note: ArduinoPortExpander::ArduinoPortExpander(const ArduinoPortExpander&)
src/arduino_port_expander.h:105:7: note:   candidate expects 1 argument, 3 provided
src/arduino_port_expander.h:105:7: note: ArduinoPortExpander::ArduinoPortExpander(ArduinoPortExpander&&)
src/arduino_port_expander.h:105:7: note:   candidate expects 1 argument, 3 provided
/config/esphome/bedroom-plant-and-light.yaml:36:23: error: could not convert '{expander}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
       return {expander};
                       ^
Compiling /data/bedroom_plants_and_led/.pioenvs/bedroom_plants_and_led/lib67b/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling /data/bedroom_plants_and_led/.pioenvs/bedroom_plants_and_led/lib67b/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling /data/bedroom_plants_and_led/.pioenvs/bedroom_plants_and_led/lib67b/ESP8266WiFi/WiFiClient.cpp.o
/config/esphome/bedroom-plant-and-light.yaml:37:3: warning: control reaches end of non-void function [-Wreturn-type]
       
   ^
*** [/data/bedroom_plants_and_led/.pioenvs/bedroom_plants_and_led/src/main.cpp.o] Error 1
========================= [FAILED] Took 12.66 seconds =========================
```

### Additional information

Worked before, one of the last updates caused the issue. cannot point out which one

oxan, 2021-12-19:
Your custom component needs to be updated to be compatible with the changes in 2021.10.

cicinovec, 2021-12-19:
What exactly should I change? I took the example from EspHome page

oxan, 2021-12-20:
@glmnet This seems to be an issue in your gist linked from the cookbook.