esphome/issues #2848: crash on ESP8266 with esphome API enabled

trombik, 2021-12-19:
### The problem

ESP8266 reboots after sometime, ranging from 10 minutes to 40 minutes, due to panic.

### Which version of ESPHome has the issue?

master, ref 1fb0a7109d8f08e3e8bb7e2e8761085506193def

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.0

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

esphome

### Example YAML snippet

```yaml
substitutions:
  myname: backgarden-sensor-soil

esphome:
  name: ${myname}
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${myname}"
    password: !secret wifi_ap_password

captive_portal:

web_server:
  ota: false

sensor:
  - platform: template
    name: Soil humidity
    id: soil_humidity
    unit_of_measurement: "%"
    accuracy_decimals: 0

[... 130 lines cut ...]
0x402375a0: mem_free at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:237
0x40100244: esphome::millis() at ??:?
0x40201726: esphome::api::APIConnection::send_buffer(esphome::api::ProtoWriteBuffer, unsigned int) at ??:?
0x40102cfc: wDev_ProcessFiq at ??:?
0x4020292c: esphome::api::APIPlaintextFrameHelper::try_read_frame_(esphome::api::APIPlaintextFrameHelper::ParsedFrame*) at ??:?
0x40223842: esphome::api::APIServerConnectionBase::send_ping_response(esphome::api::PingResponse const&) at ??:?
0x40100882: millis at ??:?
0x40100244: esphome::millis() at ??:?
0x40201d7f: esphome::api::APIConnection::loop() at ??:?
0x40100882: millis at ??:?
0x40100882: millis at ??:?
0x4010059b: esp8266::polledTimeout::timeoutTemplate<false, esp8266::polledTimeout::YieldPolicy::DoNothing, esp8266::polledTimeout::TimePolicy::TimeUnit<esp8266::polledTimeout::TimePolicy::TimeSourceMillis, 1000ull> >::expiredOneShot() const at ??:?
0x40100882: millis at ??:?
0x40105a3d: ets_timer_arm_new at ??:?
0x402207d8: esp_yield at ??:?
0x40220e3a: __delay at ??:?
0x4010025c: esphome::delay(unsigned int) at ??:?
0x4020f4cb: esphome::Application::loop() at ??:?
0x4021197c: loop at ??:?
0x402208fc: loop_wrapper() at core_esp8266_main.cpp:?
```

### Additional information

i have other boards that exhibit the same crash, but with different components. interestingly, i have a board that reboots constantly, and another that does not crash, even when the configuration files are identical. the previous firmware, built with `esphome` 2021.9.x (not 100% sure which version, though), did not crush. i noticed the crush after upgrading 2021.10.x.

tschlueter, 2022-01-17:
My esphome ESP8266 crashes too after adding a template sensor. @oxan try to remove the template sensor "soil_humidity". Even the example code (https://esphome.io/components/sensor/template.html) leads to a crash for me.

trombik, 2022-01-18:
@tschlueter thanks for the info. however, other devices without template sensor crashes as well. i'm not sure i am seeing the same problem on these esp8266 devices. the symptoms do not make sense to me because a device do and do not crash after re-flashing (OTA update) with an identical YAML. when it crashes, the uptime is always less than 15 minutes, and keep crashing. I am still trying to narrow down the cause. yesterday, i updated home assistant to the latest release. let's see.

trombik, 2022-01-19:
for the record, here is the versions I am using now:

* Home Assistant 2021.12.9
* ESPHome version 2022.1.0-dev

you can see a saw wave of uptime sensor at the beginning. so far, the uptime is more than a day.

[image]

trombik, 2022-02-27:
updated arduino to 3.x after recent bump in `esphome` release. so far. the uptime is more than five days. before the upgrade, the longest uptime never exceeded one week (but far loner than 15 minutes, which is better).

trombik, 2022-03-03:
today, the uptime exceeded 9 days.

trombik, 2022-08-06:
i'm closing for now.

closed 2022-08-06 by trombik