esphome/issues #2849: AM43 component can’t find sensor.h

petergerdes, 2021-12-19:
### The problem

AM43 blinds component won’t compile.

Tried with `  libraries:
    - "Sensor@0.1.0" ` and without.

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

am43

### Example YAML snippet

```yaml
esphome:
  name: kitchenblinds
  platform: ESP32
  board: nodemcu-32s
  libraries:
    - "Sensor@0.1.0"

wifi:
  ssid: "SSID"
  password: "Password"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "Password"

ota:
  password: "Password"

web_server:
  port: 80

esp32_ble_tracker:

ble_client:
  - mac_address: AA:BB:CC:DD:EE:FF
    id: am43_kitchen

cover:
  - platform: am43
    name: "Kitchen Blinds"
    ble_client_id: am43_kitchen
```

### Anything in the logs that might be useful for us?

```txt
Library Manager: ArduinoJson-esphomelib @ 5.13.3 has been installed!
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <Sensor> 0.1.0
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling .pioenvs/kitchenblinds/src/esphome/components/am43/am43.cpp.o
Compiling .pioenvs/kitchenblinds/src/esphome/components/am43/am43_base.cpp.o
Compiling .pioenvs/kitchenblinds/src/esphome/components/am43/cover/am43_cover.cpp.o
In file included from src/esphome/components/am43/am43.cpp:1:0:
src/esphome/components/am43/am43.h:6:46: fatal error: esphome/components/sensor/sensor.h: No such file or directory
compilation terminated.
*** [.pioenvs/kitchenblinds/src/esphome/components/am43/am43.cpp.o] Error 1
Compiling .pioenvs/kitchenblinds/src/esphome/components/api/api_connection.cpp.o
========================================================================= [FAILED] Took 9.42 seconds =========================================================================
```

onkel-joerg, 2022-01-16:
Same problem here

jesserockz, 2022-01-17:
Adding a library like that will not solve the problem as `sensor.h` is an internal ESPHome file (`esphome/components/sensor/sensor.h`)

Have you tried to do a clean build? - Overflow menu (three dots) in device card in dashboard then `Clean Build Files`, then try install again.

onkel-joerg, 2022-01-17:
Yes I tried, but unfortunately the same result

OttoWinter, 2022-01-18:
Try putting an empty `sensor` block in the config

```yaml
sensor:
```

The issue probably is that `am43` should be declaring `sensor` as an `AUTO_LOAD` if it depends on those files.

onkel-joerg, 2022-01-19:
update:
It works with the "sensor:" in the yaml file.
"open", "close", "stop" work fine, the pecentage of cover opening is also transmitted. 
I can't find battery poercentage, but I'm very new to HA and ESPhome, so I'll probably find it. ;-)

Conclusion: it works fine

best regards

buxtronix, 2022-01-19:
To get battery percentage you need to also declare some sensors:

```
sensor:
  - platform: am43
    ble_client_id: am43_kitchen
    battery_level:
      name: "AM43 battery"
    illuminance:
      name: "AM43 light"
    update_interval: 120s
```

onkel-joerg, 2022-01-19:
Thank you.
added it, compiled it and transmitted it to the ESP
works like a charm. 
Now i have to learn how to transmit all the item values to mqtt broker
(I'm quiet new in HA)

btw:
What does the (new) notification mean, that I get?

 `{'DNS compression type 69 is unknown at 12': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                              IncomingDecodeError('DNS compression type 69 is unknown at 12'),
                                              <traceback object at 0x7f7fb98880>)}`

closed 2022-01-20 by OttoWinter