esphome/issues #2850: sim800l plugin is incompatible newer cellular modules

jraynes, 2021-12-19:
### The problem

With the [impending shutdown of 2G cellular services](https://www.digi.com/blog/post/2g-3g-4g-lte-network-shutdown-updates)  I wanted to bring attention to the fact that the sim800l module in esphome does not work with newer LTE modules from Simcom. 

I have [this SIM7080G module from aliexpress](https://www.aliexpress.com/item/4001035720662.html). Instead of 2G GSM it uses LTE CAT-M which is fairly future proof from what I understand. According to Simcom's website[this module is "compatible with SIM868"](https://www.simcom.com/product/SIM7080G.html). In my own testing I can confirm this module uses the same AT commands defined in 3GPP TS 27.005 to send and receive text messages. But when I connect the module to my esp board via UART serial it does not seem to talk to the module at all:

```
[16:51:32][C][sim800l:279]: SIM800L:
[16:51:32][C][sim800l:280]:   RSSI: 0 dB
```

the entities are discovered in homeassistant however their states are stuck on "unknown" status.

I realized the default line speed for the SIM7080 module is 115200 baud so I tried to set that in my esphome YAML like so:

```
uart:
  baud_rate: 115200
  tx_pin: TX
  rx_pin: RX
```

But when I try to install it looks like the speed is hard coded in esphome:

INFO Reading configuration /config/esphome/modem.yaml...
Failed config

```
uart: [source /config/esphome/modem.yaml:48]
  - [source /config/esphome/modem.yaml:48]
    
    Component sim800l required baud rate 9600 for the uart bus.
    baud_rate: 115200 [source /config/esphome/modem.yaml:49]
    tx_pin: 
      number: 1
      mode: 
        output: True
        input: False
        open_drain: False
        pullup: False
        pulldown: False
      inverted: False
    rx_pin: 
      number: 3
      mode: 
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
      inverted: False
    rx_buffer_size: 256
    stop_bits: 1
    data_bits: 8
    parity: NONE

```

I would love to see esphome support these newer modules. I have several laying around and would be happy to help with this effort.

### Which version of ESPHome has the issue?

v2021.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.3

### What platform are you using?

ESP32

### Board

esp32devkitv1

### Component causing the issue

sim800l

dougle03, 2021-12-21:
Have you tried this on a more recent version of ESPHome? I note you're using an Oct 21 build.... 
This may not solve your problem, but it's an obvious step to take.

jesserockz, 2021-12-22:
There is a validation on the sim800l which enforces the baud rate.

Try changing these lines, either change the baud, or remove baud=9600 to not care.

https://github.com/esphome/esphome/blob/e152f128c8ed2975aa410ccde0ab85df3d3d8f4e/esphome/components/sim800l/__init__.py#L43-L45

jraynes, 2021-12-22:
@jesserockz when I remove the line setting the baud rate I get:

```
INFO Reading configuration /config/esphome/modem.yaml...
Failed config

uart: [source /config/esphome/modem.yaml:48]
  
  'baud_rate' is a required option for [0].
  - [source /config/esphome/modem.yaml:49]
    tx_pin: TX
    rx_pin: RX
```

```
and when I try to set the baud rate to anything other than 9600 I get:

INFO Reading configuration /config/esphome/modem.yaml...
Failed config

uart: [source /config/esphome/modem.yaml:48]
  - [source /config/esphome/modem.yaml:48]
    
    Component sim800l required baud rate 9600 for the uart bus.
    baud_rate: 115200 [source /config/esphome/modem.yaml:49]
```

jraynes, 2021-12-22:
@dougle03 I do not see an upgrade available:
[image]

jesserockz, 2021-12-22:
Can you try add this to your yaml:

```yaml
external_components:
  - source: github://esphome/esphome@sim800-remove-baud-rate-validation
    components: ["sim800l"]
```

jraynes, 2021-12-27:
getting this error now:

[image]

closed 2022-01-04 by oxan

dougle03, 2022-01-04:
> @dougle03 I do not see an upgrade available: [image]

You will have to wait for the HA dev team to update the container that Supervisor references. there are other ways to get a later version.