esphome/issues #2851: NodeMCUv3 pin layout is flipped

bruvv, 2021-12-20:
### The problem

When connecting mhz19 co2 sensor I use the following:
```
uart:
  rx_pin: D1
  tx_pin: D2
  baud_rate: 9600
```
When connecting it to the RX and TX of the MHZ19 sensor it is switch and the RX pin is D2 and the TX pint is D1.

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12

### What platform are you using?

ESP8266

### Board

NodeMCUV3

### Component causing the issue

MH-Z19c

oxan, 2021-12-21:
This is a common cause of confusion with UART, the RX and TX pins on the board and in the config are named from the ESP's perspective, and on sensors they're usually named from the sensor's perspective, so you need to connect RX with TX and vice versa.

jesserockz, 2021-12-22:
I have updated the docs for MHZ19 to show generic GPIO pins. And I will try to remove any further reference to `D` pins from the docs as they are board specific.

I am going to close this issue as although I can update the doc pins, its not really an issue that can be fixed, but thank you for reporting.

closed 2021-12-22 by jesserockz