esphome/issues #2852: wind sensor TX23 not working with the current TX20/TX23

samtwilliams, 2021-12-20:
### The problem

As per issue https://github.com/esphome/issues/issues/907 which is closed due to no-activity.

The issue still persists with tx23.

All wiring correct as per documentation connected to esp32. Upon log no data is being recieved.

### Which version of ESPHome has the issue?

Copyright © 2019-2021 ESPHome | Fund developmentv2021.11.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

core-2021.11.5

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1 (tried on different boards also)

### Component causing the issue

TX23

### Example YAML snippet

```yaml
esphome:
  name: esp_living_room
  platform: ESP32
  board: esp32doit-devkit-v1
#  board: nodemcu-32s

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Climate Fallback Hotspot"
    password: "X3e69v28lq2z"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: tx20
    id: tx20_id
    wind_speed:
      name: "Windspeed"
    wind_direction_degrees:
      name: "Wind direction degrees"
    pin: 
      number: GPIO04
      mode: INPUT
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp-living-room.yaml...
WARNING 'esp_living_room': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from esp_living_room.local using esphome API
INFO Successfully connected to esp_living_room.local
[16:30:25][I][app:099]: ESPHome version 2021.11.4 compiled on Dec 20 2021, 16:22:51
[16:30:25][C][wifi:488]: WiFi:
[16:30:25][C][wifi:350]:   Local MAC: 24:62:AB:FC:76:E8
[16:30:25][C][wifi:351]:   SSID: [redacted]
[16:30:25][C][wifi:352]:   IP Address: 192.168.0.131
[16:30:25][C][wifi:354]:   BSSID: [redacted]
[16:30:25][C][wifi:355]:   Hostname: 'esp_living_room'
[16:30:25][C][wifi:357]:   Signal strength: -58 dB ▂▄▆█
[16:30:25][C][wifi:361]:   Channel: 1
[16:30:25][C][wifi:362]:   Subnet: 255.255.255.0
[16:30:25][C][wifi:363]:   Gateway: 192.168.0.1
[16:30:25][C][wifi:364]:   DNS1: 192.168.2.1
[16:30:25][C][wifi:365]:   DNS2: 192.168.3.1
[16:30:25][C][logger:233]: Logger:
[16:30:25][C][logger:234]:   Level: DEBUG
[16:30:25][C][logger:235]:   Log Baud Rate: 115200
[16:30:25][C][logger:236]:   Hardware UART: UART0
[16:30:25][C][tx20:026]: Tx20:
[16:30:25][C][tx20:028]:   Wind speed: 'Windspeed'
[16:30:25][C][tx20:028]:     State Class: 'measurement'
[16:30:25][C][tx20:028]:     Unit of Measurement: 'km/h'
[16:30:25][C][tx20:028]:     Accuracy Decimals: 1
[16:30:25][C][tx20:028]:     Icon: 'mdi:weather-windy'
[16:30:25][C][tx20:029]:   Wind direction degrees: 'Wind direction degrees'
[16:30:25][C][tx20:029]:     State Class: ''
[16:30:25][C][tx20:029]:     Unit of Measurement: '°'
[16:30:25][C][tx20:029]:     Accuracy Decimals: 1
[16:30:25][C][tx20:029]:     Icon: 'mdi:sign-direction'
[16:30:25][C][tx20:031]:   Pin: GPIO4
[16:30:25][C][captive_portal:151]: Captive Portal:
[16:30:25][C][ota:082]: Over-The-Air Updates:
[16:30:25][C][ota:083]:   Address: esp_living_room.local:3232
[16:30:25][C][api:134]: API Server:
[16:30:25][C][api:135]:   Address: esp_living_room.local:6053
[16:30:25][C][api:139]:   Using noise encryption: NO
[16:30:25][C][mdns:084]: mDNS:
[16:30:25][C][mdns:085]:   Hostname: esp_living_room
```

the-cyberspace, 2021-12-25:
#907

the-cyberspace, 2021-12-31:
I have the same problem. I thought that the wind sensor was broken or i`ve done something wrong with te code but now i understand why it`s not working.

samtwilliams, 2021-12-31:
So I managed to get it working following a custom component found in another post. Looked quite a daunting implementation but actually quite straightforward. 

Mentioned by sebyeocky here https://github.com/esphome/issues/issues/907

closed 2021-12-31 by samtwilliams