esphome/issues #2854: on_boot initialization order ignored?

kaolpr, 2021-12-21:
### The problem

The board I'm trying to use with Esphome uses MCP23008 GPIO expander. It also drives its nRESET pin. In order to use exapnder, I need to set nRESET state to high before MCP23008 initialization takes place.

I tried adding `on_boot` to `esphome` configuration entry:

```yaml
on_boot:
    priority: 800.0
    then:
      - switch.turn_on: expander_reset
[...]
switch:
  - id: expander_reset
    pin: GPIO5
    platform: gpio
    restore_mode: ALWAYS_ON
```

I also tried different (than 800) `priority` values, however, in every case `expander_reset` initialization takes place after MCP23008 initialization (what in my case is show-stopper as pull-ups are used) - verified with generated `main.cpp`.

I believe I must be missing something obvious, as setting initialization order seems pretty basic feature...

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Board

ESP32-POE + custom extension

oxan, 2021-12-21:
Did you try a priority above 900?

kaolpr, 2021-12-21:
I tried 1000, but switch initialization was still long after expander init.

jesserockz, 2021-12-22:
Ok so the expander itself as priority of 900, which runs before the gpio switch of 800. And your on_boot actions wont work with a higher priority as the switch wont be ready.

You can change the `setup_priority` in yaml of each component

closed 2022-01-18 by OttoWinter