esphome/issues #2856: Error starting esphome - importlib.metadata.PackageNotFoundError: esphome

mzac, 2021-12-21:
### The problem

I just upgraded my esphome running in k3s and getting an error on both x86 and arm64 on 2021.12.2.  Reverting back to 2021.12.1 works fine.

```
Traceback (most recent call last):
  File "/usr/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/usr/bin/esphome", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: esphome
```

### Which version of ESPHome has the issue?

2021.12.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

goncalvespedro, 2021-12-21:
Having the same issue

jesserockz, 2021-12-21:
It seems something went weirdly wrong with the release. I am looking into it but please restore the HA addon backup from ESPHome 2021.12.1 or use the docker image by 2021.12.1 tag until I can remove the 2021.12.2 ones.

comet424, 2021-12-21:
i tried to find a way how to downgrade  the esphome but i have no idea

sdaudet-ua, 2021-12-21:
@jesserockz This is a new install for me, just got a bunch of new ESP boards in and I am excited to play with them! Is there anything I can do locally to get 2021.12.1 downloaded?

comet424, 2021-12-21:
that be a good feature request  and option  to choose which version of the install to install. and a easy button to downgrade etc

Aliskin-Papa, 2021-12-21:
Sam here

sdaudet-ua, 2021-12-21:
@comet424 I saw a feature request for that like 2 years ago... not sure we are going to get what we really want on that one :/

comet424, 2021-12-21:
from the downgrade  link   and it had code for  config.json   i pasted 
@@ -38,5 +38,5 @@
  "slug": "esphome",
  "uart": true,
  "url": "https://esphome.io/",
  "version": "2021.12.2"
} 
  "version": "2021.12.1"
}

in the config.json   and in the esphome  directory

but it didnt cause the esphome to downgrade.. even with reboot of HA

@sdaudet-ua aww ok..  ya it be good for any of the addons  be able to downgrade to a different version    cuz once you upgrade you cant go back

sdaudet-ua, 2021-12-21:
@comet424 The link you are talking about is a merge request for the ha community add-on store, not meant for you to copy-paste in to a local config. Basically it is the request to bring ESPHome back to 2021.12.1 for everyone that goes to download it. I am not aware of a way to specify the version being downloaded, which is the feature we would LOVE to see.

comet424, 2021-12-21:
@sdaudet-ua ah ok  here i thought it would get be down...

and hopefully me unininstalling and re installing esphome last few times   didnt erase my pre made  esphome config files  for my devices

jesserockz, 2021-12-21:
If you install the addons repo direct from esphome (`https://github.com/esphome/hassio`) then it should only offer 2021.12.1 as I have more control over that one =)

Uninstalling/reinstalling the addon will not touch your config files and is designed that way. The files reside in the HA config folder and addons cannot delete those.

comet424, 2021-12-21:
@jesserockz 
i dont know how to install it..  i went to the webpage..  but it doesnt say how do i install it..
i did try    in HA addons and click repository  and pasted the link as it works for some other github  like scheduler
but it didnt work

how do i actually install it?  it doesnt say  like click this  or run this command line to install it
so i have no idea.. how to install the 12.1

i not advanced in github .. i more of a windows user where you just have a exe  and can install different versions

sdaudet-ua, 2021-12-21:
@comet424 Go to Add-Ons, click the 3 dots in the top right and click repositories. Paste the above URL there to find the ESPHome add-on in its own category.

sdaudet-ua, 2021-12-21:
@jesserockz Thank you very much, I was able to get it going!

comet424, 2021-12-21:
@sdaudet-ua  ah ok ya thats what i did what i mentioned above .. and i got an error...  but when i re did it now..  it showed up   i see even though was error..

but i get this error now...    going to try to reboot my HA  see if that fixes it

Can't install ghcr.io/esphome/esphome-hassio-amd64:2021.12.1: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2021.12.1&fromImage=ghcr.io%2Fesphome%2Fesphome-hassio-amd64: Internal Server Error ("Get "https://ghcr.io/v2/": dial tcp: lookup ghcr.io: no such host")

jesserockz, 2021-12-21:
@comet424 That is a different/unrelated issue with supervisor. You will need to check there for that, but I think trying again (maybe even a few times) should sort it

comet424, 2021-12-21:
@sdaudet-ua 
@jesserockz 
got it installed
guess the new HA broke the esphome.  i know it broke my volumio's  you can no longer detect 2 volumios... dont know which version broke itlol

but be nice is where you see the 2021.12.1     a nice drop down box  where you select the version  and then click install. so its really easy..  like  in Unraid  you can choose which version to choose to upgrade to or downgrade to.. but a drop down box  thay way ou could chose different versions   say if 2020  was better on certain stuff..

so its installed.. but i get 502 bad gateway..  
and the log has this repeated..
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:52] INFO: Waiting for dashboard to come up...
[00:30:52] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:53] INFO: Waiting for dashboard to come up...
[00:30:53] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:54] INFO: Waiting for dashboard to come up...
[00:30:54] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:55] INFO: Waiting for dashboard to come up...
[00:30:55] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:56] INFO: Waiting for dashboard to come up...
[00:30:56] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:57] INFO: Waiting for dashboard to come up...
[00:30:57] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:58] INFO: Waiting for dashboard to come up...
[00:30:58] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:30:59] INFO: Waiting for dashboard to come up...
[00:30:59] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:31:00] INFO: Waiting for dashboard to come up...
[00:31:00] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:31:01] INFO: Waiting for dashboard to come up...
[00:31:01] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:31:03] INFO: Waiting for dashboard to come up...
[00:31:03] INFO: Starting NGINX...
nginx: [emerg] invalid port in ":" of the "listen" directive in /etc/nginx/servers/ingress.conf:2
[00:31:04] INFO: Waiting for dashboard to come up...
[00:31:04] INFO: Starting NGINX...

dougle03, 2021-12-21:
> guess the new HA broke the esphome

Nope: It's broken even when installed in a totally separate Docker instance.

dougle03, 2021-12-21:
> It seems something went weirdly wrong with the release. I am looking into it but please restore the HA addon backup from ESPHome 2021.12.1 or use the docker image by 2021.12.1 tag until I can remove the 2021.12.2 ones.

Worth noting the dev tag also fails to start....

jesserockz, 2021-12-21:
Yeah I am about to try figure out exactly where/when it started breaking.

@comet424 That is a different issue, please create a new ticket.

nlstrydom, 2021-12-21:
The ESP8266 data is still coming through, it is just the web UI that is down on my side. Couldn't restore the older version as it says it is not compliant. Might be worth just waiting for the problem to be fixed since I have comms to the devices.

greyhound3, 2021-12-21:
After stopping and uninstalling 2021.12.2, clicking on add Integrations, clicking the tree dots and clicking reload, it displayed the 2021.12.1 version. Installed this, everything is good again for me...

nlstrydom, 2021-12-21:
Thanks, that worked for me as well. I clicked the three dots on the HACS side for some dumb reason :(

Cajeen, 2021-12-21:
Same issue with same add on error.
Dashboard error: 502: Bad Gateway
Each dashboard visit posts supervisor log error: 21-12-21 01:23:15 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.32.1:63579 ssl:default [Connect call failed ('172.30.32.1', 63579)]

rolled back a version

comet424, 2021-12-21:
@jesserockz ok posted a new topic..  ya reboot didnt fix..... probably like 1 little thing that causes the issue too..  
used to have that when i used to program back in day with Visual Basic 1 thing and it all breaks

darkfessing, 2021-12-21:
l`vei faced the same bug. ESPHome on Docker, updated 14 hours ago.

closed 2021-12-22 by jesserockz

SeByDocKy, 2021-12-24:
Still have the same error with Hassio (Odroid C4) and last nightly dev build, despite an uninstall and a fresh reinstallation

http://pastebin.fr/99732

pedrolamas, 2021-12-28:
I use a custom docker stack to serve all IoT needs, and one of those containers is ESPHome.

When I pulled the `latest` I started experiencing this issue, and as the previous version did not have the problem, I just switched for that one instead, while I waited for a new `latest` version to be released.

Now I do appreciate all the work in ESPHome (the reason I pay for Nabu Casa even though I don't use it), but having said that, I don't see how this issue was closed if the problem is still not fixed... anyone pulling the latest docker container version will see this problem!

StijnVdd, 2021-12-28:
Have the same issue when starting 2021.12.2

Like pedrolamas I'm running a custom stack