esphome/issues #2858: ESPHome Failing to Start - Package Not Found ESPHome

bnichols, 2021-12-21:
### The problem

After installing ESPHome 2021.12.2 from supervisor...

```
21-12-20 20:44:29 INFO (MainThread) [supervisor.addons] Creating Home Assistant add-on data folder /data/addons/data/15ef4d2f_esphome
21-12-20 20:44:29 INFO (SyncWorker_3) [supervisor.docker.interface] Downloading docker image ghcr.io/esphome/esphome-hassio-armv7 with tag 2021.12.2.
21-12-20 20:46:03 INFO (MainThread) [supervisor.addons] Add-on '15ef4d2f_esphome' successfully installed

```
 then clicking start, the following error appears in the log:

```
[21:03:01] INFO: Starting ESPHome dashboard...
Traceback (most recent call last):
  File "/usr/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/usr/bin/esphome", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: esphome
```

My configuration below:

## System Health

version | core-2021.12.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v7
arch | armv7l
timezone | America/New_York

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 57.8 GB
disk_used | 4.3 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.0.0), Terminal & SSH (8.5.4), Mosquitto broker (5.1), ESPHome (2021.12.2)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

### Which version of ESPHome has the issue?

2021.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.3

### What platform are you using?

ESP32

goncalvespedro, 2021-12-21:
Having the same issue

comet424, 2021-12-21:
same can not program my pages.. also gett 500 bad gateway error too

wallravit, 2021-12-21:
Having the same issue, ESPHome on my Home Assistant can't be started.

jesserockz, 2021-12-21:
Closing as a duplicate of #2856 as it seems to be related to all the docker images.

closed 2021-12-21 by jesserockz

adangster1, 2021-12-21:
same here, new install.. wont start the plugin gui ..same error

Aliskin-Papa, 2021-12-21:
Sam here

Cajeen, 2021-12-21:
Same issue with same add on error. 
Dashboard error: 502: Bad Gateway
Each time I go to dashboard get supervisor log error: 21-12-21 01:23:15 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.32.1:63579 ssl:default [Connect call failed ('172.30.32.1', 63579)]