esphome/issues #2860: ESPHome dashboard can not open

kylinfedora, 2021-12-21:
### The problem

Starting ESPHome dashboard
Traceback (most recent call last):
  File "/usr/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/usr/bin/esphome", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: esphome

### Which version of ESPHome has the issue?

2021.12.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.4

### What platform are you using?

ESP8266

dougle03, 2021-12-21:
See issue #2858 - please close this one as the issue only needs one report.

jesserockz, 2021-12-21:
Duplicate of #2858

closed 2021-12-21 by jesserockz