esphome/issues #2861: xiaomi_hhccjcy01 - no data reported

nohn, 2021-12-21:
### The problem

I ordered a new xiaomi_hhccjcy01 sensor a few days ago. Unlike my older sensors, the new one doesn't report data to esphome.

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12.3

### What platform are you using?

ESP32

### Board

ESP32 Dev Board NodeMCU WROOM32

### Component causing the issue

xiaomi_hhccjcy01

### Example YAML snippet

```yaml
- platform: xiaomi_hhccjcy01
    mac_address: 'c4:7c:8d:6d:f5:d6'
    temperature:
      name: "Küche Blumenkasten Links Temperature"
    moisture:
      name: "Küche Blumenkasten Links Moisture"
    illuminance:
      name: "Küche Blumenkasten Links Illuminance"
    conductivity:
      name: "Küche Blumenkasten Links Conductivity"
```

### Anything in the logs that might be useful for us?

```txt
[...]
[09:57:16][C][xiaomi_hhccjcy01:012]: Xiaomi HHCCJCY01
[09:57:16][C][xiaomi_hhccjcy01:013]:   Temperature 'Küche Blumenkasten Links Temperature'
[09:57:16][C][xiaomi_hhccjcy01:013]:     Device Class: 'temperature'
[09:57:16][C][xiaomi_hhccjcy01:013]:     State Class: 'measurement'
[09:57:16][C][xiaomi_hhccjcy01:013]:     Unit of Measurement: '°C'
[09:57:16][C][xiaomi_hhccjcy01:013]:     Accuracy Decimals: 1
[09:57:16][C][xiaomi_hhccjcy01:014]:   Moisture 'Küche Blumenkasten Links Moisture'
[09:57:16][C][xiaomi_hhccjcy01:014]:     State Class: 'measurement'
[09:57:16][C][xiaomi_hhccjcy01:014]:     Unit of Measurement: '%'
[09:57:16][C][xiaomi_hhccjcy01:014]:     Accuracy Decimals: 0
[09:57:16][C][xiaomi_hhccjcy01:015]:   Conductivity 'Küche Blumenkasten Links Conductivity'
[09:57:16][C][xiaomi_hhccjcy01:015]:     State Class: 'measurement'
[09:57:16][C][xiaomi_hhccjcy01:015]:     Unit of Measurement: 'µS/cm'
[09:57:16][C][xiaomi_hhccjcy01:015]:     Accuracy Decimals: 0
[09:57:16][C][xiaomi_hhccjcy01:015]:     Icon: 'mdi:flower'
[09:57:16][C][xiaomi_hhccjcy01:016]:   Illuminance 'Küche Blumenkasten Links Illuminance'
[09:57:16][C][xiaomi_hhccjcy01:016]:     Device Class: 'illuminance'
[09:57:16][C][xiaomi_hhccjcy01:016]:     State Class: 'measurement'
[09:57:16][C][xiaomi_hhccjcy01:016]:     Unit of Measurement: 'lx'
[09:57:16][C][xiaomi_hhccjcy01:016]:     Accuracy Decimals: 0
[...]
[09:57:34][D][esp32_ble_tracker:719]: Found device C4:7C:8D:6D:F5:D6 RSSI=-79
[09:57:34][D][esp32_ble_tracker:740]:   Address Type: PUBLIC
[09:57:34][D][esp32_ble_tracker:742]:   Name: 'Flower care'

And that's it for this sensor. 

Other sensors report data:

[...]
[09:58:39][D][sensor:113]: 'Schlafzimmer Blumenkasten Links Temperature': Sending state 0.50000 °C with 1 decimals of accuracy
[09:58:42][D][xiaomi_ble:311]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6D:C3:67):
[09:58:42][D][xiaomi_ble:323]:   Conductivity: 116µS/cm
[09:58:42][D][sensor:113]: 'Kinderzimmer Blumenkasten Links Conductivity': Sending state 116.00000 µS/cm with 0 decimals of accuracy
[09:58:43][D][xiaomi_ble:311]: Got Xiaomi LYWSDCGQ (58:2D:34:39:DD:8A):
[09:58:43][D][xiaomi_ble:317]:   Humidity: 52.9%
[09:58:43][D][sensor:113]: 'Schlafzimmer Bett Humidity': Sending state 52.90000 % with 1 decimals of accuracy
[09:58:54][D][xiaomi_ble:311]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6D:C3:74):
[09:58:54][D][xiaomi_ble:326]:   Illuminance: 2224lx
[09:58:54][D][xiaomi_ble:341]:   Light: off
[09:58:54][D][sensor:113]: 'Schlafzimmer Blumenkasten Links Illuminance': Sending state 2224.00000 lx with 0 decimals of accuracy
[09:59:06][D][xiaomi_ble:311]: Got Xiaomi LYWSDCGQ (58:2D:34:39:D8:46):
[09:59:06][D][xiaomi_ble:314]:   Temperature: 24.2°C
[09:59:06][D][xiaomi_ble:317]:   Humidity: 35.7%
[...]
```

### Additional information

bluetoothctl on a Linux box also doesn't show data sent for this sensor. So I guess, the firmware has changed again and something needs to be sent to the sensor in order for it to report data.

jesserockz, 2021-12-22:
Quite possible that a firmware update has done something. The xiaomi app reads generally by connecting and requesting the data via BLE GATT.

If you try changing the logging to VERY_VERBOSE, you might get a specific error when it tries to decode the advertisement from this device.

nohn, 2021-12-22:
```
[07:09:13][VV][esp32_ble_tracker:493]: Parse Result:
[07:09:13][VV][esp32_ble_tracker:510]:   Address: C4:7C:8D:6D:F5:D6 (PUBLIC)
[07:09:13][VV][esp32_ble_tracker:512]:   RSSI: -83
[07:09:13][VV][esp32_ble_tracker:513]:   Name: 'Flower care'
[07:09:13][VV][esp32_ble_tracker:521]:   Ad Flag: 6
[07:09:13][VV][esp32_ble_tracker:524]:   Service UUID: 0xFE95
[07:09:13][VV][esp32_ble_tracker:538]:   Service data:
[07:09:13][VV][esp32_ble_tracker:539]:     UUID: 0xFE95
[07:09:13][VV][esp32_ble_tracker:540]:     Data: 71.20.98.00.00.D6.F5.6D.8D.7C.C4.0D (12)
[07:09:13][VV][esp32_ble_tracker:543]: Adv data: 02.01.06.03.02.95.FE.0F.16.95.FE.71.20.98.00.00.D6.F5.6D.8D.7C.C4.0D.0C.09.46.6C.6F.77.65.72.20.63.61.72.65 (36)
[07:09:13][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[07:09:13][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[07:09:13][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
```

For other devices, I see this:

```
[07:09:59][VV][esp32_ble_tracker:493]: Parse Result:
[07:09:59][VV][esp32_ble_tracker:510]:   Address: C4:7C:8D:6D:C3:67 (PUBLIC)
[07:09:59][VV][esp32_ble_tracker:512]:   RSSI: -77
[07:09:59][VV][esp32_ble_tracker:513]:   Name: 'Flower care'
[07:09:59][VV][esp32_ble_tracker:521]:   Ad Flag: 6
[07:09:59][VV][esp32_ble_tracker:524]:   Service UUID: 0xFE95
[07:09:59][VV][esp32_ble_tracker:538]:   Service data:
[07:09:59][VV][esp32_ble_tracker:539]:     UUID: 0xFE95
[07:09:59][VV][esp32_ble_tracker:540]:     Data: 71.20.98.00.B1.67.C3.6D.8D.7C.C4.0D.09.10.02.12.00 (17)
[07:09:59][VV][esp32_ble_tracker:543]: Adv data: 02.01.06.03.02.95.FE.14.16.95.FE.71.20.98.00.B1.67.C3.6D.8D.7C.C4.0D.09.10.02.12.00.0C.09.46.6C.6F.77.65.72.20.63.61.72.65 (41)
[07:09:59][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[07:09:59][VV][xiaomi_hhccjcy01:025]: parse_device(): MAC address C4:7C:8D:6D:C3:67 found.
[07:09:59][D][xiaomi_ble:311]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6D:C3:67):
[07:09:59][D][xiaomi_ble:323]:   Conductivity: 18µS/cm
```

nohn, 2021-12-22:
Similar when using bluetoothctl:

The problematic device:

```
[NEW] Device C4:7C:8D:6D:F5:D6 Flower care
```
(and that's it)

Any other device:

```
[NEW] Device C4:7C:8D:6D:C3:74 Flower care
[CHG] Device C4:7C:8D:6D:C3:74 ServiceData Key: 0000fe95-0000-1000-8000-00805f9b34fb
[CHG] Device C4:7C:8D:6D:C3:74 ServiceData Value:
  71 20 98 00 64 74 c3 6d 8d 7c c4 0d 09 10 02 2c  q ..dt.m.|.....,
  00                              
...
[CHG] Device C4:7C:8D:6D:C3:74 ServiceData Key: 0000fe95-0000-1000-8000-00805f9b34fb
[CHG] Device C4:7C:8D:6D:C3:74 ServiceData Value:
  71 20 98 00 66 74 c3 6d 8d 7c c4 0d 07 10 03 9a  q ..ft.m.|......
  01 00                    
```

jesserockz, 2021-12-22:
Can you try make this change and get the log again?
It seems that the address does not match for some reason.

```diff
--- a/esphome/components/xiaomi_hhccjcy01/xiaomi_hhccjcy01.cpp
+++ b/esphome/components/xiaomi_hhccjcy01/xiaomi_hhccjcy01.cpp
@@ -19,7 +19,7 @@ void XiaomiHHCCJCY01::dump_config() {
 
 bool XiaomiHHCCJCY01::parse_device(const esp32_ble_tracker::ESPBTDevice &device) {
   if (device.address_uint64() != this->address_) {
-    ESP_LOGVV(TAG, "parse_device(): unknown MAC address.");
+    ESP_LOGVV(TAG, "parse_device(): unknown MAC address. 0x%12X != 0x%12X", device.address_uint64(), this->address_);
     return false;
   }
   ESP_LOGVV(TAG, "parse_device(): MAC address %s found.", device.address_str().c_str());
```

nohn, 2021-12-22:
```
[09:48:45][VV][esp32_ble_tracker:493]: Parse Result:
[09:48:45][VV][esp32_ble_tracker:510]:   Address: C4:7C:8D:6D:F5:D6 (PUBLIC)
[09:48:45][VV][esp32_ble_tracker:512]:   RSSI: -54
[09:48:45][VV][esp32_ble_tracker:513]:   Name: 'Flower care'
[09:48:45][VV][esp32_ble_tracker:521]:   Ad Flag: 6
[09:48:45][VV][esp32_ble_tracker:524]:   Service UUID: 0xFE95
[09:48:45][VV][esp32_ble_tracker:538]:   Service data:
[09:48:45][VV][esp32_ble_tracker:539]:     UUID: 0xFE95
[09:48:45][VV][esp32_ble_tracker:540]:     Data: 71.20.98.00.00.D6.F5.6D.8D.7C.C4.0D (12)
[09:48:45][VV][esp32_ble_tracker:543]: Adv data: 02.01.06.03.02.95.FE.0F.16.95.FE.71.20.98.00.00.D6.F5.6D.8D.7C.C4.0D.0C.09.46.6C.6F.77.65.72.20.63.61.72.65 (36)
[09:48:45][VV][xiaomi_lywsdcgq:020]: parse_device(): unknown MAC address.
[09:48:45][VV][xiaomi_lywsdcgq:020]: parse_device(): unknown MAC address.
[09:48:45][VV][xiaomi_lywsdcgq:020]: parse_device(): unknown MAC address.
[09:48:45][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address. 0x    8D6DF5D6 != 0x        C47C
[09:48:45][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address. 0x    8D6DF5D6 != 0x        C47C
[09:48:45][VV][xiaomi_hhccjcy01:026]: parse_device(): MAC address C4:7C:8D:6D:F5:D6 found.
[09:48:45][VV][xiaomi_ble:102]: parse_xiaomi_message(): payload has wrong size (0)!
```

jesserockz, 2021-12-22:
So it seems like sometimes the mac is parsed weirdly, and it did find the device correctly, but then the payload was wrong/mangled. 

I will try to take another look at that data you posted tomorrow and see if I can work it out.

nohn, 2021-12-22:
@jesserockz if you like, I can Paypal you the money to order one to your address (this is way cheaper than sending one from Germany to NZ)

remyderuysscher, 2022-05-03:
I have this problem too, it's only 1 device

burnbabyburn, 2022-07-12:
error still persists :(

> So it seems like sometimes the mac is parsed weirdly, and it did find the device correctly, but then the payload was wrong/mangled.
> 
> I will try to take another look at that data you posted tomorrow and see if I can work it out.

provi1, 2023-07-25:
i have the same problem.

Also esp32_ble_tracker: only is not showing any bluetooth MAC informations and device names. even with 
logger:
  level: VERY_VERBOSE 
it´s not working.

only way to see the scan result is using this workaround:
text_sensor:
  - platform: ble_scanner
    name: "BLE Devices Scanner"  

this show up some bluetooth devices, but not the flower mi one.
at the same time i can find it with windows 10 pc, linux maschine and iPhone.
so i used the MAC Address from one of these sources.

but now i got:
[20:47:26][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[20:47:26][W][component:204]: Component esp32_ble_tracker took a long time for an operation (0.05 s).
[20:47:26][W][component:205]: Components should block for at most 20-30ms.
[20:47:26][V][esp32_ble:178]: (BLE) gap_event_handler - 3

The FloweMi Sensor has firmware v3.3.6
I´m Using HA 2023.7.1