esphome/issues #2863: DHT sensor configured via port expander (sx1509) errors out during build

mirko, 2021-12-21:
### The problem

I'm having a SX1509 GPIO expander configured via which I'm trying to configure a DHT22 temp/humid sensor.
The config file check passes, though during building the firmware the process errors out with:

```src/main.cpp: In function 'void setup()':
src/main.cpp:256:40: error: no matching function for call to 'esphome::dht::DHT::set_pin(esphome::sx1509::SX1509GPIOPin*&)'
   dht_dht->set_pin(sx1509_sx1509gpiopin);
                                        ^
src/main.cpp:256:40: note: candidate is:
In file included from src/esphome.h:2:0,
                 from src/main.cpp:3:
src/esphome/components/dht/dht.h:39:8: note: void esphome::dht::DHT::set_pin(esphome::InternalGPIOPin*)
   void set_pin(InternalGPIOPin *pin) { pin_ = pin; }
        ^
src/esphome/components/dht/dht.h:39:8: note:   no known conversion for argument 1 from 'esphome::sx1509::SX1509GPIOPin*' to 'esphome::InternalGPIOPin*'
```

### Which version of ESPHome has the issue?

2021.10.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

sensor:dht - sx1509

### Example YAML snippet

```yaml
sx1509:
  - id: sx1509_hub
    address: 0x3E

sensor:
  - platform: dht
    #model: AM2302
    pin:
      sx1509: sx1509_hub
      number: 8
    temperature:
      name: "temp"
    humidity:
      name: "humid"
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp: In function 'void setup()':
src/main.cpp:256:40: error: no matching function for call to 'esphome::dht::DHT::set_pin(esphome::sx1509::SX1509GPIOPin*&)'
   dht_dht->set_pin(sx1509_sx1509gpiopin);
                                        ^
src/main.cpp:256:40: note: candidate is:
In file included from src/esphome.h:2:0,
                 from src/main.cpp:3:
src/esphome/components/dht/dht.h:39:8: note: void esphome::dht::DHT::set_pin(esphome::InternalGPIOPin*)
   void set_pin(InternalGPIOPin *pin) { pin_ = pin; }
        ^
src/esphome/components/dht/dht.h:39:8: note:   no known conversion for argument 1 from 'esphome::sx1509::SX1509GPIOPin*' to 'esphome::InternalGPIOPin*'
```

jesserockz, 2021-12-22:
I don't tink this would work. The DHTs require very precise timing of the pin reading and writing and thats not something that will work via an expander.

So really the bug here is that the config should not be valid.

mirko, 2021-12-22:
> So really the bug here is that the config should not be valid.

Hence the report focusing on the build failing, however before limiting that based on assumptions, can we verify the expander will mess with respective timing requirements first?

jesserockz, 2021-12-22:
> is there the possibility we can verify the expander messing with respective timing requirements first?

The DHT timing pulses are in microseconds, not something you can do while also communicating over I2C or SPI with an expander.

closed 2021-12-22 by jesserockz