esphome/issues #2865: ESPHome Update broke Aukey Switches

whizzzkid, 2021-12-22:
### The problem

I was using https://www.esphome-devices.com/devices/Aukey-SH-PA1-Smart-Plug/ configuration to run the switches, after the most recent update something seems to have messed up all of my switches. The relay doesn't turn on, neither does the `blue` led.

### Which version of ESPHome has the issue?

>=2021.11.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

jesserockz, 2021-12-22:
You need to provide more information (config/logs/etc), otherwise nobody can help you. That's why we have those fields in the issue form.

whizzzkid, 2021-12-22:
@jesserockz agreed however I don't have anything other than that. The same config (in the link) was working for more than a year and now it's not. Looks like there was a regression which is not reported in the logs and cannot be diagnosed as the logs display the normal behavior. However, it doesn't toggle the relay physically. The logs insists that the switch is on, but not really.

whizzzkid, 2022-01-04:
@jesserockz any updates on this?

megabitdragon, 2022-01-05:
I have the exact same issue after updating the ESPHome addon to version 2021.12.3, also using EPS8266 platform.

megabitdragon, 2022-01-05:
One more piece of information, I rolled back to version 1.18.0 and everything works as expected.

jesserockz, 2022-01-05:
One thing I just noticed is that you have GPIO15 (the relay) set up as both an output and a binary_sensor. 
A pin cannot do both input and output at the same time and is bound to cause issues. 
Move your `on_state` code to `on_turn_on` and `on_turn_off` of the light and try again.

I see you got that from the esphome-devices template so it is not your fault, but it is just plain wrong and should not be done like that.

whizzzkid, 2022-01-05:
@jesserockz that was absolutely it:

```sh
# Basic Config
esphome:
  name: switch-esp-hall-light-001
  platform: ESP8266
  board: esp01_1m
  platformio_options:
    upload_speed: 115200

wifi:
  ssid: "[REDACTED]"
  password: "[REDACTED]"
  
  # Optional manual IP
  manual_ip:
    static_ip: [REDACTED]
    gateway: [REDACTED]
    subnet: [REDACTED]

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "[REDACTED]"
    password: "[REDACTED]"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
ota:

output:
  - platform: gpio
    pin:
      number: 15
    id: plug_outlet

  - platform: gpio
    pin:
      number: GPIO2
      inverted: yes
    id: blue_led

light:
  - platform: binary
    name: "Hall Light 001"
    id: switch_esp_hall_light_001
    output: plug_outlet
    on_turn_on:
      - output.turn_on: blue_led
    on_turn_off:
      - output.turn_off: blue_led

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO13
    id: switch_esp_hall_light_001_button
    on_press:
      light.toggle: switch_esp_hall_light_001

status_led:
  # Red LED
  pin:
    number: GPIO0
    inverted: yes
```

This now works.

closed 2022-01-05 by whizzzkid